ms-iot/ROSOnWindows #6: [roswtf] WARNING Communication with node [/rosout] is very slow

kejxu, 2018-10-10:
```
C:\temp\test_ws\src\my_tut>roswtf
Loaded plugin tf.tfwtf
Package: my_tut
================================================================================
Static checks summary:
Found 1 error(s).
[31m[1mERROR[0m Local network configuration is invalid: Local hostname [DESKTOP-E0BPROA] resolves to [2001:4898:d8:27:d993:bad8:bf0d:182f,fe80::5464:3b68:972d:8714%7,2001:4898:d8:27:5464:3b68:972d:8714], which does not appear to be a local IP address ['10.131.8.171'].
================================================================================
Beginning tests of your ROS graph. These may take awhile...
analyzing graph...
... done analyzing graph
running graph rules...
ERROR: connection refused to [http://DESKTOP-E0BPROA:60645/]
... done running graph rules
Online checks summary:
Found 2 warning(s).
Warnings are things that may be just fine, but are sometimes at fault
[1mWARNING[0m Communication with node [/rosout] is very slow
[1mWARNING[0m The following nodes are unexpectedly connected:
 * /rosout->/rqt_gui_py_node_8208 (/rosout_agg)
```

ooeygui, 2019-05-17:
This is related to localhost

seanyen, 2019-06-07:
It could be related to: https://github.com/ms-iot/ros_comm/pull/56

imcnanie, 2019-09-04:
Any updates on this?

ooeygui, 2019-09-19:
@imcnanie We changed the setup script to use 127.0.01 which prevented a DNS lookup on each connection.

seanyen, 2020-02-24:
I am closing it now since the fix is merged by the upstream.

closed 2020-02-24 by seanyen