ms-iot/ROSOnWindows #30: [ros_comm] class_loader refers to a vector which has a dynmamic sized ABI

ooeygui, 2018-10-25:
**Describe the bug**
class_loader has a std::vector in its interface, which when exported on Windows results in a build break:

`warning C4251: 'class_loader::impl::AbstractMetaObjectBase::associated_class_loaders_': class 'std::vector<class_loader::ClassLoader *,std::allocator<_Ty>>' needs to have dll-interface to be used by clients of class 'class_loader::impl::AbstractMetaObjectBase'`

the associated_class_loaders member should be allocated, and a pointer stored.
`ClassLoaderVector associated_class_loaders_;`

ooeygui, 2018-10-25:
Actually safe to ignore.

traversaro, 2019-01-15:
Related comments on how to disable `C4251` in a clean way:  https://github.com/robotology/yarp/pull/1647#issuecomment-381676728 .

closed 2019-10-24 by ooeygui