ms-iot/ROSOnWindows #31: It is not possible to compile with Debug configuration a C++ project that links TinyXML

traversaro, 2018-10-28:
**Describe the bug**
Any C++ project that links to TinyXML fails to compile in Debug, because only the Release version of TinyXML is provided in `tinyxml` package in the chocolatey `ros-win` channel.

**To Reproduce**
After executing `setup.bat`, try to compile in Debug a simple project that uses TinyXML https://gist.github.com/traversaro/c27938365ca06f0f95facc887275a42e : 
~~~
git clone https://gist.github.com/traversaro/c27938365ca06f0f95facc887275a42e test_tinyxml
cd test_tinyxml
md build
cd build 
cmake -A x64 ..
cmake --build . --config Debug
~~~
The last command results in  the following error: 
~~~

C:\src\test_tinyxml\build>cmake --build . --config Debug
<...>

Build FAILED.

"C:\src\test_tinyxml\build\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\src\test_tinyxml\build\tinyxml_test.vcxproj" (default target) (3) ->
(Link target) ->
  LINK : warning LNK4098: defaultlib 'MSVCRT' conflicts with use of other libs; use /NODEFAULTLIB:library [C:\src\test_tinyxml\build\tinyxml_test.vcxproj]

"C:\src\test_tinyxml\build\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\src\test_tinyxml\build\tinyxml_test.vcxproj" (default target) (3) ->
(Link target) ->
  tinyxml.lib(tinyxml.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in tinyxml_test.obj [C:\src\test_tinyxml\build\tinyxml_test.vcxproj]
  tinyxml.lib(tinyxml.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in tinyxml_test.obj [C:\src\test_tinyxml\build\tinyxml_test.v
cxproj]
  tinyxml.lib(tinyxmlparser.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in tinyxml_test.obj [C:\src\test_tinyxml\build\tinyxml_test.vcxproj]
  tinyxml.lib(tinyxmlparser.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in tinyxml_test.obj [C:\src\test_tinyxml\build\tinyxml_
test.vcxproj]
  tinyxml.lib(tinyxmlerror.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in tinyxml_test.obj [C:\src\test_tinyxml\build\tinyxml_test.vcxproj]
  tinyxml.lib(tinyxmlerror.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in tinyxml_test.obj [C:\src\test_tinyxml\build\tinyxml_t
est.vcxproj]
  C:\src\test_tinyxml\build\Debug\tinyxml_test.exe : fatal error LNK1319: 6 mismatches detected [C:\src\test_tinyxml\build\tinyxml_test.vcxproj]

    1 Warning(s)
    7 Error(s)

Time Elapsed 00:00:05.31

C:\src\test_tinyxml\build>
~~~

**Expected behavior**
I was expecting to be able to compile any C++ program that links ROS dependencies also in Debug configuration.

**Environment information:**
 - OS version: `Windows Version 1709 - OS Build 16299.192`
 - Chocolatey Versions: 
~~~
C:\src\test-tinyxml\build>choco list  --local-only
Chocolatey v0.10.11
assimp 4.0.1.1809242322
boost 1.66.0.1809242326
bullet3 2.87.0.1809242326
bzip2 1.0.6.1809242322
chocolatey 0.10.11
console_bridge 0.4.0.1809242322
cppunit 1.12.1.1809242322
eigen 3.3.4.1809242322
freeglut 3.0.0.1809242322
google-mock 1.8.0.1809242326
google-test 1.8.0.1809242322
libcurl 7.58.0.1809242322
libgraphviz 2.41.0.1810022039
libopencv 3.4.1.1809242322
log4cxx 0.10.0.1809242322
lz4 1.8.1.1809242322
ogre 1.10.11.1809100020
openssl 1.1.1.1809242322
pkg-config 0.29.2.1809242322
poco 1.8.1.1809242322
pyqt5 5.10.1.1809242322
pyside2 5.10.1.1809242322
qt5-sdk 5.10.1.1810112239
ros-catkin-tools 0.0.1.1810092059
ros-cmake 3.11.4.1809242322
ros-melodic-desktop 20181027.1
ros-python2 2.7.15.1809242354
sbcl 0.0.0.1809242326
sdl 1.2.15.1809242322
sdl_image 1.2.12.1809242322
sip 4.19.8.1809242322
tinyxml 2.6.2.1809242322
tinyxml2 6.1.0.1809242322
urdfdom 1.0.0.1809242322
urdfdom_headers 1.0.0.1809242322
vcpython27 9.0.30729.1809090720
yaml-cpp 0.5.3.1809242322
37 packages installed.
~~~

traversaro, 2018-10-28:
Related discourse comment: https://discourse.ros.org/t/moveit-now-enabled-for-ros1-on-windows/6509/2 .

ooeygui, 2018-11-01:
Hi @traversaro,
We only released release binaries of the dependencies. 
Could you let us know what you are looking for from the debug version? Is it just symbols or are you looking for extra validation?

traversaro, 2018-11-01:
Hi @ooeygui , 
whenever I develop new software, I typically compile it in  `Debug` for several reasons:
* as you mention, the possibility of viewing the source code symbols during debugging (and this can be also be done compiling in `RelWithDebInfo` ,
* to enable checks in the code that use the standard [`assert`](https://en.cppreference.com/w/cpp/error/assert) macro are enabled (and this typically is only possible in CMake using the `Debug` mode, as `RelWithDebInfo` defines the `NDEBUG` macro. 

If building C++ code in Debug is not supported using ROSOnWindows, I think it could be beneficial to document this choice. If this is already documented somewhere, my bad, I probably missed it.

ooeygui, 2018-11-02:
Thank you for the reply. Looking at the docs it is not documented. I will fix that. (And I also see some opportunity to make the docs more friendly).

Do you think it is reasonable to have a separate ros and rosdeps path for debug? 
I.e. c:\opt\rosd and c:\opt\rosdepsd?

I'll keep this issue open.

traversaro, 2018-11-03:
> Do you think it is reasonable to have a separate ros and rosdeps path for debug?
> I.e. c:\opt\rosd and c:\opt\rosdepsd?

I guess you mean that `c:\opt\rosdepd` will contain the exact same content of `c:\opt\rosdep``, but compiled in `Debug` mode, that would in theory work, but with several downsides: 
* The Visual Studio CMake generator is a multi-config generator (see https://cgold.readthedocs.io/en/latest/glossary/multi-config.html), meaning that the configuration used for compilation is not chosen at CMake configuration time, but rather can be changed directly from the GUI. Having two completely different installation prefixes for Release and Debug would mean that it would be necessary to run again CMake (and sourcing a different `setup.bat`) when switching from Release to Debug
* All the content that are shared between Release and Debug builds (headers, executables, resources) would be duplicated, using in an unnecessary way disk space and possibly confusing the users

The strategy typically used in this cases (as far as I know) instead is to install both Debug and Release libraries in the same prefix. In that case it is clearly necessary to avoid naming collision between Debug and Release libraries, and there are two solution that I am aware of: 
* Add a postfix to the name of the debug libraries. For library that use CMake to build, this can be done by setting [`CMAKE_DEBUG_POSTFIX`](https://cmake.org/cmake/help/v3.10/variable/CMAKE_DEBUG_POSTFIX.html) to `d` or some similar suffix.
* Install debug libraries in `<prefix>/debug/lib` and `<prefix>/debug/bin` instead of `bin` and `lib`. This is the solution used by [`vcpkg`](https://github.com/Microsoft/vcpkg)  

In case the library exports CMake targets, the loading of the libraries for all configuration is typically ensured by this code: 
~~~
# Load information for each installed configuration.
get_filename_component(_DIR "${CMAKE_CURRENT_LIST_FILE}" PATH)
file(GLOB CONFIG_FILES "${_DIR}/<Package>Targets-*.cmake")
foreach(f ${CONFIG_FILES})
  include(${f})
endforeach()
~~~
that is automatically generated by the CMake `install(EXPORT` command. 

For library that do not install their CMake exported targets, the assumption about the debug names is typically contained in the `Find<Package>.name` script, see for example: https://github.com/Kitware/CMake/blob/d9190f3308fb67f01f061038a06006af73a77da1/Modules/FindProtobuf.cmake#L267 .

Another advantage of not going for the two prefixes for Release and Debug is that for libraries that have an ABI compatible between Release and Debug (for example libraries that expose a C-based API, or that do not use STL containers in the API) it is possible to just ship the Debug version.

ooeygui, 2020-11-20:
Closing out this issue as it has been corrected.

closed 2020-11-20 by ooeygui