ms-iot/ROSOnWindows #39: eigen include directory

gladijos, 2018-11-16:
**Describe the bug**
today eigen include directory should be equal C:/opt/rosdeps/x64/include/eigen3. It required by  files C:\opt\ros\melodic\x64\lib\pkgconfig\orocos-kdl.pc and C:\opt\ros\melodic\x64\lib\pkgconfig\orocos_kdl.pc

**To Reproduce**
*call " choco upgrade ros-melodic-desktop -y"

**Expected behavior**
files orocos_kdl.pc  and orocos-kdl.pc shouldn't contain absolute paths.

**Environment information:**
 - OS version: `windows 10 x64 Pro`
 - Chocolatey Versions: 
Chocolatey v0.10.11
assimp 4.0.1.1809242322
boost 1.66.0.1809242326
bullet3 2.87.0.1809242326
bzip2 1.0.6.1809242322
chocolatey 0.10.11
chocolatey-core.extension 1.3.3
console_bridge 0.4.0.1809242322
cppunit 1.12.1.1809242322
eigen 3.3.4.1809242322
git 2.19.1
git.install 2.19.1
google-mock 1.8.0.1809242326
google-test 1.8.0.1809242322
libcurl 7.58.0.1809242322
libgraphviz 2.41.0.1810022039
libopencv 3.4.1.1809242322
log4cxx 0.10.0.1809242322
lz4 1.8.1.1809242322
ogre 1.10.11.1809100020
openssl 1.1.1.1809242322
pkg-config 0.29.2.1809242322
poco 1.8.1.1809242322
pyqt5 5.10.1.1809242322
pyside2 5.10.1.1809242322
qt5-sdk 5.10.1.1810112239
ros-catkin-tools 0.0.1.1811160029
ros-cmake 3.11.4.1809242322
ros-melodic-desktop 20181116.1
ros-python2 2.7.15.1809242354
sbcl 0.0.0.1809242326
sdl 1.2.15.1809242322
sdl_image 1.2.12.1809242322
sip 4.19.8.1809242322
tinyxml 2.6.2.1809242322
tinyxml2 6.1.0.1809242322
urdfdom 1.0.0.1809242322
urdfdom_headers 1.0.0.1809242322
vcpython27 9.0.30729.1809090720
yaml-cpp 0.5.3.1809242322

gladijos, 2018-11-16:
Actually if you have installed smthing that use eigen before the folder "C:\opt\rosdeps\x64\include\eigen3" wouldn't exist. - this is the main problem.

seanyen-msft, 2018-11-16:
If you have orocos_kdl installed (which is a catkin package), then rosdep should resolve eigen as a dependency for you. Can you elaborate more what's the use case you are hitting problems?

gladijos, 2018-11-17:
I had installed ros2 on win10. I installed it with choco using. I think choco installed eigen for ros2. After this I tried to install ros1 with choco help and this path "C:\opt\rosdeps\x64\include\eigen3" hadn't existed after installation. But I noticed that there are eigen3 folder and archive exist in choco directories ("C:\ProgramData\choco..."). After this I deleted choco directory and C:\opt\ directory at all. I installed choco and ros2 from white list. and everything went fine!

ooeygui, 2018-11-25:
Thank you for this bug. ROS1 and ROS2 must co-exist.

IoTDan, 2022-10-11:
Resolved in current builds. Please reactivate if still an issue

closed 2022-10-11 by IoTDan