ms-iot/ROSOnWindows #44: rviz fail to start because it could not find or load the Qt platform plugin

bbw2008927, 2018-12-22:
Hi,
After install binary of ros-win 20181221.1.
But I try to run rosrun rviz rviz. I get error, can't find Qt plateform.  and I have run setup.bat before rosrun.
roscore have run without no problem in other.
The procedure of install, choco upgrade ros-melodic-desktop -y, no error happen.
I can find folder of C:\opt\rosdeps\x64\plugins\platforms
And add C:\opt\rosdeps\x64 into path of OS. But there is still issue.

seanyen-msft, 2018-12-22:
Thank for trying out ROS on Windows!

As of the error which cannot find QT platform, can you describe more what message you saw? Or screenshot?

Also it would be helpful to post the following output about your environment:
```
:: assume you have roscore running in another command prompt.
:: first, run the setup.bat
c:\opt\ros\melodic\x64\setup.bat

:: dump the following env variables
set ROS
set QT
set PATH

:: dump what's installed locally
choco list -lo

:: enable verbose logging for rviz
rosrun rviz rviz -l
```

bbw2008927, 2018-12-22:
Hi, 
Info of env variable:
1. ROS:
ROS_DISTRO=melodic
ROS_ETC_DIR=C:/opt/ros/melodic/x64/etc/ros
ROS_MASTER_URI=http://localhost:11311
ROS_PACKAGE_PATH=C:/opt/ros/melodic/x64\share
ROS_PYTHON_VERSION=2
ROS_ROOT=C:/opt/ros/melodic/x64/share/ros
ROS_VERSION=1
2. Qt, Only one line.
QtMsBuild=C:\Users\Administrator\AppData\Local\QtMsBuild
3. My PATH is very long, so I pick all the ROS involved outpu.
The detail file is attached.
C:/opt/ros/melodic/x64\lib;
C:/opt/rosdeps/x64\lib;
C:/opt/ros/melodic/x64\bin;
C:/opt/rosdeps/x64\bin;
C:\opt\python27amd64\;
C:\opt\python27amd64\Scripts;
C:\ProgramData\chocolatey\bin;
C:\opt\rosdeps\x64;
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin;
C:\Program Files (x86)\Microsoft Visual 
[PATH.txt](https://github.com/ms-iot/ROSOnWindows/files/2704936/PATH.txt)
Studio\2017\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\Ninja
4. The choco list result:
Chocolatey v0.10.11
assimp 4.0.1.1809242322
boost 1.66.0.1809242326
bullet3 2.87.0.1809242326
bzip2 1.0.6.1809242322
chocolatey 0.10.11
chocolatey-core.extension 1.3.3
console_bridge 0.4.0.1809242322
cppunit 1.12.1.1809242322
eigen 3.3.4.1809242322
git 2.20.1
git.install 2.20.1
google-mock 1.8.0.1809242326
google-test 1.8.0.1809242322
libcurl 7.58.0.1809242322
libgraphviz 2.41.0.1810022039
libopencv 3.4.1.1809242322
log4cxx 0.10.0.1809242322
lz4 1.8.1.1809242322
ogre 1.10.11.1809100020
openssl 1.1.1.1809242322
pkg-config 0.29.2.1809242322
poco 1.8.1.1809242322
pyqt5 5.10.1.1809242322
pyside2 5.10.1.1809242322
qt5-sdk 5.10.1.1810112239
ros-catkin-tools 0.0.1.1811160029
ros-cmake 3.11.4.1809242322
ros-melodic-desktop 20181221.1
ros-python2 2.7.15.1809242354
sbcl 0.0.0.1809242326
sdl 1.2.15.1809242322
sdl_image 1.2.12.1809242322
sip 4.19.8.1809242322
tinyxml 2.6.2.1809242322
tinyxml2 6.1.0.1809242322
urdfdom 1.0.0.1809242322
urdfdom_headers 1.0.0.1809242322
vcpython27 9.0.30729.1809090720
yaml-cpp 0.5.3.1809242322
39 packages installed.
Besides, 
The error only tell me can't find Qt plugin, no other error.
screenshot is attached:
[image]

seanyen-msft, 2018-12-22:
It's very likely that your environment is missing QT_PLUGIN_PATH. Can you try to run the following?

```
:: first, run the setup.bat
c:\opt\ros\melodic\x64\setup.bat

:: manually set QT_PLUGIN_PATH
set QT_PLUGIN_PATH=c:\opt\ros\melodic\x64\plugins

:: run rviz
rosrun rviz rviz -l
```

If it works, then I guess there is some problem in `C:\opt\ros\melodic\x64\etc\catkin\profile.d\1.ros_misc.bat`, particularly the output of `where qt5core.dll`. Can you try to run it manually to see any interesting values there?

traversaro, 2018-12-22:
We had a similar problem when trying to run RViz  on machine in which another Qt installation was already present in the User Path. In that case, I typically suggest to open the RViz binary with https://github.com/lucasg/Dependencies from the ROS command line  and verify which Qt dll is actually loaded by the program.

bbw2008927, 2018-12-23:
Hi
Rviz can open in my side.
Notice! the Qt_pulgin_path is C:\opt\rosdeps\x64\plugins.
thanks

bbw2008927, 2019-01-02:
Hi 
After I update full version by choco update ros-melodic-desktop_full -y
I get can't find plugin again.
I also try to lucasg/dependencies to analysis reason. 
But I still have no idea.
[image]

seanyen-msft, 2019-02-19:
Closed it because `set QT_PLUGIN_PATH=c:\opt\rosdeps\x64\plugins` seems to work for this case.

closed 2019-02-19 by seanyen-msft

WestbrookZero, 2020-12-19:
when i set QT_PLUGIN_PATH=c:\opt\ros\melodic\x64\plugins  QT_QPA_PLATFORM_PLUGIN_PATH=C:\opt\ros\melodic\x64\plugins\platforms, i still got the error:
qt.qpa.plugin: Could not load the Qt platform plugin "windows" in "C:\opt\ros\melodic\x64\plugins\platforms" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: direct2d (from C:\opt\ros\melodic\x64\plugins\platforms), minimal (from C:\opt\ros\melodic\x64\plugins\platforms), offscreen (from C:\opt\ros\melodic\x64\plugins\platforms), windows (from C:\opt\ros\melodic\x64\plugins\platforms), direct2d, minimal, offscreen, windows.