ms-iot/ROSOnWindows #45: rViz died immediately after starting

amitbiswas26, 2018-12-22:
Hi, 
I was trying out the TurtleBot3 project, however when I try to launch the simulation(**roslaunch turtlebot3_fake turtlebot3_fake.launch**), rViz seems to be crashing immediately after starting:

...
ROS_MASTER_URI=http://localhost:11311
process[turtlebot3_fake_node-1]: started with pid [3768]
process[robot_state_publisher-2]: started with pid [3964]
process[rviz-3]: started with pid [7880]
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[31m[rviz-3] process has died [pid 7880, exit code -1073741819, cmd C:\opt\ros\melodic\x64\lib\rviz\rviz.exe -d C:\workspaces\turtlebot3\install\share\turtlebot3_fake\rviz\turtlebot3_fake.rviz __name:=rviz __log:=C:\Users\Amit\.ros\log\b3e15940-05f4-11e9-96b5-000d3a58c4a0\rviz-3.log].
log file: C:\Users\Amit\.ros\log\b3e15940-05f4-11e9-96b5-000d3a58c4a0\rviz-3*.log[0m

Any idea what might be the reason?

[image]

amitbiswas26, 2018-12-22:
Here is the [Crash Dump](https://resmppaidisk.blob.core.windows.net/ros/rviz.exe.5784.dmp)

And yes, it repros with **roscore** already running and then running **rosrun rviz rviz -l**

<details>
<summary>Click to expand!</summary>

```
C:\>rosrun rviz rviz -l
[ INFO] [1545508283.541700100]: rviz version 1.13.1
[ INFO] [1545508283.541853900]: compiled against Qt version 5.10.1
[ INFO] [1545508283.542180000]: compiled against OGRE version 1.10.11 (Xalafu)
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[ INFO] [1545508284.194594700]: Creating resource group General
[ INFO] [1545508284.196111400]: Creating resource group Internal
[ INFO] [1545508284.196671700]: Creating resource group Autodetect
[ INFO] [1545508284.196997500]: SceneManagerFactory for type 'DefaultSceneManager' registered.
[ INFO] [1545508284.197440500]: Registering ResourceManager for type Material
[ INFO] [1545508284.197743300]: Registering ResourceManager for type Mesh
[ INFO] [1545508284.198038700]: Registering ResourceManager for type Skeleton
[ INFO] [1545508284.198324900]: MovableObjectFactory for type 'ParticleSystem' registered.
[ INFO] [1545508284.198645100]: ArchiveFactory for archive type FileSystem registered.
[ INFO] [1545508284.198933500]: ArchiveFactory for archive type Zip registered.
[ INFO] [1545508284.199299900]: ArchiveFactory for archive type EmbeddedZip registered.
[ INFO] [1545508284.199592500]: DDS codec registering
[ INFO] [1545508284.199889600]: ETC codec registering
[ INFO] [1545508284.200174100]: stb_image - v2.15 - public domain JPEG/PNG reader
[ INFO] [1545508284.201664200]: Supported formats: jpeg,jpg,png,bmp,psd,tga,gif,pic,ppm,pgm,hdr
[ INFO] [1545508284.201944800]: Registering ResourceManager for type HighLevelGpuProgram
[ INFO] [1545508284.202201000]: Registering ResourceManager for type Compositor
[ INFO] [1545508284.202551100]: MovableObjectFactory for type 'Entity' registered.
[ INFO] [1545508284.203017300]: MovableObjectFactory for type 'Light' registered.
[ INFO] [1545508284.203364100]: MovableObjectFactory for type 'BillboardSet' registered.
[ INFO] [1545508284.203671100]: MovableObjectFactory for type 'ManualObject' registered.
[ INFO] [1545508284.203913100]: MovableObjectFactory for type 'BillboardChain' registered.
[ INFO] [1545508284.204151000]: MovableObjectFactory for type 'RibbonTrail' registered.
[ INFO] [1545508284.204619600]: *-*-* OGRE Initialising
[ INFO] [1545508284.204882100]: *-*-* Version 1.10.11 (Xalafu)
[ INFO] [1545508284.205146400]: OverlayElementFactory for type Panel registered.
[ INFO] [1545508284.205404100]: OverlayElementFactory for type BorderPanel registered.
[ INFO] [1545508284.205641400]: OverlayElementFactory for type TextArea registered.
[ INFO] [1545508284.205878600]: Registering ResourceManager for type Font
[ INFO] [1545508284.206108700]: Loading library RenderSystem_GL
[ INFO] [1545508284.208183800]: Installing plugin: GL RenderSystem
[ INFO] [1545508284.208527300]: OpenGL Rendering Subsystem created.
[ INFO] [1545508284.210513900]: Plugin successfully installed
[ INFO] [1545508284.210999500]: Loading library Plugin_OctreeSceneManager
[ INFO] [1545508284.211991700]: Installing plugin: Octree Scene Manager
[ INFO] [1545508284.212269800]: Plugin successfully installed
[ INFO] [1545508284.212551200]: Loading library Plugin_ParticleFX
[ INFO] [1545508284.213540800]: Installing plugin: ParticleFX
[ INFO] [1545508284.213858000]: Particle Emitter Type 'Point' registered
[ INFO] [1545508284.214105800]: Particle Emitter Type 'Box' registered
[ INFO] [1545508284.214338700]: Particle Emitter Type 'Ellipsoid' registered
[ INFO] [1545508284.214638900]: Particle Emitter Type 'Cylinder' registered
[ INFO] [1545508284.214872700]: Particle Emitter Type 'Ring' registered
[ INFO] [1545508284.215109100]: Particle Emitter Type 'HollowEllipsoid' registered
[ INFO] [1545508284.215339300]: Particle Affector Type 'LinearForce' registered
[ INFO] [1545508284.215599700]: Particle Affector Type 'ColourFader' registered
[ INFO] [1545508284.215866300]: Particle Affector Type 'ColourFader2' registered
[ INFO] [1545508284.216093800]: Particle Affector Type 'ColourImage' registered
[ INFO] [1545508284.216524700]: Particle Affector Type 'ColourInterpolator' registered
[ INFO] [1545508284.216778500]: Particle Affector Type 'Scaler' registered
[ INFO] [1545508284.217019100]: Particle Affector Type 'Rotator' registered
[ INFO] [1545508284.217714400]: Particle Affector Type 'DirectionRandomiser' registered
[ INFO] [1545508284.220682600]: Particle Affector Type 'DeflectorPlane' registered
[ INFO] [1545508284.220951400]: Plugin successfully installed
[ INFO] [1545508284.221179200]: CPU Identifier & Features
[ INFO] [1545508284.221393700]: -------------------------
[... 6 lines cut ...]
[ INFO] [1545508284.222733800]:  *          MMX: yes
[ INFO] [1545508284.222938200]:  *       MMXEXT: yes
[ INFO] [1545508284.223112800]:  *        3DNOW: no
[ INFO] [1545508284.223287500]:  *     3DNOWEXT: no
[ INFO] [1545508284.223500500]:  *         CMOV: yes
[ INFO] [1545508284.223677400]:  *          TSC: yes
[ INFO] [1545508284.223852800]:  *INVARIANT TSC: no
[ INFO] [1545508284.224027500]:  *          FPU: yes
[ INFO] [1545508284.224201300]:  *          PRO: yes
[ INFO] [1545508284.224374600]:  *           HT: no
[ INFO] [1545508284.224588800]: -------------------------
[ INFO] [1545508284.224770600]: *** Starting Win32GL Subsystem ***
[... cut ...]
[ INFO] [1545508284.234745800]: Created Win32Window 'OgreWindow(0)' : 19x48, 32bpp
[ INFO] [1545508284.245013100]: GL_VERSION = 1.1.0
[ INFO] [1545508284.245198800]: GL_VENDOR = Microsoft Corporation
[ INFO] [1545508284.245894000]: GL_RENDERER = GDI Generic
[ INFO] [1545508284.246987800]: GL_EXTENSIONS = GL_WIN_swap_hint GL_EXT_bgra GL_EXT_paletted_texture
[ INFO] [1545508284.248163700]: ***************************
[ INFO] [1545508284.249423100]: *** GL Renderer Started ***
[ INFO] [1545508284.250361500]: ***************************

C:\>```
</details>

seanyen-msft, 2018-12-23:
@amitbiswas26 it seems the crash happens when RViz calls createrenderwindow from OGRE. One more thing I noticed is that "GL_RENDERER = GDI Generic". Did you run it on a VM or over remote desktop? If yes, did you have the same issue if you are running it on a physical machine?

amitbiswas26, 2018-12-24:
@seanyen-msft, Yes, I am accessing through remote desktop and the machine is an Azure VM. I haven't yet tried on a local physical machine, will let you know once I set it up.
So is this an OGRE issue with remote desktop? If so, are there any workarounds?

ooeygui, 2019-01-03:
Thanks for the report. We'll look into it right away.

ooeygui, 2019-05-17:
Renaming as this is a cloud rendering issue

c650, 2019-07-22:
Bump. I believe I am experiencing this issue in a HyperV Windows VM right now.

YoushaaMurhij, 2020-03-21:
This issue is not fixed, yet!

ooeygui, 2020-03-23:
@YoushaaMurhij rViz and Gazebo require a GPU instance when running in the cloud. Can you confirm that you've used an NV6, NV12 or an NV24 instance as described in the [ROS Azure VM Doc](https://ms-iot.github.io/ROSOnWindows/ROSAtMS/ROS1AzureVM.html)

ooeygui, 2020-07-02:
We are working with OpenGL on Dx12 which will enable GPUless running.

closed 2020-07-02 by ooeygui

yujinkimmm, 2022-02-17:
@YoushaaMurhij @c650 @amitbiswas26 The [OpenGL and OpenCL compatibility pack ](https://www.microsoft.com/en-us/p/opencl-and-opengl-compatibility-pack/9nqpsl29bfff?SilentAuth=1&wa=wsignin1.0&activetab=pivot:overviewtab)is now available for download. Please let us know if this solves your problems and if you have further questions!