ms-iot/ROSOnWindows #48: Problem compiling Gazebo plugin that uses the gazebo rendering Camera header

traversaro, 2019-01-02:
**Describe the bug**
Compilation of a Gazebo plugin that uses the header `<gazebo/rendering/Camera.hh>` is not working. 

**To Reproduce**
After executing `C:\opt\ros\melodic\x64\setup.bat` and `C:\opt\rosdeps\x64\share\gazebo-9\setup.bat`, try to compile in Release a simple Gazebo plugin that includes `<gazebo/rendering/Camera.hh>` (based on http://gazebosim.org/tutorials/?tut=plugins_hello_world) https://gist.github.com/traversaro/8b47312cb53a1228aa59ead900522c17 : 
~~~
git clone https://gist.github.com/traversaro/8b47312cb53a1228aa59ead900522c17 test_gazebo_rendering_camera
cd test_gazebo_rendering_camera
md build
cd build 
cmake -A x64 ..
cmake --build . --config Release
~~~
The last command results in  the following error: 
~~~
"C:\src\gazebo_plugin_tutorial\build\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\src\gazebo_plugin_tutorial\build\hello_world.vcxproj" (default target) (3) ->
(ClCompile target) ->
  C:\opt\rosdeps\x64\include\OGRE/Terrain/OgreTerrainPaging.h(33): fatal error C1083: Cannot open include file: 'OgrePagedWorldSection.h': No such file or directory [C:\src\gazebo_plugin_tutorial\build\hello_world.vcxproj]

    829 Warning(s)
    1 Error(s)

Time Elapsed 00:00:07.22
~~~

Uncommenting the following lines in the CMakeLists.txt: 
~~~
# find_package(OGRE REQUIRED CONFIG)
# target_include_directories(hello_world PUBLIC ${OGRE_Paging_INCLUDE_DIRS})
~~~
the compilation works as expected. For some reason, this workaround is not necessary on Linux and macOS. Note that `gazebo_pkg_ros` has a similar workaround, but based on `pkg-config`: https://github.com/ros-simulation/gazebo_ros_pkgs/blob/melodic-devel/gazebo_plugins/CMakeLists.txt#L59 .

**Expected behavior**
I would expect the plugin to compile correctly without workarounds, as it is the case on Linux and macOS . 

**Environment information:**
 - OS version: `winver`
Version 1709 (OS Build 16299.192) 
 - Chocolatey Versions: `choco list --local-only`
~~~
C:\src\gazebo_plugin_tutorial\build>choco list --local-only
Chocolatey v0.10.11
assimp 4.0.1.1809242322
boost 1.66.0.1809242326
bullet3 2.87.0.1809242326
bzip2 1.0.6.1809242322
cairo 1.15.12.1810221716
ceres 1.14.0.1812111905
chocolatey 0.10.11
clapack 3.2.1.1811082319
console_bridge 0.4.0.1809242322
cppunit 1.12.1.1809242322
cppzmq 4.2.2.1811162207
dlfcn-win32 1.1.1.1811162226
eigen 3.3.4.1809242322
freeglut 3.0.0.1809242322
freeimage 3.17.0.1811160037
gazebo9 9.4.1.1901022119
gflags 2.2.1.1811082321
glog 0.3.5.1811082321
google-mock 1.8.0.1809242326
google-test 1.8.0.1809242322
gtk2 2.22.1.1809242322
hg 4.6.1
ilmbase 2.2.1.1811210800
jasper 2.0.14.1811210919
jxrlib 1.1.0.1811210809
lcms 2.8.0.1811210903
libcurl 7.58.0.1809242322
libflann 1.9.1.1809090720
libgraphviz 2.41.0.1810022039
libjpeg-turbo 1.5.3.1809242322
liblzma 5.2.4.1811210837
libogg 1.3.3.1809090720
libopencv 3.4.1.1809242322
libpcl 1.8.1.1809090720
libpng 1.6.35.1809242322
libraw 0.19.0.1811210804
libtbb-dev 2018.6.0.1812011750
libtheora 1.1.1.1809090720
libwebp 0.6.1.1811210731
log4cxx 0.10.0.1809242322
lua 5.3.4.1810221716
lz4 1.8.1.1809242322
metis 5.1.0.1811082350
ogre 1.10.11.1809100020
openblas 0.2.20.1811082312
openexr 2.2.1.1811210832
openjpeg 2.3.0.1811210736
openssl 1.1.1.1809242322
pkg-config 0.29.2.1809242322
poco 1.8.1.1809242322
protobuf 3.6.1.1811130545
pyqt5 5.10.1.1809242322
pyside2 5.10.1.1809242322
qt5-sdk 5.10.1.1810112239
qwt 6.1.3.1811212304
ros-catkin-tools 0.0.1.1811160029
ros-cmake 3.11.4.1809242322
ros-melodic-cartographer_ros 20190102.1
ros-melodic-desktop 20190102.1
ros-melodic-desktop_full 20190102.1
ros-python2 2.7.15.1809242354
sbcl 0.0.0.1809242326
sdl 1.2.15.1809242322
sdl_image 1.2.12.1809242322
sip 4.19.8.1809242322
suitesparse 1.4.0.1811082321
tiff 4.0.9.1811210742
tinyxml 2.6.2.1809242322
tinyxml2 6.1.0.1809242322
urdfdom 1.0.0.1809242322
urdfdom_headers 1.0.0.1809242322
vcpython27 9.0.30729.1809090720
yaml-cpp 0.5.3.1809242322
zeromq 4.2.5.1811162203
zlib 1.2.11.1809242322
75 packages installed.

C:\src\gazebo_plugin_tutorial\build>
~~~

seanyen, 2019-03-24:
@traversaro Sorry for the delay on this.

It seems to me that [`gazebo-config.cmake`](https://bitbucket.org/osrf/gazebo/src/cfb74cf0e6bd1979f68da00480e5e2e77aff00be/cmake/gazebo-config.cmake.in#lines-175) is quite sensitive to what CMake files in current search path. For example, depending on `FindOgre.cmake` or `OgreConfig.cmake` is used, `OGRE_xxx_INCLUDE_DIRS` or `OGRE-xxx_INCLUDE_DIRS` is defined. However, `gazebo-config.cmake` only deals with `OGRE_xxx_INCLUDE_DIRS`.

For now, I will accommodate this by updating `gazebo-config.cmake` in our `gazebo9` chocolatey package like this: 

```cmake
# Find OGRE
find_package(OGRE REQUIRED COMPONENTS Terrain Paging)
list(APPEND GAZEBO_INCLUDE_DIRS ${OGRE_INCLUDE_DIRS})
list(APPEND GAZEBO_LIBRARIES ${OGRE_LIBRARIES})
# When including OGRE, also include the Terrain and Paging components
list(APPEND GAZEBO_INCLUDE_DIRS
  ${OGRE_Terrain_INCLUDE_DIRS}
  ${OGRE_Paging_INCLUDE_DIRS}
  ${OGRE-Terrain_INCLUDE_DIRS}
  ${OGRE-Paging_INCLUDE_DIRS})
list(APPEND GAZEBO_LIBRARIES
  ${OGRE_Terrain_LIBRARIES}
  ${OGRE_Paging_LIBRARIES}
  ${OGRE-Terrain_LIBRARIES}
  ${OGRE-Paging_LIBRARIES})
```

Let me know if it fixes your issue. Meanwhile, I will need to open a new discussion with Gazebo team on this.

seanyen, 2019-03-25:
Adding conversation link in Gazebo side. https://bitbucket.org/osrf/gazebo/pull-requests/3090/