ms-iot/ROSOnWindows #49: RVIZ can't display point cloud from PointCloud2 message

bbw2008927, 2019-01-03:
**Describe the bug**
When I  publish Point Cloud2 message,  I find out rviz can get the message information but can't display.
This is rviz screen shot: 
[image]
This is the error in the cmd windows:
[image]

**To Reproduce**
Generate random point cloud, I just generate 100 random point.

* Catkin workspace setup instructions
  * Clone locations
  * build command line
  * Launch command line

**Expected behavior**
A clear and concise description of what you expected to happen.

**Environment information:**
 - OS version: Win10 1809
 - Chocolatey Versions: 
Chocolatey v0.10.11
assimp 4.0.1.1809242322
boost 1.66.0.1809242326
bullet3 2.87.0.1809242326
bzip2 1.0.6.1809242322
chocolatey 0.10.11
chocolatey-core.extension 1.3.3
console_bridge 0.4.0.1809242322
cppunit 1.12.1.1809242322
cppzmq 4.2.2.1811162207
dlfcn-win32 1.1.1.1811162226
eigen 3.3.4.1809242322
freeimage 3.17.0.1811160037
gazebo9 9.4.1.1812181926
git 2.20.1
git.install 2.20.1
google-mock 1.8.0.1809242326
google-test 1.8.0.1809242322
gtk2 2.22.1.1809242322
hg 4.6.1
ilmbase 2.2.1.1811210800
jasper 2.0.14.1811210919
jxrlib 1.1.0.1811210809
lcms 2.8.0.1811210903
libcurl 7.58.0.1809242322
libflann 1.9.1.1809090720
libgraphviz 2.41.0.1810022039
libjpeg-turbo 1.5.3.1809242322
liblzma 5.2.4.1811210837
libogg 1.3.3.1809090720
libopencv 3.4.1.1809242322
libpng 1.6.35.1809242322
libraw 0.19.0.1811210804
libtbb-dev 2018.6.0.1812011750
libtheora 1.1.1.1809090720
libwebp 0.6.1.1811210731
log4cxx 0.10.0.1809242322
lz4 1.8.1.1809242322
ogre 1.10.11.1809100020
openexr 2.2.1.1811210832
openjpeg 2.3.0.1811210736
openssl 1.1.1.1809242322
pkg-config 0.29.2.1809242322
poco 1.8.1.1809242322
protobuf 3.6.1.1811130545
pyqt5 5.10.1.1809242322
pyside2 5.10.1.1809242322
qt5-sdk 5.10.1.1810112239
qwt 6.1.3.1811212304
ros-catkin-tools 0.0.1.1811160029
ros-cmake 3.11.4.1809242322
ros-melodic-desktop 20190102.1
ros-melodic-desktop_full 20190101.1
ros-python2 2.7.15.1809242354
sbcl 0.0.0.1809242326
sdl 1.2.15.1809242322
sdl_image 1.2.12.1809242322
sip 4.19.8.1809242322
tiff 4.0.9.1811210742
tinyxml 2.6.2.1809242322
tinyxml2 6.1.0.1809242322
urdfdom 1.0.0.1809242322
urdfdom_headers 1.0.0.1809242322
vcpython27 9.0.30729.1809090720
yaml-cpp 0.5.3.1809242322
zeromq 4.2.5.1811162203
zlib 1.2.11.1809242322
66 packages installed.

Thanks

bbw2008927, 2019-01-03:
And this is RVIZ dependency information, I think it is OK.
[image]

ooeygui, 2019-01-03:
Hi @bbw2008927,
Let me see if I can help you. Could you post the whole screenshot of rViz? I suspect that it is a driver issue. 

Are you running rViz on native hardware or through a VM?

bbw2008927, 2019-01-04:
Thanks for your answer.
There is no VM. And these are the screenshot of rviz and cmd win.
The GPU I use is GTX1050ti. NVIDIA driver version is 399.24.
[image]
[image]

bbw2008927, 2019-01-12:
Maybe it is really from driver issue. I change a new computer, the rviz can display point cloud.

ooeygui, 2019-01-25:
@bbw2008927 I'm sorry for the delay in looking at this. What video card were you using when it didn't render?

bbw2008927, 2019-02-01:
Hi
These are config of my laptop:
GPU: Intel UHD Graphics 630
         NVIDIA GTX1050Ti
Notice: The dual-GPU is based on MUXless not MUXed.
This laptop can run Ubuntu 18.04 with NVIDIA GPU driver.
But it can't run Ubuntu 16.04 with NVIDIA GPU driver.
The new PC I change, the GPU is only GTX1070Ti, There is no intel HD graphics card.
I really want to run ROS on my laptop, So I can solve problem on-site, not go back office room.
Recently, there is something wrong with office 365. I miss the reminder e-mail.
I'm sorry about it.
Look forward your response

bbw2008927, 2019-03-10:
Hi, ooeygui
Thanks for your help. problem solved.
You have to force standalone video card to render rviz. don't use intel HD 630 to render.
This is my config in nvidia panel:
[image]
This is result in rviz:
[image]
Please close this task.

seanyen-msft, 2019-03-10:
@bbw2008927 Good to hear that it is finally resolved! Close it per request.

closed 2019-03-10 by seanyen-msft

HarelAshwal, 2020-06-21:
Hi Guys,

I'm having the exact same issue on my laptop.
This is my configuration: **Lenovo Thinkpad T470s**

Name	Intel(R) HD Graphics 620
PNP Device ID	PCI\VEN_8086&DEV_5916&SUBSYS_224B17AA&REV_02\3&11583659&0&10
Adapter Type	Intel(R) HD Graphics Family, Intel Corporation compatible
Adapter Description	Intel(R) HD Graphics 620
Adapter RAM	1.00 GB (1,073,741,824 bytes)
Installed Drivers	C:\WINDOWS\System32\DriverStore\FileRepository\igdlh64.inf_amd64_a2d2bce6b74074ef\igdumdim64.dll,C:\WINDOWS\System32\DriverStore\FileRepository\igdlh64.inf_amd64_a2d2bce6b74074ef\igd10iumd64.dll,C:\WINDOWS\System32\DriverStore\FileRepository\igdlh64.inf_amd64_a2d2bce6b74074ef\igd10iumd64.dll,C:\WINDOWS\System32\DriverStore\FileRepository\igdlh64.inf_amd64_a2d2bce6b74074ef\igd12umd64.dll
Driver Version	26.20.100.7812
INF File	oem22.inf (iKBLD_w10_DS section)
Color Planes	Not Available
Color Table Entries	4294967296
Resolution	1920 x 1080 x 60 hertz
Bits/Pixel	32
Memory Address	0xEB000000-0xEBFFFFFF
Memory Address	0xA0000000-0xAFFFFFFF
I/O Port	0x0000FEC0-0x0000FEFF
IRQ Channel	IRQ 4294967287
Driver	C:\WINDOWS\SYSTEM32\DRIVERSTORE\FILEREPOSITORY\IGDLH64.INF_AMD64_A2D2BCE6B74074EF\IGDKMD64.SYS (26.20.100.7812, 23.38 MB (24,518,472 bytes), 19/06/2020 13:43)

What do I need to configure to make it works?

bbw2008927, 2020-06-22:
Hi HarelAshwal,
My solution is that don't use intel HD graphics card to run RVIZ. If Intel GPU is only choice. I have no cue about it.

ooeygui, 2020-06-22:
Hello @HarelAshwal ,
Thank you for using ROS on Windows. I'm sorry you are experiencing problems. rViz and Gazebo use the Ogre open source game engine. Because of the plugin model, Ogre needs to be configured to use OpenGL. OpenGL Drivers vary in capability on Windows, and Ogre/rViz require specific features in order to render point clouds (and lidar points) efficiently. 

We are working with the Windows Graphics team to test [OpenGL on DirectX 12](https://devblogs.microsoft.com/directx/in-the-works-opencl-and-opengl-mapping-layers-to-directx/). We will post on Social Media when this is made available.

sunny127, 2020-11-02:
I managed to get around the problem even on my laptop with no GeForce dedicated card, only an Intel card.
1) Use ROS2
2) Use RVIZ2
3) In RVIZ2, change "PointCloud2" "Style" to "Points".
4) Make sure you see messages coming at RVIZ2-> "PointCloud2" ->"Status"->"Topic" ( "Messages received" should be incremented ).
5) Using your mouse, click and hold right-click and drag the cursor to the bottom so the platform will be very small and far. Then you should see point cloud.

yujinkimmm, 2022-02-17:
@HarelAshwal The [OpenCL and OpenGL compatibility pack](https://www.microsoft.com/en-us/p/opencl-and-opengl-compatibility-pack/9nqpsl29bfff?SilentAuth=1&wa=wsignin1.0&activetab=pivot:overviewtab) is available for download now. With DirectX, supported apps will also run with hardware acceleration. Let us know if you have any questions.