ms-iot/ROSOnWindows #50: Robot_state_publisher dies on an invalid URDF

ooeygui, 2019-01-06:
**Describe the bug**
robot_state_publisher does not handle the parse error from urdf model parsing, instead crashes without an error description.

**To Reproduce**
Create a URDF with an invalid link; such as two roots.
```
<?xml version="1.0" ?>

<robot name="sample" xmlns:xacro="http://ros.org/wiki/xacro">
    <link name="camera_frame"/>
    <link nmae="foo" />
</robot>
```

Then use urdf_tutorial to run this:
roslaunch urdf_tutorial display.launch model:=sample_ws\urdf\workcell.xacro

**Expected behavior**
robot_state_publisher should output an error instead of crashing

gavanderhoorn, 2019-01-07:
This may actually be an upstream issue: https://github.com/ros/robot_state_publisher/issues/81.

ooeygui, 2019-05-17:
Closing as an upstream issue

closed 2019-05-17 by ooeygui