ms-iot/ROSOnWindows #51: ros-catkin-tools install problem.

ColmP, 2019-01-10:
Hello,

I am new to ROSOnWindows and have run into a consistent problem with installing ros-catkin-tool.
I get the following error:
ERROR: The running command stopped because the preference variable “ErrorActionPreference” or common parameter is set to Stop: Exception: The upgrade of ros-catkin-tools was NOT successful. Error while running ‘C:\ProgramData\chocolatey\lib\ros-catkin-tools\tools\chocolateyinstall.ps1’. See log for details.
This is mentioned on the troubleshooting page but consistently fails for us, it is not transient. We have tried this on two different Windows 10 systems with the same results. We are running Windows 10 Enterprise, Version 1607, Build 14393.2670

I am wondering if anyone has some recommendations on how to get this working.

Best Regards,
Colm

ooeygui, 2019-01-10:
@ColmP I'm sorry you are hitting an error, lets see if we can get you unblocked. This is the first report of this kind; I'd love your help root causing.

What locale are you in?
Can you visit https://roswin.azurewebsites.net in a web browser and see if the page displays?
Could you attach the chocolaty log file itself?

ColmP, 2019-01-10:
Thank you. 
I am able to see the web page. I am located in Boston.
Attached is the chocolaty log file.
[chocolatey.log](https://github.com/ms-iot/ROSOnWindows/files/2746567/chocolatey.log)
At this point we have tried 3 systems. 1 worked, 2 failed with the same error.
I am guessing that there is some dependency that I am missing.
Best Regards,
Colm

ooeygui, 2019-01-10:
From the log file:
Error [Error 2] The system cannot find the file specified while executing command git clone -q https://github.com/ros-infrastructure/rospkg c:\users\cprende\appdata\local\temp\chocolatey\pip-install-mr1kjd\rospkg

Can you check to see if the command `git` works in that command window?
If git is missing, you can install it with `choco install git -y`.

If it is there and runs, I'd like to see if the temp folder is there - maybe it is getting blocked by a malware scanner?

ColmP, 2019-01-10:
The git command does not work in the command window, sorry that I missed that. 
Git is installed but must be missing from the path. One the install of ROSOnWindows that worked for us, git runs in the command window.
I added the git program folder to my path ( I had upgraded from a previous git-bash install ) and was able to install ros-catkin-tools. I am currently re-installing the full ros-kinetic-desktop.
I will confirm whether this works or not.
Best Regards,
Colm