ms-iot/ROSOnWindows #53: rosrun does not work with camera calibration

ooeygui, 2019-01-15:
Working through the camera calibration tutorial, rosrun cannot find the installed calibration python script.
http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration

Using the command
`rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.0245 image:=/cv_camera/image_raw camera:=/cv_camera`

I receive the error:
`[rosrun] Couldn't find executable named cameracalibrator.py`

rospack shows 
`camera_calibration C:/opt/ros/melodic/x64\share\camera_calibration`
but this location only has a package.xml.

I suspect that rosrun should also look in the lib\site-packages

closed 2019-10-24 by ooeygui

dejongyeong, 2020-04-13:
Hi. I would like to know is that solved? I'm facing the same issue as well

ooeygui, 2020-04-13:
This bug was that rosrun could not find the python file. We have fixed that problem. (Verified today)

Can you share more information? What do you see in the console?

dejongyeong, 2020-04-13:
@ooeygui When I try to run the command, it gave me this:
[image]

and in the folder, there's no python file.
[image]

Thanks.

ooeygui, 2020-04-13:
That looks like WSL, not Native ROS on Windows. You cannot run UI applications like calibration in WSL. check out http://aka.ms/ros for running ros on windows.