ms-iot/ROSOnWindows #55: turtlebot3_slam: Cannot locate rosdep definition for [gmapping]

sw6y15, 2019-01-16:
**Describe the bug**
I've installed ros1 in win10 success by following [get started](https://ms-iot.github.io/ROSOnWindows/GettingStarted/Setup.html). When I install [Turtlebot3](https://ms-iot.github.io/ROSOnWindows/Turtlebot/Turtlebot3.html), I got an error:
[image]

**To Reproduce**
Steps to reproduce the behavior:
* Create a new workspace
`c:\opt\ros\melodic\x64\setup.bat`
`rosdep update`
`rosdep install --from-paths src --ignore-src -r -y`

**Expected behavior**
A clear and concise description of what you expected to happen.

**Environment information:**
 - OS version: Win 10 v1809
 - Chocolatey Versions: `choco list --local-only`
Chocolatey v0.10.11
assimp 4.0.1.1809242322
boost 1.66.0.1809242326
bullet3 2.87.0.1809242326
bzip2 1.0.6.1809242322
chocolatey 0.10.11
chocolatey-core.extension 1.3.3
chocolatey-windowsupdate.extension 1.0.3
console_bridge 0.4.0.1809242322
cppunit 1.12.1.1809242322
cppzmq 4.2.2.1811162207
dlfcn-win32 1.1.1.1811162226
eigen 3.3.4.1809242322
freeimage 3.17.0.1811160037
gazebo9 9.4.1.1901022119
git 2.20.1
git.install 2.20.1
google-mock 1.8.0.1809242326
google-test 1.8.0.1809242322
gtk2 2.22.1.1809242322
hg 4.6.1
ilmbase 2.2.1.1811210800
jasper 2.0.14.1811210919
jxrlib 1.1.0.1811210809
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.4
KB3035131 1.0.2
lcms 2.8.0.1811210903
libcurl 7.58.0.1809242322
libflann 1.9.1.1809090720
libgraphviz 2.41.0.1810022039
libjpeg-turbo 1.5.3.1809242322
liblzma 5.2.4.1811210837
libogg 1.3.3.1809090720
libopencv 3.4.1.1809242322
libpcl 1.8.1.1809090720
libpng 1.6.35.1809242322
libraw 0.19.0.1811210804
libtbb-dev 2018.6.0.1812011750
libtheora 1.1.1.1809090720
libwebp 0.6.1.1811210731
log4cxx 0.10.0.1809242322
lz4 1.8.1.1809242322
ogre 1.10.11.1809100020
openexr 2.2.1.1811210832
openjpeg 2.3.0.1811210736
openssl 1.1.1.1809242322
pkg-config 0.29.2.1809242322
poco 1.8.1.1809242322
protobuf 3.6.1.1811130545
pyqt5 5.10.1.1809242322
pyside2 5.10.1.1809242322
qt5-sdk 5.10.1.1810112239
qwt 6.1.3.1811212304
ros-catkin-tools 0.0.1.1901130741
ros-cmake 3.11.4.1809242322
ros-melodic-desktop_full 20190115.1
ros-python2 2.7.15.1809242354
sbcl 0.0.0.1809242326
sdl 1.2.15.1809242322
sdl_image 1.2.12.1809242322
sip 4.19.8.1809242322
tiff 4.0.9.1811210742
tinyxml 2.6.2.1809242322
tinyxml2 6.1.0.1809242322
urdfdom 1.0.0.1809242322
urdfdom_headers 1.0.0.1809242322
vcpython27 9.0.30729.1809090720
vcredist140 14.16.27024.1
yaml-cpp 0.5.3.1809242322
zeromq 4.2.5.1811162203
zlib 1.2.11.1809242322
73 packages installed.

seanyen-msft, 2019-01-16:
@sw6y15
Thanks for reporting! Are you looking for SLAM stack for turtlebot3? As of today, [Google Cartographer](https://github.com/googlecartographer/cartographer) is one known to work with ROS1 on Windows (You're welcome to make contribution to OpenSLAM gmapping Windows port too!). To learn more, take a look the SLAM example on this [page](https://ms-iot.github.io/ROSOnWindows/Turtlebot/Turtlebot3.html).

sw6y15, 2019-01-17:
Ok, I will look at cartographer. Thank you very much.

closed 2019-01-17 by sw6y15