ms-iot/ROSOnWindows #60: Error while building: Could NOT find gazebo_ros

amitbiswas26, 2019-01-19:
Hi, I am getting this error while trying to build (_catkin_make install -DCMAKE_BUILD_TYPE=RelWithDebInfo_):
This is on a fresh machine, just finished installing ROS Win, and was trying to run the Turtle bot simulation.

```
-- Could NOT find gazebo_ros (missing: gazebo_ros_DIR)
-- Could not find the required component 'gazebo_ros'. 
```

[image]

seanyen-msft, 2019-01-19:
@amitbiswas26 Did you `choco upgrade ros-melodic-desktop_full -y`? And checking `choco list -lo` and `rospack list` would help identify the missing modules.

amitbiswas26, 2019-01-19:
@seanyen-msft Thanks, `choco upgrade ros-melodic-desktop_full -y` seems to have fixed it. Should we add it in the instructions [here ](https://ms-iot.github.io/ROSOnWindows/Turtlebot/Turtlebot3.html)before the Build Nodes section?

seanyen-msft, 2019-01-20:
@amitbiswas26 I added few lines in Build Notes. Hope that helps!

amitbiswas26, 2019-01-20:
@seanyen-msft Thanks

seanyen-msft, 2019-01-23:
Sounds like it's resolved. Closed it.

closed 2019-01-23 by seanyen-msft