ms-iot/ROSOnWindows #61: rViz does not run in the devel environment due to plugins not being found.

ooeygui, 2019-01-20:
When running rViz in a devel environment, rViz cannot load plugins, however it does work in the install environment. I suspect this is a general issue with plugins in development.

Repro:
Checkout rViz into a catkin workspace.
catkin_make
devel\setup.bat

Result:
[ERROR] [1547964534.724646000]: PluginlibFactory: The plugin for class 'rviz/Orbit' failed to load.  Error: Could not find library corresponding to plugin rviz/Orbit. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.

Expect:
rViz to run without error.

ooeygui, 2019-01-28:
https://github.com/ms-iot/pluginlib/pull/7

seanyen-msft, 2019-02-20:
The change is merged. Considered it close.

closed 2019-02-20 by seanyen-msft