ms-iot/ROSOnWindows #64: [MoveIt] Can't import moveit_commander (no _moveit_roscpp_initializer)

giovent, 2019-01-25:
Hi, MoveIt works correctly, but I'd like to use the python interface and I can't import moveit_commander (from python 2.7). Is it possible to use moveit_commander on ROS on Windows?

```
>>> import moveit_commander
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\opt\ros\melodic\x64\lib\site-packages\moveit_commander\__init__.py", line 2, in <module>
    from roscpp_initializer import *
  File "C:\opt\ros\melodic\x64\lib\site-packages\moveit_commander\roscpp_initializer.py", line 35, in <module>
    from moveit_ros_planning_interface import _moveit_roscpp_initializer
ImportError: cannot import name _moveit_roscpp_initializer
```

ooeygui, 2019-01-25:
Thank you for the report.

seanyen-msft, 2019-01-28:
@giovent Can you try out the latest ros-melodic-moveit (`choco upgrade ros-melodic-moveit -y`)? I did some fix-up on some Python module install path, and now the moveit_commander should work with Python script.

giovent, 2019-01-28:
The latest version (20190127.1) fixed it, I can import moveit_commander correctly, many thanks!

seanyen-msft, 2019-01-28:
@giovent Sounds good. I closed this issue. Thanks!

closed 2019-01-28 by seanyen-msft