ms-iot/ROSOnWindows #66: RViz does not display Temperature

lrapetti, 2019-01-31:
**Describe the bug**
When I try to add a `TemperatureDisplay`, I am getting the following error:
```
GLHardwareVertexBuffer::lock - Error: failed to Discard the buffer
```
[image]

I see a similar error was described in #49, where it seem to be related to driver (my graphic processor is: Intel HD Graphics 630 ).

seanyen-msft, 2019-01-31:
@lrapetti Thanks for reporting. Have you tried to update the graphic driver to the latest to see if the issue goes away? https://downloadcenter.intel.com/product/98909/Intel-HD-Graphics-630

lrapetti, 2019-01-31:
@seanyen-msft From the device manager it seems I'm on the latest version. 
If I try to manually download and install the latest version of the driver from the link you sent me (`25.20.100.6519`), the installation procedure fails with the following error message: `The driver being installed is not validated for this computer. Please obtain the appropriate driver from the computer manufacturer`.

Some further information on the computer:
- OS: `Microsoft Windows 10 Pro 10.0.17134 Build 17134`
- Model: `Dell Precision 5520`

traversaro, 2019-06-14:
@seanyen I see tha the issue still has the `more-information-needed`, feel free to let us know (I work in the same lab of @lrapetti) any additional information you think it could make sense to provide, thanks a lot!

seanyen, 2019-06-14:
@traversaro Thanks for the reminder!

@lrapetti, do you have a chance to update your machine drivers? https://www.dell.com/support/home/ca/en/cabsdt1/product-support/product/precision-15-5520-laptop/drivers

Also, is the issue about RViz not able to display temperature or "GLHardwareVertexBuffer::lock - Error: failed to Discard the buffer" problem? Sharing `rviz` configuration file might be also helpful.

lrapetti, 2019-06-26:
> Also, is the issue about RViz not able to display temperature or `GLHardwareVertexBuffer::lock - Error: failed to Discard the buffer` problem?

The error I see is `GLHardwareVertexBuffer::lock - Error: failed to Discard the buffer`, and at the same time the temperature does not shown.

> Sharing `rviz` configuration file might be also helpful.

The `rviz` configuration can be found at https://github.com/robotology/human-dynamics-estimation/blob/master/conf/ros/rviz/HDERviz.rviz, and it has been tested on a different machine (non Windows) and the temperature showed up correctly as in the sphere that you can see in the following screenshot:
[image]

lrapetti, 2019-06-26:
> @lrapetti, do you have a chance to update your machine drivers? https://www.dell.com/support/home/ca/en/cabsdt1/product-support/product/precision-15-5520-laptop/drivers

I have also tried to update the machine drivers, but nothing changed.

seanyen, 2020-05-13:
@lrapetti @traversaro  Sorry for super long overdue on this issue. I think I am able to find a environment to repro this issue, and would like to check if my observation matches to what you saw.

So currently the ROS on Windows is using `OGRE 1.10.11` as the rendering backend, and in my repro case (I am using `Intel UHD 630`), the visualization of temperature only works when the styles are "Point", "Square" and "Sphere".

Does that match to what the original problems reported here?

And just for your information, after I switched to use OGRE 1.12.6 and did a private RViz build. It seems that the temperature display can work for all styles. Atm, I think there is that the newer version of OGRE comes with some improvement which can help this case.

lrapetti, 2020-05-14:
> the visualization of temperature only works when the styles are "Point", "Square" and "Sphere".
> 
> Does that match to what the original problems reported here?

we are actually trying to visualize the temperature on spheres, like in the picture in https://github.com/ms-iot/ROSOnWindows/issues/66#issuecomment-505860273 (obtained on a different machine)

seanyen, 2020-09-22:
Since the behavior depends on an external library, I am closing it. For anyone hitting this issue, you can use Noetic release which is using the OGRE 1.12 as the backend, and which comes with the more recent bug fixes.

closed 2020-09-22 by seanyen