ms-iot/ROSOnWindows #70: Failed to loat gazebo_ros_controll.dll

OnesoulKang, 2019-02-05:
When I launch my launch file, error 'Failed to load gazbo_ros_control.dll' is occur.

There are 'gazebo_ros_control.dll' & 'gazebo_ros_control.lib' in "C:\opt\ros\melodic\x64\lib"

I confirmed that the package is in "C:/opt/ros/melodic/x64\share\gazebo_ros_contro"l using the command 'rospack find gazebo_ros_control'.

How can I fix this error?

Here is my launch file.
<?xml version="1.0" encoding="UTF-8"?>
<launch>
    
        <include file="$(find gazebo_ros)/launch/empty_world.launch">
            <arg name="verbose" value="true"/>
        </include>

    <group ns="/mrm">
        
        <arg name="model" default="$(find mrm_description)/urdf/mrm.xacro"/>
        <arg name="x" default="0"/>
        <arg name="y" default="0"/>
        <arg name="z" default="0.5"/>

        <param name="robot_description" command="$(find xacro)/xacro $(arg model)" />

        
        <node name="mybot_spawn" pkg="gazebo_ros" type="spawn_model" output="screen"
            args="-urdf -param robot_description -model mrm -x $(arg x) -y $(arg y) -z $(arg z)"/>

        
        <rosparam command="load" file="$(find mrm_description)/config/joints.yaml"/>

        
        <node name="controller_spawner" pkg="controller_manager" type="spawner"
            respawn="false" output="screen" ns="/mrm"
            args="--namespace=/mrm
            joint_state_controller
            joint1_position_cotroller
            joint2_position_cotroller
            --timeout 60">
        </node>
    </group>
</launch>

and this is my xacro file
<?xml version="1.0" ?>

<robot name="mrm" xmlns:xacro="http://www.ros.org/wiki/xacro">
    
    <xacro:include filename="$(find mrm_description)/urdf/links_joints.xacro" />
    <xacro:include filename="$(find mrm_description)/urdf/robot_parameters.xacro" />
    

    
    <m_link_box name="${l00}"
                origin_rpy="0 0 0" origin_xyz="0 0 0"
                mass ="1024"
                ixx="170.667" ixy="0.0" ixz="0.0"
                iyy="170.667" iyz="0.0"
                izz="170.667"
                size="1 1 1" />

    <m_joint name="${l00}__${l01}" type="revolute"
             axis_xyz="0 0 1"
             origin_rpy="0 0 0" origin_xyz="0 0 0.5"
             parent="${l00}" child="${l01}" />
    
    <transmission name="trans_${l00}_${l01}">
        <type>transmission_interface/SimpleTransmission</type>
        <joint name="${l00}__${l01}">
            <hardwareInterface>hardware_interface/EffortJointInterface</hardwareInterface>
        </joint>
        <actuator name="motor_${l00}_${l01}">
            <hardwareInterface>hardware_interface/EffortJointInterface</hardwareInterface>
            <mechanicalReduction>1</mechanicalReduction>
        </actuator>
    </transmission>

    <m_link_cylinder name="${l01}"
                origin_xyz="0 0 0.2" origin_rpy="0 0 0"
                mass ="157.633"
                ixx="13.235" ixy="0" ixz="0"
                iyy="13.235" iyz="0"
                izz="9.655"
                radius="0.35" length="0.4" />

    <m_joint name="${l01}_${l02}" type="revolute"
             axis_xyz="0 1 0"
             origin_rpy="0 0 0" origin_xyz="0 0 0.4"
             parent="${l01}" child="${l02}" />

    <transmission name="trans_${l01}_${l02}">
        <type>transmission_interface/SimpleTransmission</type>
        <joint name="${l01}__${l02}">
            <hardwareInterface>hardware_interface/EffortJointInterface</hardwareInterface>
        </joint>
        <actuator name="motor_${l01}_${l02}">
            <hardwareInterface>hardware_interface/EffortJointInterface</hardwareInterface>
            <mechanicalReduction>1</mechanicalReduction>
        </actuator>
    </transmission>

    <m_link_cylinder name="${l02}"
                origin_xyz="0 0 0.4" origin_rpy="0 0 0"
                mass ="57.906"
                ixx="12.679" ixy="0" ixz="0"
                iyy="12.679" iyz="0"
                izz="0.651"
                radius="0.15" length="0.8" />
    

    <gazebo>
        <plugin name="gazebo_ros_control" filename="libgazebo_ros_control.so"/>
    </gazebo>
</robot>

ooeygui, 2019-02-06:
Hi @OnesoulKang,
I'd like to reproduce this locally. Can you post the commands to setup a workspace? (including clone locations)

@seanyen-msft I don't recall - do we automatically remap Linux filenames to Windows? (I see a `libgazebo_ros_control.so` in the Xacro)

seanyen-msft, 2019-02-06:
@OnesoulKang one way to debug it is to use [process monitor](https://docs.microsoft.com/en-us/sysinternals/downloads/procmon) to watch where does the gazebo looks for the DLL.

@ooeygui yes, the Gazebo does the trick to remap the so to dll internally if it is Windows system. My guess would be search path issue.

OnesoulKang, 2019-02-07:
@ooeygui 
I don't know much about this field, so I think it'd be nice if you could give me some more details on what I should tell you.

I usually use this commands to set my working environment.
c:\opt\ros\melodic\x64\setup.bat
c:\catkin_ws\devel\setup.bat

seanyen-msft, 2019-02-07:
@OnesoulKang On a second look at your launch file, can you add something like below: You can see an example [here](https://github.com/ms-iot/turtlebot3_simulations/blob/melodic-devel/turtlebot3_gazebo/launch/turtlebot3_gazebo_cartographer_demo.launch):

```
  <env name="GAZEBO_PLUGIN_PATH" value="$(find gazebo_plugins)"/>
```

Gazebo honors `%GAZEBO_PLUGIN_PATH%` to search the plugin DLLs.

OnesoulKang, 2019-02-08:
@seanyen-msft 

Thank you for your answer. 
but after I add the command, the same error still occurs.

The path I have identified by "rospack find gazebo_plugins" is "C:/opt/ros/melodic/x64\share\gazebo_plugins".

There are several files in the folder.

[image]

seanyen-msft, 2019-02-08:
@OnesoulKang hmm, can you modify my example and add `C:\opt\ros\melodic\x64\lib` to GAZEBO_PLUGIN_PATH:

```xml
<env name="GAZEBO_PLUGIN_PATH" value="C:\opt\ros\melodic\x64\lib"/>
```

In general, plugin DLL paths need to be added in GAZEBO_PLUGIN_PATH.

ooeygui, 2019-02-09:
@OnesoulKang No worries. In order for me to reproduce it locally, I need to setup a catkin workspace just like yours. For example, can you send me the git setup instructions?

For example, in our UR3 docs, we have this:
```
c:\opt\ros\melodic\x64\setup.bat
mkdir c:\catkin_ws\src
cd c:\catkin_ws\src
git clone https://github.com/ms-iot/universal_robot -b init_windows
git clone https://github.com/ms-iot/ur_modern_driver -b init_windows
echo > universal_robot\ur_gazebo\CATKIN_IGNORE
cd c:\catkin_ws
catkin_make
c:\catkin_ws\devel\setup.bat
```
It would be great to get something similar for your setup.

OnesoulKang, 2019-02-11:
@seanyen-msft 
I've tried the way you told me, but the same error still happens.
There is no DLL file (gazbo_ros_control.dll) in the folder. So I copy&pasted the file from another computer(with Ubuntu 14.04, ROS Indigo). But the file still cannot be found.

Error Message
`[Plugin.hh:188] Failed to load plugin C:\opt\rosdeps\x64\lib\gazebo-9\plugins\gazebo_ros_control.dll: "C:\opt\rosdeps\x64\lib\gazebo-9\plugins\gazebo_ros_control.dll"` 

@ooeygui 
`c:\opt\ros\melodic\x64\setup.bat`
`mkdir c:\catkin_ws\src`
`cd c:\catkin_ws\src`
and I downloaded a project from bitbucket.
https://bitbucket.org/theconstructcore/my-robotic-manipulator
`catkin_make`
`c:\catkin_ws\devel\setup.bat`
`roslaunch mrm_description spawn.launch`

seanyen-msft, 2019-02-11:
@OnesoulKang 
Thank for the information what repo you are using! Look like you are following the tutorials [here](http://www.theconstructsim.com/ros-projects-my-robotic-manipulator-part-4-ros-urdf-transmission-gazebo-controllers/)?

After looking this demo, it seems to depend on [`simulation_gazebo`](https://bitbucket.org/theconstructcore/simulation_gazebo), so I am assuming you are running both:
`roslaunch simulation_gazebo main.launch`
`roslaunch mrm_description spawn.launch`

And there are indeed Windows-specific bugs in `ros_control` related package, where I discovered some DLLs consumed by `gazebo_ros_control.dll` are not yet placed to the correct location. I made a few fixes and the next daily build should have it fixed. Can you try to do `choco upgrade ros-melodic-desktop_full -y` later? 
(I will ping this thread again once the new build is released.)

seanyen-msft, 2019-02-11:
@OnesoulKang The new build is out, please give it a try! (Run `choco upgrade ros-melodic-desktop_full -y`)

OnesoulKang, 2019-02-12:
@seanyen-msft 
The error is gone! I really appreciate your help!

I think now we can close this issue!

Thanks!

closed 2019-02-12 by OnesoulKang