ms-iot/ROSOnWindows #73: rostopic seems to be broken

kejxu, 2019-02-15:
When I run any rostopic command, I get...
```python
Traceback (most recent call last):
  File "C:\opt\ros\melodic\x64\bin\rostopic", line 35, in <module>
    rostopic.rostopicmain()
  File "C:\opt\ros\melodic\x64\lib\site-packages\rostopic\__init__.py", line 2101, in rostopicmain
    import rosbag
  File "C:\opt\ros\melodic\x64\lib\site-packages\rosbag\__init__.py", line 33, in <module>
    from .bag import Bag, Compression, ROSBagException, ROSBagFormatException, ROSBagUnindexedException
  File "C:\opt\ros\melodic\x64\lib\site-packages\rosbag\bag.py", line 55, in <module>
    import gnupg
  File "C:\opt\python27amd64\lib\site-packages\gnupg\__init__.py", line 23, in <module>
    from . import gnupg
  File "C:\opt\python27amd64\lib\site-packages\gnupg\gnupg.py", line 42, in <module>
    from ._meta    import GPGBase
  File "C:\opt\python27amd64\lib\site-packages\gnupg\_meta.py", line 134, in <module>
    class GPGBase(object):
  File "C:\opt\python27amd64\lib\site-packages\gnupg\_meta.py", line 72, in __new__
    if cls._find_agent():
  File "C:\opt\python27amd64\lib\site-packages\gnupg\_meta.py", line 115, in _find_agent
    if (proc.name() == "gpg-agent") and proc.is_running():
  File "C:\opt\python27amd64\lib\site-packages\psutil\__init__.py", line 609, in name
    name = self._proc.name()
  File "C:\opt\python27amd64\lib\site-packages\psutil\_pswindows.py", line 635, in wrapper
    return fun(self, *args, **kwargs)
  File "C:\opt\python27amd64\lib\site-packages\psutil\_pswindows.py", line 687, in name
    return py2_strencode(os.path.basename(self.exe()))
  File "C:\opt\python27amd64\lib\site-packages\psutil\_pswindows.py", line 635, in wrapper
    return fun(self, *args, **kwargs)
  File "C:\opt\python27amd64\lib\site-packages\psutil\_pswindows.py", line 701, in exe
    return py2_strencode(convert_dos_path(cext.proc_exe(self.pid)))
WindowsError: [Error 0] The operation completed successfully
```

kejxu, 2019-02-15:
this issue seems to be triggered by `psutil`'s [recent upgrade to 5.5.0](https://pypi.org/project/psutil/#history) on 2019/01/23. Have not further investigated why this error is triggered, but rolling back to the previous version (5.4.8) would fix this.

If you are blocked by this issue, do the following:
1. check the version of `psutil` installed by your (python 2.x) pip
```
>pip show psutil
Name: psutil
Version: 5.5.0
Summary: Cross-platform lib for process and system monitoring in Python.
Home-page: https://github.com/giampaolo/psutil
Author: Giampaolo Rodola
Author-email: g.rodola@gmail.com
License: BSD
Location: c:\opt\python27amd64\lib\site-packages
Requires:
Required-by: gnupg
```
2. if it shows `Version: 5.5.0`, do the following to roll back to 5.4.8
```
>pip install psutil==5.4.8 --force-reinstall
Collecting psutil==5.4.8
  Using cached https://files.pythonhosted.org/packages/0f/fb/6aecd2c8c9d0ac83d789eaf9f9ec052dd61dd5aea2b47ffa4704175d7a2a/psutil-5.4.8-cp27-none-win_amd64.whl
Installing collected packages: psutil
  Found existing installation: psutil 5.5.0
    Uninstalling psutil-5.5.0:
      Successfully uninstalled psutil-5.5.0
Successfully installed psutil-5.4.8
```
3. check `psutil` again
```
>pip show psutil
Name: psutil
Version: 5.4.8
Summary: Cross-platform lib for process and system monitoring in Python.
Home-page: https://github.com/giampaolo/psutil
Author: Giampaolo Rodola
Author-email: g.rodola@gmail.com
License: BSD
Location: c:\opt\python27amd64\lib\site-packages
Requires:
Required-by: gnupg
```

**tl;dr**, roll back `psutil` version with `pip install psutil==5.4.8 --force-reinstall`

kejxu, 2019-02-15:
opened https://github.com/giampaolo/psutil/issues/1410

seanyen-msft, 2019-02-15:
I made some changes in https://github.com/ms-iot/rosdistro-db/tree/init_windows to version-lock the `psutil` to 5.4.8.

kejxu, 2019-02-15:
`psutil` recently upgraded to `5.5.1` and `rostopic` works fine with the newer version, the version lock for `5.4.8` has been removed in https://github.com/ms-iot/rosdistro-db/commit/189aaff3e10566e03d7de915286225bcf5960a20

closed 2019-02-15 by kejxu