ms-iot/ROSOnWindows #77: [roscore]Problem when running roscore  error: ? [31m[rosout-1] process has died [pid 2716, exit code -1073741515...

WangZP10969, 2019-02-20:
**Describe the bug**
A loop error message appears when running roscore:
[roslaunch][INFO] 2019-02-20 10:19:37,971: [rosout-1] restarting process
[roslaunch][INFO] 2019-02-20 10:19:37,971: process[rosout-1]: restarting os process
[roslaunch][INFO] 2019-02-20 10:19:37,973: process[rosout-1]: start w/ args [[u'c:\\opt\\ros\\melodic\\x64\\lib\\rosout\\rosout.exe', u'__name:=rosout', u'__log:=C:\\Users\\XASY-PC\\.ros\\log\\e53261c0-34b5-11e9-9a06-10f005138886\\rosout-1.log']]
[roslaunch][INFO] 2019-02-20 10:19:37,973: process[rosout-1]: cwd will be [C:\Users\XASY-PC\.ros]
[roslaunch][INFO] 2019-02-20 10:19:37,974: process[rosout-1]: started with pid [10964]
[roslaunch][ERROR] 2019-02-20 10:19:38,177: [rosout-1] process has died [pid 10964, exit code -1073741515, cmd c:\opt\ros\melodic\x64\lib\rosout\rosout.exe __name:=rosout __log:=C:\Users\XASY-PC\.ros\log\e53261c0-34b5-11e9-9a06-10f005138886\rosout-1.log].
log file: C:\Users\XASY-PC\.ros\log\e53261c0-34b5-11e9-9a06-10f005138886\rosout-1*.log

**To Reproduce**
1.
I use vs code as the terminal and then install the vs2017 Community Edition.
I use the following parameters to run the terminal, and run vs code with admin.
"terminal.integrated.shellArgs.windows": "C:\\Windows\\System32\\cmd.exe /k 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\Common7\\Tools\\VsDevCmd.bat' -arch=amd64 -host_arch=amd64",

2.
The running prompt is as follows, It seems to be normal:
**********************************************************************
** Visual Studio 2017 Developer Command Prompt v15.9.7
** Copyright (c) 2017 Microsoft Corporation
**********************************************************************

C:\Users\XASY-PC>

3.
Next I installed choco, the prompt is normal
Finally I installed ros, using the following instructions:
> choco upgrade ros-melodic-desktop -y
The running prompt is as follows：
ERROR: Cannot find python runtime under c:\opt\python27amd64
The install of ros-catkin-tools was NOT successful.
Error while running 'C:\ProgramData\chocolatey\lib\ros-catkin-tools\tools\chocolateyinstall.ps1'.
 See log for details.

4.
I think it may be that I don't have python installed, so I installed python with the following instructions.
> choco install python2 --params '"/InstallDir:C:\opt\python27amd64"'
There is a warning to run, but the installation is successful
WARNING: Environment variable PYTHONHOME points to different version: C:\opt\python27amd64\

5.
Next I ran the following command, but gave me a wrong prompt
> c:\opt\ros\melodic\x64\setup.bat
Traceback (most recent call last):
  File "C:\opt\ros\melodic\x64\bin\roscore", line 36, in <module>
    from rosmaster.master_api import NUM_WORKERS
  File "C:\opt\ros\melodic\x64\lib\site-packages\rosmaster\__init__.py", line 35, in <module>
    from .main import rosmaster_main
  File "C:\opt\ros\melodic\x64\lib\site-packages\rosmaster\main.py", line 43, in <module>
    import rosmaster.master
  File "C:\opt\ros\melodic\x64\lib\site-packages\rosmaster\master.py", line 47, in <module>
    import rosmaster.master_api
  File "C:\opt\ros\melodic\x64\lib\site-packages\rosmaster\master_api.py", line 72, in <module>
    from rosmaster.util import xmlrpcapi
  File "C:\opt\ros\melodic\x64\lib\site-packages\rosmaster\util.py", line 48, in <module>
    from defusedxml.xmlrpc import monkey_patch
ImportError: No module named defusedxml.xmlrpc

I searched for this problem and then ran the following command
Pip install defusedxml

6.
After running bat again, the prompt is as follows:
Traceback (most recent call last):
  File "C:\opt\ros\melodic\x64\bin\roscore", line 72, in <module>
    import roslaunch
  File "C:\opt\ros\melodic\x64\lib\site-packages\roslaunch\__init__.py", line 39, in <module>
    import rospkg
ImportError: No module named rospkg

pip install rospkg
Then I can run bat successfully.

7.
After I ran roscore, the above loop prompt appeared.
[roslaunch][ERROR] 2019-02-20 10:00:11,440: [rosout-1] process has died [pid 6544, exit code -1073741515, cmd c:\opt\ros\melodic\x64\lib\rosout\rosout.exe __name:=rosout __log:=C:\Users\XASY-PC\.ros\log\2e344a80-34b3-11e9-a423-10f005138886\rosout-1.log].
log file: C:\Users\XASY-PC\.ros\log\2e344a80-34b3-11e9-a423-10f005138886\rosout-1*.log
[roslaunch][INFO] 2019-02-20 10:00:11,454: [rosout-1] restarting process
[roslaunch][INFO] 2019-02-20 10:00:11,460: process[rosout-1]: restarting os process
[roslaunch][INFO] 2019-02-20 10:00:11,461: process[rosout-1]: start w/ args [[u'c:\\opt\\ros\\melodic\\x64\\lib\\rosout\\rosout.exe', u'__name:=rosout', u'__log:=C:\\Users\\XASY-PC\\.ros\\log\\2e344a80-34b3-11e9-a423-10f005138886\\rosout-1.log']]
[roslaunch][INFO] 2019-02-20 10:00:11,461: process[rosout-1]: cwd will be [C:\Users\XASY-PC\.ros]
[roslaunch][INFO] 2019-02-20 10:00:11,464: process[rosout-1]: started with pid [9064]

**Expected behavior**
The record in master.log is as follows
[rosmaster.main][INFO] 2019-02-20 10:19:22,015: initialization complete, waiting for shutdown
[rosmaster.main][INFO] 2019-02-20 10:19:22,016: Starting ROS Master Node
[xmlrpc][INFO] 2019-02-20 10:19:22,016: XML-RPC server binding to 0.0.0.0:11311
[rosmaster.master][INFO] 2019-02-20 10:19:22,019: Master initialized: port[11311], uri[http://WangZP-PC:11311/]
[xmlrpc][INFO] 2019-02-20 10:19:22,019: Started XML-RPC server [http://WangZP-PC:11311/]
[xmlrpc][INFO] 2019-02-20 10:19:22,020: xml rpc node: starting XML-RPC server
[rosmaster.master][INFO] 2019-02-20 10:19:32,769: +PARAM [/run_id] by /roslaunch
[rosmaster.master][INFO] 2019-02-20 10:19:32,769: +PARAM [/roslaunch/uris/host_wangzp_pc__10612] by /roslaunch
[xmlrpc][ERROR] 2019-02-20 10:19:37,135: Traceback (most recent call last):
  File "C:\opt\python27amd64\lib\SocketServer.py", line 596, in process_request_thread
    self.finish_request(request, client_address)
  File "C:\opt\python27amd64\lib\SocketServer.py", line 331, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "C:\opt\python27amd64\lib\SocketServer.py", line 652, in __init__
    self.handle()
  File "C:\opt\python27amd64\lib\BaseHTTPServer.py", line 342, in handle
    self.handle_one_request()
  File "C:\opt\python27amd64\lib\BaseHTTPServer.py", line 310, in handle_one_request
    self.raw_requestline = self.rfile.readline(65537)
  File "C:\opt\python27amd64\lib\socket.py", line 480, in readline
    data = self._sock.recv(self._rbufsize)
error: [Errno 10054] 

**Environment information:**
 - OS version: 1803 OS 17134.590  
 - Chocolatey Versions: 
Chocolatey v0.10.11
asio 1.12.1
assimp 4.0.1.1809242322
boost 1.66.0.1809242326
bullet3 2.87.0.1809242326
bzip2 1.0.6.1809242322
chocolatey 0.10.11
chocolatey-core.extension 1.3.3
chocolatey-windowsupdate.extension 1.0.2
cmake 3.13.4
cmake.install 3.13.4
console_bridge 0.4.0.1809242322
cppunit 1.12.1.1809242322
eigen 3.3.4
git 2.20.1
git.install 2.20.1
google-mock 1.8.0.1809242326
google-test 1.8.0.1809242322
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20170509
KB3033929 1.0.3
KB3035131 1.0.1
libcurl 7.58.0.1809242322
libgraphviz 2.41.0.1809191808
libopencv 3.4.1.1809242322
log4cxx 0.10.0.1809242322
lz4 1.8.1.1809242322
ogre 1.10.11.1809100020
openssl 1.1.1.1809242322
pkg-config 0.29.2.1809242322
poco 1.8.1.1809242322
pyqt5 5.10.1.1809242322
pyside2 5.10.1.1809242322
python2 2.7.15
python3 3.7.2
qt5-sdk 5.10.1.1809242322
ros-cmake 3.11.4.1809242322
ros-melodic-desktop 20190219.1
ros-python2 2.7.15.1809242354
sbcl 0.0.0.1809242326
sdl 1.2.15.1809242322
sdl_image 1.2.12.1809242322
sip 4.19.8.1809242322
tinyxml 2.6.2.1809242322
tinyxml-usestl 2.6.2
tinyxml2 6.0.0
urdfdom 1.0.0.1809242322
urdfdom_headers 1.0.0.1809242322
vcpython27 9.0.30729.1809090720
vcredist140 14.16.27024.1
yaml-cpp 0.5.3.1809242322
51 packages installed.

The above is the process of my problems, I look forward to your help, thank you

seanyen-msft, 2019-02-20:
@WangZP10969 Thank for reporting this! It smells like the installation doesn't get alone under vscode command prompt. We will take a look on this combination.

Also, I noticed `ros-catkin-tools` is not installed successfully, can you do `choco upgrade ros-catkin-tools -y` directly to see if it resolves the problem?

WangZP10969, 2019-02-20:
> @WangZP10969 Thank for reporting this! It smells like the installation doesn't get alone under vscode command prompt. We will take a look on this combination.
>
> Also, I noticed `ros-catkin-tools` is [...]

1.
I tried using the command choco upgrade ros-catkin-tools -y
The tips given are as follows:
Chocolatey v0.10.11
Upgrading the following packages:
ros-catkin-tools
By upgrading you accept licenses for the packages.
ros-catkin-tools is not installed. Installing...
Progress: Downloading ros-catkin-tools 0.0.1.1901231908... 100%
Progress: Downloading ros-catkin-tools 0.0.1.1901231908... 100%

ros-catkin-tools v0.0.1.1901231908
ros-catkin-tools package files upgrade completed. Performing other installation steps.
Downloading ROS for Windows patch requirements file...
Check and upgrade pip if needed...
Collecting pip==18.1
  Using cached https://files.pythonhosted.org/packages/c2/d7/90f34cb0d83a6c5631cf71dfe64cc1054598c843a92b400e55675cc2ac37/pip-18.1-py2.py3-none-any.whl
Collecting setuptools
  Using cached https://files.pythonhosted.org/packages/d1/6a/4b2fcefd2ea0868810e92d519dacac1ddc64a2e53ba9e3422c3b62b378a6/setuptools-40.8.0-py2.py3-none-any.whl
Installing collected packages: pip, setuptools
  Found existing installation: pip 19.0.2
    Uninstalling pip-19.0.2:
      Successfully uninstalled pip-19.0.2
  Found existing installation: setuptools 39.0.1
    Uninstalling setuptools-39.0.1:
      Successfully uninstalled setuptools-39.0.1
Successfully installed pip-18.1 setuptools-40.8.0
Install released ROS catkin tools...
Collecting rosdep (from -r C:\ProgramData\chocolatey\lib\ros-catkin-tools\tools\requirements.txt (line 1))
  Downloading https://files.pythonhosted.org/packages/57/b9/735a9f5aee7b79863ca2fb654258e3d78ec665f9b7b9ef8d621e09244c05/rosdep-0.15.1.tar.gz (80kB)
Collecting rosinstall_generator (from -r C:\ProgramData\chocolatey\lib\ros-catkin-tools\tools\requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/75/91/66fcf3cadc67729e89745fddaa3eaf9ae49cbf65daa3403cf0db80c193fe/rosinstall_generator-0.1.14.tar.gz
Collecting wstool (from -r C:\ProgramData\chocolatey\lib\ros-catkin-tools\tools\requirements.txt (line 3))
  Using cached https://files.pythonhosted.org/packages/66/7b/a3e4195605644e54658101acc13593703bde9fb67731e4565ac225a5bdb5/wstool-0.1.17.tar.gz
Collecting rosinstall (from -r C:\ProgramData\chocolatey\lib\ros-catkin-tools\tools\requirements.txt (line 4))
  Using cached https://files.pythonhosted.org/packages/38/c5/2c466034d05ad84ca15179d1f496958c6caad9432928bf7436711cc3e64e/rosinstall-0.7.8.tar.gz
Requirement already satisfied, skipping upgrade: catkin_pkg>=0.4.0 in c:\opt\python27amd64\lib\site-packages (from rosdep->-r C:\ProgramData\chocolatey\lib\ros-catkin-tools\tools\requirements.txt (line 1)) (0.4.10)
Requirement already satisfied, skipping upgrade: rospkg>=1.0.37 in c:\opt\python27amd64\lib\site-packages (from rosdep->-r C:\ProgramData\chocolatey\lib\ros-catkin-tools\tools\requirements.txt (line 1)) (1.1.7)
Collecting rosdistro>=0.7.0 (from rosdep->-r C:\ProgramData\chocolatey\lib\ros-catkin-tools\tools\requirements.txt (line 1))
  Using cached https://files.pythonhosted.org/packages/89/73/1afdde48b6531a51776cc6a74c0f90b4bd994824ad231d28c5e5ce7d73be/rosdistro-0.7.2.tar.gz
Requirement already satisfied, skipping upgrade: PyYAML>=3.1 in c:\opt\python27amd64\lib\site-packages (from rosdep->-r C:\ProgramData\chocolatey\lib\ros-catkin-tools\tools\requirements.txt (line 1)) (3.13)
Requirement already satisfied, skipping upgrade: setuptools in c:\opt\python27amd64\lib\site-packages (from rosinstall_generator->-r C:\ProgramData\chocolatey\lib\ros-catkin-tools\tools\requirements.txt (line 2)) (40.8.0)
Collecting vcstools>=0.1.38 (from wstool->-r C:\ProgramData\chocolatey\lib\ros-catkin-tools\tools\requirements.txt (line 3))
  Using cached https://files.pythonhosted.org/packages/1e/0c/fd44a903c31a12827f31c6fb15d8ba2f3b731ad1e54f1a004c98f1fa6541/vcstools-0.1.40.tar.gz
Requirement already satisfied, skipping upgrade: docutils in c:\opt\python27amd64\lib\site-packages (from catkin_pkg>=0.4.0->rosdep->-r C:\ProgramData\chocolatey\lib\ros-catkin-tools\tools\requirements.txt (line 1)) (0.14)
Requirement already satisfied, skipping upgrade: python-dateutil in c:\opt\python27amd64\lib\site-packages (from catkin_pkg>=0.4.0->rosdep->-r C:\ProgramData\chocolatey\lib\ros-catkin-tools\tools\requirements.txt (line 1)) (2.8.0)
Requirement already satisfied, skipping upgrade: pyparsing in c:\opt\python27amd64\lib\site-packages (from catkin_pkg>=0.4.0->rosdep->-r C:\ProgramData\chocolatey\lib\ros-catkin-tools\tools\requirements.txt (line 1)) (2.3.1)
Requirement already satisfied, skipping upgrade: argparse in c:\opt\python27amd64\lib\site-packages (from catkin_pkg>=0.4.0->rosdep->-r C:\ProgramData\chocolatey\lib\ros-catkin-tools\tools\requirements.txt (line 1)) (1.4.0)
Requirement already satisfied, skipping upgrade: six>=1.5 in c:\opt\python27amd64\lib\site-packages (from python-dateutil->catkin_pkg>=0.4.0->rosdep->-r C:\ProgramData\chocolatey\lib\ros-catkin-tools\tools\requirements.txt (line 1)) (1.12.0)
Installing collected packages: rosdistro, rosdep, rosinstall-generator, vcstools, wstool, rosinstall
  Running setup.py install for rosdistro: started
    Running setup.py install for rosdistro: finished with status 'done'
  Running setup.py install for rosdep: started
    Running setup.py install for rosdep: finished with status 'done'
[... cut ...]
I used cmd to run the test, the prompt message is the same, but an error message will pop up at the same time.

The code could not be executed because boost_date_time-vc141-mt-x64-1_66.dll could not be found. Reinstalling the program may resolve this issue
Unable to continue executing code because boost_system-vc141-mt-x64-1_66.dll could not be found. Reinstalling the program may resolve this issue
Unable to continue executing code because log4cxx.dll could not be found. Reinstalling the program may resolve this issue

kejxu, 2019-02-20:
> Next I installed choco, the prompt is normal
> Finally I installed ros, using the following instructions:

thanks again for reporting the issue, the problems are seemingly lack of dependency. a few things to check here:
- are you following the [setup instructions](https://github.com/ms-iot/ROSOnWindows/blob/master/GettingStarted/Setup.md) on ROSOnWindows?
- have you added `ros-win` Nuget source with `choco source add -n=ros-win -s="https://roswin.azurewebsites.net/api/v2" --priority=1`?

btw, thanks for the detailed steps! these are valuable knowledge of how to integrate ROS with vscode =)

WangZP10969, 2019-02-21:
> thanks again for reporting the issue, the problems are seemingly lack of dependency. a few things to check here:
>
> * are you following the [setup [...]

Yes, I did it according to the instructions in this document. Below I will give a record of my installation process.
I tried it a few months ago and it was a success. The document was still on another linked page. And I have not done anything else.

Process record：
```
**********************************************************************
** Visual Studio 2017 Developer Command Prompt v15.9.7
** Copyright (c) 2017 Microsoft Corporation
**********************************************************************

C:\Users\XASY-PC>@"%SystemRoot%\System32\WindowsPowerShell\v1.0\powershell.exe" -NoProfile -InputFormat None -ExecutionPolicy Bypass
-Command "iex ((New-Object System.Net.WebClient).DownloadString('https://chocolatey.org/install.ps1'))" && SET "PATH=%PATH%;%ALLUSERSPROFILE%\chocolatey\bin"
Getting latest version of the Chocolatey package for download.
Getting Chocolatey from https://chocolatey.org/api/v2/package/chocolatey/0.10.11.
Extracting C:\Users\XASY-PC\AppData\Local\Temp\chocolatey\chocInstall\chocolatey.zip to C:\Users\XASY-PC\AppData\Local\Temp\chocolatey\chocInstall...
Installing chocolatey on this machine
Creating ChocolateyInstall as an environment variable (targeting 'Machine')
  Setting ChocolateyInstall to 'C:\ProgramData\chocolatey'
WARNING: It's very likely you will need to close and reopen your shell 
  before you can use choco.
Restricting write permissions to Administrators
We are setting up the Chocolatey package repository.
The packages themselves go to 'C:\ProgramData\chocolatey\lib'
  (i.e. C:\ProgramData\chocolatey\lib\yourPackageName).
A shim file for the command line goes to 'C:\ProgramData\chocolatey\bin'
  and points to an executable in 'C:\ProgramData\chocolatey\lib\yourPackageName'.

Creating Chocolatey folders if they do not already exist.

WARNING: You can safely ignore errors related to missing log files when 
  upgrading from a version of Chocolatey less than 0.9.9. 
  'Batch file could not be found' is also safe to ignore. 
  'The system cannot find the file specified' - also safe.
警告: Not setting tab completion: Profile file does not exist at 
'C:\Users\XASY-PC\Documents\WindowsPowerShell\Microsoft.PowerShell_profile.ps1'.
Chocolatey (choco.exe) is now ready.
You can call choco from anywhere, command line or powershell by typing choco.
Run choco /? for a list of functions.
You may need to shut down and restart powershell and/or consoles
 first prior to using choco.
Ensuring chocolatey commands are on the path
Ensuring chocolatey.nupkg is in the lib folder

C:\Users\XASY-PC>choco install git -y
Chocolatey v0.10.11
Installing the following packages:
git
By installing you accept licenses for the packages.
git v2.20.1 already installed.
 Use --force to reinstall, specify a version to install, or try upgrade.

Chocolatey installed 0/1 packages. 
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).

Warnings:
 - git - git v2.20.1 already installed.
 Use --force to reinstall, specify a version to install, or try upgrade.

C:\Users\XASY-PC>choco source add -n=ros-win -s="https://roswin.azurewebsites.net/api/v2" --priority=1
Chocolatey v0.10.11
Nothing to change. Config already set.

C:\Users\XASY-PC>choco upgrade ros-melodic-desktop -y
Chocolatey v0.10.11
Upgrading the following packages:
ros-melodic-desktop
[... 250 lines cut ...]
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES

auto-starting new master
process[master]: started with pid [10068]
ROS_MASTER_URI=http://WangZP-PC:11311/
setting /run_id to 33e3f580-3589-11e9-857a-10f005138886
process[rosout-1]: started with pid [10996]
started core service [/rosout]
[31m[rosout-1] process has died [pid 10996, exit code -1073741515, cmd c:\opt\ros\melodic\x64\lib\rosout\rosout.exe __name:=rosout __log:=C:\Users\XASY-PC\.ros\log\33e3f580-3589-11e9-857a-10f005138886\rosout-1.log].
log file: C:\Users\XASY-PC\.ros\log\33e3f580-3589-11e9-857a-10f005138886\rosout-1*.log[0m
[rosout-1] restarting process
process[rosout-1]: started with pid [9012]
[31m[rosout-1] process has died [pid 9012, exit code -1073741515, cmd c:\opt\ros\melodic\x64\lib\rosout\rosout.exe __name:=rosout __log:=C:\Users\XASY-PC\.ros\log\33e3f580-3589-11e9-857a-10f005138886\rosout-1.log].
log file: C:\Users\XASY-PC\.ros\log\33e3f580-3589-11e9-857a-10f005138886\rosout-1*.log[0m
[rosout-1] restarting process
^C
C:\Windows\System32>[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

kejxu, 2019-03-04:
just want to update on this thread, we are currently focusing on [upstreaming](https://github.com/ms-iot/ROSOnWindows/projects) our `ms-iot` payload to ROS mainline, as a result, everything else have to be de-prioritized.

I will revisit and try to reproduce this issue later this week, and update on this thread afterwards =)

NevilleKitala, 2019-03-11:
I would like to report the exact same issue seems to have occurred with me as well. I have read through his logs and errors and they seem similar to mine. I am curious if there might be a solution yet.

WangZP10969, 2019-03-11:
> I would like to report the exact same issue seems to have occurred with me as well. I have read through his logs and errors and they seem similar to mine. I am curious if there might be a solution yet.

The effect of running on my computer is this, there is no solution yet. I have been busy with something else recently. After a while I will try to use other computers to test if there will be similar problems.

WangZP10969, 2019-03-14:
> I would like to report the exact same issue seems to have occurred with me as well. I have read through his logs and errors and they seem similar to mine. I am curious if there might be a solution yet.

I tried it again today and the results are consistent with the previous ones.
Throughout the installation process, I am skeptical about the steps to manually install Python. The installation instructions do not mention the way Python is installed. 
If you don't install Python, you can't execute bat, and the Python installation location in bat is also a fixed path. 
This makes me think if it need a specific way to install Python?

seanyen, 2019-03-14:
@WangZP10969

> This makes me think if it need a specific way to install Python?

In ROS on Windows, we installed a specific Python runtime (c:\opt\python27amd64) for your ROS setup. You can find the package to learn more: https://roswin.azurewebsites.net/packages/ros-python2/

> process[rosout-1]: started with pid [10996]
started core service [/rosout]
�[31m[rosout-1] process has died [pid 10996, exit code -1073741515, cmd c:\opt\ros\melodic\x64\lib\rosout\rosout.exe __name:=rosout __log:=C:\Users\XASY-PC\.ros\log\33e3f580-3589-11e9-857a-10f005138886\rosout-1.log].
log file: C:\Users\XASY-PC\.ros\log\33e3f580-3589-11e9-857a-10f005138886\rosout-1*.log�[0m
[rosout-1] restarting process
process[rosout-1]: started with pid [9012]
�[31m[rosout-1] process has died [pid 9012, exit code -1073741515, cmd c:\opt\ros\melodic\x64\lib\rosout\rosout.exe __name:=rosout __log:=C:\Users\XASY-PC\.ros\log\33e3f580-3589-11e9-857a-10f005138886\rosout-1.log].
log file: C:\Users\XASY-PC\.ros\log\33e3f580-3589-11e9-857a-10f005138886\rosout-1*.log�[0m
[rosout-1] restarting process

From your log, error `-1073741515` to hex should be `0xC0000135` and which usually means some DLL dependencies not found. And I suggest to use [Dependencies](https://github.com/lucasg/Dependencies) tool to analyze the missing modules.

WangZP10969, 2019-03-14:
> @WangZP10969
> 
> > This makes me think if it need a specific way to install Python?
> 
> In ROS on Windows, we installed a specific Python runtime (c:\opt\python27amd64) for your ROS setup. You can find the package to learn more: https://roswin.azurewebsites.net/packages/ros-python2/
> 
> > process[rosout-1]: started with pid [10996]
> > started core service [/rosout]
> > �[31m[rosout-1] process has died [pid 10996, exit code -1073741515, cmd c:\opt\ros\melodic\x64\lib\rosout\rosout.exe __name:=rosout __log:=C:\Users\XASY-PC.ros\log\33e3f580-3589-11e9-857a-10f005138886\rosout-1.log].
> > log file: C:\Users\XASY-PC.ros\log\33e3f580-3589-11e9-857a-10f005138886\rosout-1*.log�[0m
> > [rosout-1] restarting process
> > process[rosout-1]: started with pid [9012]
> > �[31m[rosout-1] process has died [pid 9012, exit code -1073741515, cmd c:\opt\ros\melodic\x64\lib\rosout\rosout.exe __name:=rosout __log:=C:\Users\XASY-PC.ros\log\33e3f580-3589-11e9-857a-10f005138886\rosout-1.log].
> > log file: C:\Users\XASY-PC.ros\log\33e3f580-3589-11e9-857a-10f005138886\rosout-1*.log�[0m
> > [rosout-1] restarting process
> 
> From your log, error `-1073741515` to hex should be `0xC0000135` and which usually means some DLL dependencies not found. And I suggest to use [Dependencies](https://github.com/lucasg/Dependencies) tool to analyze the missing modules.
> 
> Or you can try to directly run `C:\opt\ros\melodic\x64\lib\rosout\rosout.exe` to see any useful error messages after `c:\opt\melodic\x64\setup.bat` runs.

Thank you for your help.
I am trying to run rosout.exe directly, prompting the missing rosconsole.dll, roscpp.dll, roscpp_serialzation.dll, rostime.dll.
Is it because I forgot to install something?

seanyen, 2019-03-14:
@WangZP10969 Those should be under `C:\opt\ros\melodic\x64\bin` and this path should be added to `%PATH%` after `setup.bat` invokes. Did you find those DLLs (`rosconsole.dll`, `roscpp.dll`, `roscpp_serialzation.dll`, `rostime.dll`) there? They should come as `ros-melodic-xyz` installed.

FYI, here is my `%PATH%` after `setup.bat` is called: `PATH=C:/opt/ros/melodic/x64\bin;C:/opt/ros/melodic/x64\lib;C:/opt/rosdeps/x64\bin;C:/opt/rosdeps/x64\lib;C:\opt\python27amd64\;C:\opt\python27amd64\Scripts;...`

If it is not the case, I'd suggest to investigate a bit in `setup.bat` and `_setup_util.py` under `C:\opt\ros\melodic\x64`.

WangZP10969, 2019-03-14:
@seanyen 
I found those dlls in the bin directory, then the environment variables after I run bat are as follows

PATH=C:/opt/ros/melodic/x64\bin;C:/opt/ros/melodic/x64\lib;C:\opt\python27amd64\;C:\opt\python27amd64\Scripts;

seanyen, 2019-03-14:
@WangZP10969 

What if you patch `%PATH%`? Run this after `setup.bat`: (Assuming `c:\opt\rosdeps\x64` exists.)
```
set PATH=C:/opt/ros/melodic/x64\bin;C:/opt/ros/melodic/x64\lib;C:/opt/rosdeps/x64\bin;C:/opt/rosdeps/x64\lib;C:\opt\python27amd64\;C:\opt\python27amd64\Scripts;%PATH%
```

Then run `roscore` again?

WangZP10969, 2019-03-14:
@seanyen 

Actually there is no C:/opt/rosdeps in my directory.
Only /ros and /python27amd64 in the opt directory

After I set it up, run the path as follows:
PATH=C:/opt/ros/melodic/x64\bin;C:/opt/ros/melodic/x64\lib;C:/opt/rosdeps/x64\bin;C:/opt/rosdeps/x64\lib; C:\opt\python27amd64\;C:\opt\python27amd64\Scripts;C:/opt/ros/melodic/x64\bin;C:/opt/ros/melodic/x64\lib;C:\opt\python27amd64\ ;C:\opt\python27amd64\Scripts;

Running roscore should be the same before:
Process[rosout-1]: started with pid [10968]
?[31m[rosout-1] process has died [pid 10968, exit code -1073741515, cmd C:\opt\ros\melodic\x64\lib\rosout\rosout.exe __name:=rosout __log:=C:\Users \XASY-PC\.ros\log\7cc8065e-4637-11e9-a570-10f005138883\rosout-1.log].
Log file: C:\Users\XASY-PC\.ros\log\7cc8065e-4637-11e9-a570-10f005138883\rosout-1*.log?[0m
[rosout-1] restarting process

seanyen, 2019-03-14:
@WangZP10969 
I guessed something went wrong when the previous install ran `rosdep`.

To manually recover that, after `setup.bat`, can you do:
```
:: it is okay to see "ERROR: default sources list file already exists:"
rosdep init

:: make sure you see "Hit https://raw.githubusercontent.com/ms-iot/rosdistro-db/init_windows/rosdep/win-chocolatey.yaml" on the output
rosdep update

:: now we trigger rosdep install manually with force reinstall.
pushd c:\opt\ros\melodic\x64
rosdep install --from-paths . --ignore-src --reinstall -r -y
```

It is going to pull large amount of files so please be patient. After it finishes, you should find lots of files downloaded to `c:\opt\rosdeps\x64`.

WangZP10969, 2019-03-15:
@seanyen 
First of all, thank you very much for your help.

Yesterday I tried to install rosdep, but there may be some unsuccessful ones that caused the situation to not improve.

I think it may be affected by a lot of things in the current environment, so I will test the full process in a clean environment in the virtual machine.

The process is as follows:
1. First install Ros-Python. There is a problem with this. I don't know **how to install the Nuget package to opt**. I used to install normal Python and then unpack the Nuget package to copy the contents for overwriting. What is the way to install a formal process?
2. Follow https://ms-iot.github.io/ROSOnWindows/GettingStarted/Setup.html
Process installation
3. If bat or roscore cannot be executed normally, repair it by the following method
Choco upgrade ros-catkin-tools
...rosdep install --from-paths . --ignore-src --reinstall -r -y
Pip install defusedxml
Pip install rospkg
...

This test may not be completed very quickly, I am busy with some work, once I have the results, I will update it in time.

WangZP10969, 2019-03-19:
I think the reason I found the problem is because I couldn't install ros completely. After using the tutorial, the ros is installed very fast, but obviously there are a lot of things missing, at least the rosdeps folder is not installed.
The installation record is as follows, no error message is prompted
What is the reason for this happening?

```
C:\Windows\System32>choco upgrade ros-melodic-desktop -y
Chocolatey v0.10.13
2 validations performed. 1 success(es), 1 warning(s), and 0 error(s).

Validation Warnings:
 - A pending system reboot request has been detected, however, this is
   being ignored due to the current Chocolatey configuration.  If you
   want to halt when this occurs, then either set the global feature
   using:
     choco feature enable -name=exitOnRebootDetected
   or pass the option --exit-when-reboot-detected.

Upgrading the following packages:
ros-melodic-desktop
By upgrading you accept licenses for the packages.
ros-melodic-desktop is not installed. Installing...
Progress: Downloading ros-melodic-desktop 20190318.1... 100%
Progress: Downloading ros-melodic-desktop 20190318.1... 100%

ros-melodic-desktop v20190318.1
ros-melodic-desktop package files upgrade completed. Performing other installation steps.
Extracting 64-bit C:\ProgramData\chocolatey\lib\ros-melodic-desktop\tools\drop.zip to c:\opt\ros...
c:\opt\ros
running rosdep...
 The upgrade of ros-melodic-desktop was successful.
  Software installed to 'c:\opt\ros'

Chocolatey upgraded 1/1 packages.
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).
```

maxsvetlik, 2019-04-03:
I've also run into this problem today and have come up with a workaround.
I was installing ROS on Windows as according to the installation instructions but facing the same issues as @WangZP10969.

After running `choco upgrade ros-melodic-desktop_full -y` choco would download and install the package, then output something along the lines to **`starting rosdeps...`** and immediately after, output **`chocolatey could not create tempory file at FILEPATH`**. Chocolatey would then report than `ros-melodic-desktop_full` had been installed correctly. Which I guess is true, but chocolatey had not successfully initialized, run or reasoned about rosdeps 

Changing the cache location for chocolatey allowed for the temporary file to be successfully created, which in turn allowed for the rosdep to be initialized and run correctly. To be specific, here are the steps I took for recovery:

1. choco uninstall ros-melodic-desktop_full
2. choco config set cacheLocation c:\new_temp
3. choco install ros-melodic-desktop_full

(sorry for the lack of exact output; my chocolatey logs got deleted as a casualty)
I installed ROS on Windows 3 weeks ago on a different system and did not encounter this problem, so this seems to be a recent development

WangZP10969, 2019-04-04:
> I've also run into this problem today and have come up with a workaround.
> I was installing ROS on Windows as according to the installation instructions but facing the same issues as @WangZP10969. [...]

@maxsvetlik  Thank you very much for your reply, I will try your way.

WangZP10969, 2019-04-22:
@maxsvetlik I try this way to clear the all of the ros, and set choco timeout very long, its worded normals
I think the problem was the install it not complete.
Thank all of you very much!

closed 2019-04-22 by WangZP10969

kejxu, 2019-04-22:
> I think the problem was the install it not complete.
> Thank all of you very much!

Thanks for trying it out and sharing the outcome =)