ms-iot/ROSOnWindows #78: [spawn_urdf-4] process has died

TVarkey, 2019-02-22:
**Describe the bug**
log file: C:\Users\NAME\.ros\log\3523f36e-36e5-11e9-a3db-106530272d8c\gazebo_gui-3*.log
[ERROR] [1550869401.276000, 6.017000]: Spawn service failed. Exiting.
[gazebo-2] process has finished cleanly
log file: C:\Users\NAME\.ros\log\3523f36e-36e5-11e9-a3db-106530272d8c\gazebo-2*.log
[spawn_urdf-4] process has died [pid 19968, exit code 1, cmd python c:\opt\ros\melodic\x64\lib\gazebo_ros\spawn_model -urdf -model turtlebot3_waffle -x -2.0 -y -0.5 -z 0.0 -param robot_description __name:=spawn_urdf __log:=C:\Users\NAME\.ros\log\3523f36e-36e5-11e9-a3db-106530272d8c\spawn_urdf-4.log].
log file: C:\Users\NAME\.ros\log\3523f36e-36e5-11e9-a3db-106530272d8c\spawn_urdf-4*.log
===========REQUIRED process [rviz-10] has died!

**To Reproduce**
Followed the instructions in the web page below
https://ms-iot.github.io/ROSOnWindows/Turtlebot/Turtlebot3.html

In the section in the web page above "Let’s try out something more!"
roslaunch turtlebot3_gazebo turtlebot3_gazebo_cartographer_demo.launch

ADDITIONAL INFORMATION:
[rospy.internal][ERROR] 2019-02-22 14:03:21,272: Traceback (most recent call last):
  File "C:\opt\ros\melodic\x64\lib\site-packages\rospy\impl\tcpros_base.py", line 733, in receive_once
    self.stat_bytes += recv_buff(sock, b, p.buff_size)
  File "C:\opt\ros\melodic\x64\lib\site-packages\rospy\impl\tcpros_base.py", line 104, in recv_buff
    d = sock.recv(buff_size)
error: [Errno 10054] An existing connection was forcibly closed by the remote host

[rospy.internal][ERROR] 2019-02-22 14:03:21,272: Traceback (most recent call last):
  File "C:\opt\ros\melodic\x64\lib\site-packages\rospy\impl\tcpros_base.py", line 733, in receive_once
    self.stat_bytes += recv_buff(sock, b, p.buff_size)
  File "C:\opt\ros\melodic\x64\lib\site-packages\rospy\impl\tcpros_base.py", line 104, in recv_buff
    d = sock.recv(buff_size)
error: [Errno 10054] An existing connection was forcibly closed by the remote host

[rosout][ERROR] 2019-02-22 14:03:21,276: Spawn service failed. Exiting.
[rospy.core][INFO] 2019-02-22 14:03:21,276: signal_shutdown [atexit]
[rospy.internal][INFO] 2019-02-22 14:03:22,289: topic[/rosout] removing connection to /rosout
[rospy.internal][INFO] 2019-02-22 14:03:22,289: topic[/clock] removing connection to http://myhostname:51198/
[rospy.impl.masterslave][INFO] 2019-02-22 14:03:22,289: atexit
[rospy.internal][WARNING] 2019-02-22 14:03:22,292: Unknown error initiating TCP/IP socket to myhostname:51197 (http://myhostname:51198/): Traceback (most recent call last):
  File "C:\opt\ros\melodic\x64\lib\site-packages\rospy\impl\tcpros_base.py", line 557, in connect
    self.socket.connect((dest_addr, dest_port))
  File "C:\opt\python27amd64\lib\socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 10061] No connection could be made because the target machine actively refused it

**Expected behavior**
Gazebo running a simulated world with a simulated turtlebot, RViz running the mapping progress, and a simulation node to drive the turtlebot random walking.

**Environment information:**
 - OS version: `winver` Windows Version 1703 (OS Build 15063.1563) -- Windows 10 Enterprise
 - Chocolatey Versions: `choco list --local-only`  Chocolatey v0.10.11

seanyen-msft, 2019-02-22:
@TVarkey Thanks for reporting the issues! I noticed `http://myhostname:51198/` was used for the connection. Are you trying to communicate the nodes across different machines? Have you tried to play with (or turn off) the firewall to see if it helps with the error of `the target machine actively refused it`.

TVarkey, 2019-02-22:
@seanyen-msft I am just following instructions in the link provided in the "To Reproduce" and I don't believe there are any communications across machines. Since I am not communicating across machines, is there a way around without turning off the firewall? Thank you.

seanyen-msft, 2019-02-22:
@TVarkey can you try to run the following under a command prompt with admin privilege? (It is for debugging purpose only. It's not recommended to permanently switch off the firewall.)

```console
> netsh advfirewall set allprofiles state off
```

TVarkey, 2019-02-22:
@seanyen-msft I ran that command and I am still seeing the same issue
Also, want to point out that, I have not seen any of the gazebo logs such as the gazebo-2*.log, so far

log file: C:\Users\UserName\.ros\log\b2f734a1-36fa-11e9-87b2-106530272d8c\gazebo-2*.log
[spawn_urdf-4] process has died [pid 10676, exit code 1, cmd python c:\opt\ros\melodic\x64\lib\gazebo_ros\spawn_model -urdf -model turtlebot3_waffle -x -2.0 -y -0.5 -z 0.0 -param robot_description __name:=spawn_urdf __log:=C:\Users\UserName\.ros\log\b2f734a1-36fa-11e9-87b2-106530272d8c\spawn_urdf-4.log].
log file: C:\Users\UserName\.ros\log\b2f734a1-36fa-11e9-87b2-106530272d8c\spawn_urdf-4*.log
================================================================================REQUIRED process [rviz-10] has died!

ooeygui, 2019-02-23:
Could you attach master.log from C:\Users\UserName.ros\log\*runid*?

TVarkey, 2019-02-23:
[master.log](https://github.com/ms-iot/ROSOnWindows/files/2897180/master.log)

TVarkey, 2019-02-23:
@ooeygui See attached log. Sorry... Did not mean to close. Thanks for your help.

TVarkey, 2019-02-25:
@seanyen-msft Anything else required from me? Please let me know. Thanks, Thomas

TVarkey, 2019-03-06:
@seanyen-msft @ooeygui Do you need more information? If not, can you please remove the label for more-information-needed? If you need anything else from me, please let me know

ooeygui, 2019-03-27:
Sorry for the delay - I'm investigating this.

ooeygui, 2019-03-27:
Typically when we see:

```An existing connection was forcibly closed by the remote host```

It is due to the Firewall blocking access to the ROS process. Can you verify that the firewall rules allow that ROS process?

bakerhillpins, 2019-04-10:
So I'm seeing the same problems here but tied to [spawn_urdf-3]. I've attached the log files.

`[ INFO] [1554903509.360539600]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1554903509.375889100]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1554903509.652757200]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1554903509.658113500]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1554903513.987533500, 0.058000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[31m[ERROR] [1554903517.413000, 0.282000]: Spawn service failed. Exiting.
[0m[gazebo-1] process has finished cleanly
log file: C:\Users\xx\.ros\log\dc9d1e8f-5b95-11e9-949c-4439c48fd245\gazebo-1*.log
[gazebo_gui-2] process has finished cleanly
log file: C:\Users\xx\.ros\log\dc9d1e8f-5b95-11e9-949c-4439c48fd245\gazebo_gui-2*.log
[spawn_urdf-3] process has died [pid 352, exit code 1, cmd python C:\opt\ros\melodic\x64\lib\gazebo_ros\spawn_model -urdf -model turtlebot3_waffle -x -2.0 -y -0.5 -z 0.0 -param robot_description __name:=spawn_urdf __log:=C:\Users\xx\.ros\log\dc9d1e8f-5b95-11e9-949c-4439c48fd245\spawn_urdf-3.log].
log file: C:\Users\xx\.ros\log\dc9d1e8f-5b95-11e9-949c-4439c48fd245\spawn_urdf-3*.log`

[spawn_urdf-3.log](https://github.com/ms-iot/ROSOnWindows/files/3064045/spawn_urdf-3.log)
[spawn_urdf-3-stdout.log](https://github.com/ms-iot/ROSOnWindows/files/3064046/spawn_urdf-3-stdout.log)
[master.log](https://github.com/ms-iot/ROSOnWindows/files/3064047/master.log)

When I first ran the tutorial (same as listed by OP) I had to click through several layers of Firewall permissions, of which I granted all network permissions to all processes that prompted for it. It's a single node on a single PC.

I closed all the processes/node, disabled all firewalls via control panel and tried again with the same results. Let me know if you'd like me to try anything else.

bakerhillpins, 2019-04-11:
For what it's worth: I thought I'd spend some time on this. I'll come back and edit this if I find anything interesting as I go rather than a stream of comments.

I thought I'd start by trying to manually relaunch the spawn_urdf node after the error occurred so I dug through the launch file and the log file to find/validate the command line which I determined to be:
`rosrun gazebo_ros spawn_model -urdf -model turtlebot3_waffle -x -2.0 -y -.05 -z 0.0 -param robot_description __name:=spawn_urdf`

Interestingly when I run this the node spins up but complains that it can't find the /gazebo/spawn_urdf_model service:

> [INFO] [1554991461.045000, 0.000000]: Loading model XML from ros parameter robot_description
> [INFO] [1554991461.078000, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
> [33m[WARN] [1554991462.100000, 0.000000]: wait_for_service(/gazebo/spawn_urdf_model): failed to contact, will keep trying
> [0m

Odd part is that this service seems to be up and running; [Clipped output]

> C:\Windows\System32>rosservice list
> /cartographer_node/get_loggers
> /cartographer_node/set_logger_level
> /cartographer_occupancy_grid_node/get_loggers
> [snip]
> /gazebo/set_parameters
> /gazebo/set_physics_properties
> /gazebo/spawn_sdf_model
> /gazebo/spawn_urdf_model
> /gazebo/unpause_physics
> /gazebo_gui/get_loggers
> /gazebo_gui/set_logger_level
> [snip]

So I wonder if the forcibly closed error is due to the gazebo node having some sort of issue when executing the service call originating from spawn_urdf? That seems reasonable considering the log file entry since it seems to have found the service when run from the launch file situation:

> [rosout][INFO] 2019-04-10 09:38:33,831: Calling service /gazebo/spawn_urdf_model
[rospy.internal][ERROR] 2019-04-10 09:38:37,411: Traceback (most recent call last):
  File "C:\opt\ros\melodic\x64\lib\site-packages\rospy\impl\tcpros_base.py", line 733, in receive_once
    self.stat_bytes += recv_buff(sock, b, p.buff_size)
  File "C:\opt\ros\melodic\x64\lib\site-packages\rospy\impl\tcpros_base.py", line 104, in recv_buff
    d = sock.recv(buff_size)
error: [Errno 10054] An existing connection was forcibly closed by the remote host

seanyen, 2019-04-11:
@bakerhillpins Thanks for the diagnostics! `/gazebo/spawn_urdf_model` is advertised by `gazebo_ros_api_plugin` from `gazebo_ros`. Maybe you can enable `DEBUG` level messages from `gazebo_ros`?

Also since the plugin is loaded by Gazebo, logs under `c:\Users\<account name>\.gazebo\` might be worth to check as well.

bakerhillpins, 2019-04-12:
@seanyen I've not tried enabling DEBUG yet. I'll put it on my list. 

I looked in the gazebo logs and found 2 things I thought I'd mention.
1st, the log has an error looking for the autorace model.config. Seems odd since the tutorial is referencing the cartographer demo which, as I understand it, uses the Turtlebot World with the Waffle device. There is no config file in that directory, but there are in the world and waffle dirs.

> (1554903513 9405700) [Err] [..\gazebo\gui\InsertModelWidget.cc:433] Missing model.config for model "C:/Users/xx/catkin_ws/src/turtlebot3/install\share\turtlebot3_gazebo\models\turtlebot3_autorace"
(1554903513 14483700) model.config file is missing in directory["C:/Users/xx/catkin_ws/src/turtlebot3/install\share\turtlebot3_gazebo\models\turtlebot3_autorace"]

2nd, note that the first half of the directory separators are Linux style. The [RosOnWindows](https://ms-iot.github.io/ROSOnWindows/GettingStarted/UsingROSonWindows.html) page mentions an attempt to convert. Unsure if it matters, but there it is. I can resolve the directory with either separator model via windows explorer but it may have a conversion engine.

bakerhillpins, 2019-04-12:
When I manually execute the gazebo client it seems to crash and exit every time leaving an error in the Windows Application Event log. (remove the .log extension)
[gzclientError.evtx.log](https://github.com/ms-iot/ROSOnWindows/files/3074511/gzclientError.evtx.log)

Looking back through the log this error has occurred each time I've executed the client.

ooeygui, 2020-02-10:
This has been tracked for awhile; but has not manifested in several releases. Closing. 

If you do come across a new repro, please attach a stack trace of the crashing process.

YoushaaMurhij, 2020-03-21:
I am facing the exact same problem! URDF files won't show in Gazebo!

ooeygui, 2020-03-23:
@YoushaaMurhij Thank you for reporting. Can you point me to the repo which reproduces this crash?

YoushaaMurhij, 2020-03-24:
> @YoushaaMurhij Thank you for reporting. Can you point me to the repo which reproduces this crash?
It is Turtlebot3 repo.
I followed the instructions in the web page below:
https://ms-iot.github.io/ROSOnWindows/Turtlebot/Turtlebot3.html

uchuuengineer, 2020-03-27:
I noticed I have the same problem when running  
`roslaunch turtlebot3_gazebo turtlebot3_gazebo_cartographer_demo.launch`
on my work computer, running Windows 10 Enterprise. However, when I repeated the same steps on my personal PC, I did not have this error at all and it worked fine. On my work laptop, I tried to reinstall from scratch, but every time when running the launch file I get the following output:
```
NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    flat_world_imu_node (turtlebot3_slam/flat_world_imu_node)
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    move_base (move_base/move_base)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    spawn_urdf (gazebo_ros/spawn_model)
    turtlebot3_drive (turtlebot3_gazebo/turtlebot3_drive)

auto-starting new master
process[master]: started with pid [35508]
ROS_MASTER_URI=http://127.0.0.1:11311
setting /run_id to a98ddd0f-7051-11ea-99a0-00216afa3394
process[rosout-1]: started with pid [32392]
started core service [/rosout]
process[gazebo-2]: started with pid [38248]
process[gazebo_gui-3]: started with pid [43208]
process[spawn_urdf-4]: started with pid [116112]
process[robot_state_publisher-5]: started with pid [72360]
process[move_base-6]: started with pid [92348]
ERROR: cannot launch node of type [cartographer_ros/cartographer_node]: cartographer_ros
ROS path [0]=C:\opt\ros\melodic\x64\share\ros
ROS path [1]=C:/catkin_ws/src
ROS path [2]=C:\opt\ros\melodic\x64\share
ERROR: cannot launch node of type [cartographer_ros/cartographer_occupancy_grid_node]: cartographer_ros
ROS path [0]=C:\opt\ros\melodic\x64\share\ros
ROS path [1]=C:/catkin_ws/src
ROS path [2]=C:\opt\ros\melodic\x64\share
process[flat_world_imu_node-9]: started with pid [102356]
process[rviz-10]: started with pid [71272]
process[turtlebot3_drive-11]: started with pid [87908]
[ INFO] [1585330666.913992100]: TurtleBot3 Simulation Node Init
[ INFO] [1585330666.919415400]: robot_model : WAFFLE
[ INFO] [1585330666.919582500]: turning_radius_ : 0.000000
[ INFO] [1585330666.919971700]: front_distance_limit_ = 0.000000
[ INFO] [1585330666.920419800]: side_distance_limit_ = 0.000000
Traceback (most recent call last):
  File "C:\opt\ros\melodic\x64\lib\gazebo_ros\spawn_model", line 34, in <module>
    from tf.transformations import quaternion_from_euler
  File "C:\opt\ros\melodic\x64\lib\site-packages\tf\__init__.py", line 31, in <module>
    from .listener import Transformer, TransformListener, TransformerROS
  File "C:\opt\ros\melodic\x64\lib\site-packages\tf\listener.py", line 33, in <module>
    import sensor_msgs.msg
ImportError: No module named msg
[ INFO] [1585330667.224164000]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1585330667.225161200]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1585330667.227411600]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1585330667.227150500]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[spawn_urdf-4] process has died [pid 116112, exit code 1, cmd python C:\opt\ros\melodic\x64\lib\gazebo_ros\spawn_model -urdf -model turtlebot3_WAFFLE -x -2.0 -y -0.5 -z 0.0 -param robot_description __name:=spawn_urdf __log:=C:\Users\cdobrin1\.ros\log\a98ddd0f-7051-11ea-99a0-00216afa3394\spawn_urdf-4.log].
log file: C:\Users\cdobrin1\.ros\log\a98ddd0f-7051-11ea-99a0-00216afa3394\spawn_urdf-4*.log
[ INFO] [1585330669.417506800, 0.083000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1585330669.662622600, 0.324000000]: Physics dynamic reconfigure ready.
[ERROR] [1585330669.707273700, 0.368000000]: PluginlibFactory: The plugin for class 'Submaps' failed to load.  Error: According to the loaded plugin descriptions the class Submaps with base class type rviz::Display does not exist. Declared types are  rtabmap_ros/Info rtabmap_ros/MapCloud rtabmap_ros/MapGraph rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/WrenchStamped rviz_imu_plugin/Imu rviz_plugin_tutorials/Imu
[ WARN] [1585330674.470789100, 5.081000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.1 timeout was 0.1.
```

The turtlebot world appears in Gazebo, but there is no robot traveling through it like there is supposed to be. I believe I am having the same problem as the others on this issue.

seanyen, 2020-03-27:
This one looks fishy to me. Can you try `choco upgrade ros-melodic-desktop_full -y --pre` to the latest daily build and see if this problem goes away?

Recently there are some Python3 migration movement which might break some Python packages, and the fix is merged [here](https://github.com/ros/catkin/pull/1070).

```
Traceback (most recent call last):
  File "C:\opt\ros\melodic\x64\lib\gazebo_ros\spawn_model", line 34, in <module>
    from tf.transformations import quaternion_from_euler
  File "C:\opt\ros\melodic\x64\lib\site-packages\tf\__init__.py", line 31, in <module>
    from .listener import Transformer, TransformListener, TransformerROS
  File "C:\opt\ros\melodic\x64\lib\site-packages\tf\listener.py", line 33, in <module>
    import sensor_msgs.msg
ImportError: No module named msg
```