ms-iot/ROSOnWindows #80: multiple files named <name> in package <package>

kejxu, 2019-02-27:
**Describe the bug**
A clear and concise description of what the bug is.

when trying to run `roslaunch turtlebot3_bringup turtlebot3_model.launch`, an error like this would occur:
```
>roslaunch turtlebot3_bringup turtlebot3_model.launch
[31mRLException: multiple files named [turtlebot3_model.launch] in package [turtlebot3_bringup]:
- C:\ros\catkin_ws\turtlebot3\install_isolated\share\turtlebot3_bringup\launch\turtlebot3_model.launch
- c:\ros\catkin_ws\turtlebot3\install_isolated\share\turtlebot3_bringup\launch\turtlebot3_model.launch
Please specify full path instead[0m
[31mThe traceback for the exception was written to the log file[0m
```

**To Reproduce**
Steps to reproduce the behavior:
* Catkin workspace setup instructions
  * Clone locations
  * build command line
  * Launch command line

source build `turtlebot3` and execute `roslaunch turtlebot3_bringup turtlebot3_model.launch`

seanyen-msft, 2019-02-27:
Did this still happen even with this fix? https://github.com/ms-iot/ros/pull/23

kejxu, 2019-02-27:
this is due to a bug in [ros_environment/env-hooks/1.ros_package_path.bat.em](https://github.com/ros/ros_environment/blob/melodic/env-hooks/1.ros_package_path.bat.em)

`os.path.normcase` would change `C:\` to `c:\` causing error later during string comparison in `rospkg` since case sensitive comparison will be used

this is already fixed for ms-iot build with https://github.com/ms-iot/ros_environment/pull/9
same fix has been submitted for ros mainline: https://github.com/ros/ros_environment/pull/14

kejxu, 2019-02-27:
> Did this still happen even with this fix? [ms-iot/ros#23](https://github.com/ms-iot/ros/pull/23)

no, https://github.com/ms-iot/ros/pull/23 would help work around this problem in `roslib\packages.py`. reason why I'm reworking this change:
- trying to minimize adding any (new) usage of `os.path.normpath(os.path.normcase())`, this kind of changes are expected to get noticeable resistance from ROS before they get convinced to update all path comparisons
- similar error could happen when a path from `ROS_PACKAGE_PATH` is compared against `CMAKE_PREFIX_PATH`; this is actual root cause, env dump before updating `ros_package_path.bat`:
```
CMAKE_PREFIX_PATH=C:/ros/catkin_ws/turtlebot3/install_isolated;C:/opt/ros/melodic/x64;C:/opt/rosdeps/x64
ROS_PACKAGE_PATH=c:\ros\catkin_ws\turtlebot3\install_isolated\share;C:\opt\ros\melodic\x64\share
```
imo, it would be easier/safer correct the cases in `ROS_PACKAGE_PATH`

closed 2019-02-27 by kejxu