ms-iot/ROSOnWindows #82: Status of tutorials on Windows?

imcardle, 2019-03-06:
**Describe the bug**
I'm new to ROS and MoveIt! so I have been following the tutorials for Windows here:
[ROSOnWindows - MoveIt Tutorials](https://ms-iot.github.io/ROSOnWindows/Moveit/moveit_tutorials.html)

However, I'm running into problems with rViz not starting up properly (error message "Entry Point Not Found" when trying to call robot_state_publisher.exe).

Can anyone offer a workaround or solution for this issue please?

**To Reproduce**
Execute the following tutorial demo:
roslaunch panda_moveit_config demo.launch

**Expected behavior**
rViz should start up without any error message.

**Environment information:**
ROS Distro: Melodic
OS Version: Windows 10
Binary build (MoveIt!):  20190305.1

**Error message**
[image]

seanyen-msft, 2019-03-06:
Thank @imcardle for reporting this. I am looking into that now.

seanyen-msft, 2019-03-06:
@imcardle I am not able to repro this issue on my end, So it would be nice you can provide some more info for me to troubleshoot:

* Console output of `roslaunch panda_moveit_config demo.launch -v`
* Console output of `choco list -lo`, which is to know what version you have.
* Archive for the logs under `C:\Users\<your username>\.ros\log`, which is to know more details what's happening.

Thanks!

imcardle, 2019-03-07:
@seanyen-msft Sure, I've edited the original post to add the requested diagnostics/logs.  Hope that helps!

seanyen-msft, 2019-03-07:
@imcardle Thanks! Can I also ask for your details OS information? Console output of `systeminfo | findstr /C:"OS"` would help.

And is there any outstanding windows updates? Sometimes it could cause this type of failure. Run `start ms-settings:windowsupdate` to check it out.

imcardle, 2019-03-07:
@seanyen-msft No problem - I've added the OS information to the environment details in the OP above, and I'm right up to date with the windows updates.

imcardle, 2019-03-14:
Hi @seanyen - have you any update on this issue?  In the meantime, I tried to get the tutorial running on a Linux virtual machine - but have run into another issue at the exact same point in the process!  I'm wondering if, although the symptoms are different, maybe the cause is the same?  You might like to take a look at it, just in case:
[https://github.com/ros-planning/moveit_tutorials/issues/297](https://github.com/ros-planning/moveit_tutorials/issues/297)

ooeygui, 2019-10-24:
@imcardle If you are still hitting this, please reopen.

closed 2019-10-24 by ooeygui