ms-iot/ROSOnWindows #83: Devspace package path should be ordered first in ROS_PACKAGE_PATH

seanyen, 2019-03-13:
**Describe the bug**
After you source `devel_isolated\setup.bat`, all the develop space search path should be coming first in `%ROS_PACKAGE_PATH%`.

**To Reproduce**
Steps to reproduce the behavior:
- create a catkin workspace.
- `catkin_make_isolated`.
- run `devel_isolated\setup.bat`.
- run `set ROS_PACKAGE_PATH`.

**Expected behavior**
I expect the installation path should come last in the list. However, I get: (notice where `C:\opt\ros\melodic\x64\share` shows up.)

`ROS_PACKAGE_PATH=C:/diag_ws/src/diagnostics/diagnostic_updater;C:/diag_ws/src/diagnostics/diagnostic_aggregator;C:/diag_ws/src/diagnostics/diagnostic_analysis;C:/diag_ws/src/diagnostics/self_test;C:\opt\ros\melodic\x64\share;C:/diag_ws/src/diagnostics/rosdiagnostic;C:/diag_ws/src/diagnostics/diagnostic_common_diagnostics;C:/diag_ws/src/diagnostics/test_diagnostic_aggregator;C:/diag_ws/src/diagnostics/diagnostics`

ooeygui, 2019-03-24:
Thank you! This is exactly what I was just debugging :-)
This also repros when debugging a single package.

Step to reproduce:

*  Create a new catkin workspace: c:\ws\a
* git clone -b init_windows https://github.com/ms-iot/image_pipeline
* catkin_make
* devel\setup.bat

Observed:
rospack list | findstr depth_image_proc
depth_image_proc C:\opt\ros\melodic\x64\share\depth_image_proc

Expected:
rospack list | findstr depth_image_proc
depth_image_proc C:/ws/a/src\image_pipeline\depth_image_proc

seanyen, 2019-10-14:
Closed it since the pull request is merged at the upstream.

closed 2019-10-14 by seanyen