ms-iot/ROSOnWindows #85: RViz MoveIt plugins not working for melodic on Windows 10

johnwason, 2019-03-15:
**Describe the bug**

Adding a "Motion Planning" plugin in RViz fails with the error:

```
PluginlibFactory: The plugin for class 'moveit_rviz_plugin/MotionPlanning' failed to load.  Error: Failed to load library C:/opt/ros/melodic/x64\bin/moveit_motion_planning_rviz_plugin.dll. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = C:/opt/ros/melodic/x64\bin/moveit_motion_planning_rviz_plugin.dll)
```

**To Reproduce**

rosrun rviz rviz

Click Add -> Motion Planning

**Expected behavior**
RVis should load the plugin without error.

**Environment information:**
 - OS version: Windows 1803
 - Chocolatey Versions: ros-melodic-moveit 20190314.1

seanyen, 2019-03-24:
@johnwason Thank for reporting! Can you try `choco upgrade ros-melodic-moveit -y` to get the latest bits and see if it worked? A recent broken merge on https://github.com/ms-iot/moveit seems to be the cause and it should be fixed now.

alecGraves, 2019-03-24:
Most recent version is working for me. Thanks @seanyen

johnwason, 2019-03-25:
Thanks @seanyen, that seems to have fixed the loading problem.

seanyen, 2019-03-25:
I closed it since it is fixed.

closed 2019-03-25 by seanyen