ms-iot/ROSOnWindows #87: link error about libpcl

bbw2008927, 2019-03-24:
**Describe the bug**
When compiling node which contain PCL library. Error report as below.
error LNK2019: can't resolve extern symbol "__declspec(dllimport) class boost::system::error_category con
st & __cdecl boost::system::detail::generic_category_ncx(void)" (__imp_?generic_category_ncx@detail@system@boost@@YAAEBVerror_category@23@XZ)，This symbol is referred in  "public: __cdecl boost::thread_exception::thread_exception(int,char const *)"
(??0thread_exception@boost@@QEAA@HPEBD@Z) 

**To Reproduce**
The CmakeList contain PCL config as below.
find_package(PCL 1.8 REQUIRED)
include_directories(${PCL_INCLUDE_DIRS})
link_directories(${PCL_LIBRARY_DIRS})
add_executable(pcl_sample pcl_sample.cxx)
target_link_libraries(pcl_sample ${catkin_LIBRARIES} ${PCL_LIBRARIES})
 
And in the main function, just simple definition.
pcl::PointCloud<pcl::PointXYZ> cloud;

Run catkin_make in workspace.
The source code will be attached in next commit.
**Expected behavior**
Compile pass.

**Environment information:**
 - OS version: 1809 17763.379
 - Chocolatey Versions:
Chocolatey v0.10.13
2 validations performed. 1 success(es), 1 warning(s), and 0 error(s).

Validation Warnings:
 - A pending system reboot request has been detected, however, this is
   being ignored due to the current command being used 'list'.
   It is recommended that you reboot at your earliest convenience.

assimp 4.0.1.1809242322
boost 1.66.0.1809242326
bullet3 2.87.0.1809242326
bzip2 1.0.6.1901242108
chocolatey 0.10.13
chocolatey-core.extension 1.3.3
chocolatey-windowsupdate.extension 1.0.3
console_bridge 0.4.0.1809242322
cppunit 1.12.1.1809242322
cppzmq 4.2.2.1811162207
dlfcn-win32 1.1.1.1811162226
eigen 3.3.4.1809242322
freeimage 3.17.0.1811160037
gazebo9 9.4.1.1901022119
git 2.21.0
git.install 2.21.0
google-mock 1.8.0.1809242326
google-test 1.8.0.1809242322
gtk2 2.22.1.1809242322
hg 4.9.1
ilmbase 2.2.1.1811210800
jasper 2.0.14.1811210919
jxrlib 1.1.0.1811210809
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.4
KB3035131 1.0.2
lcms 2.8.0.1811210903
libcurl 7.58.0.1809242322
libflann 1.9.1.1809090720
libgraphviz 2.41.0.1810022039
libjpeg-turbo 1.5.3.1809242322
liblzma 5.2.4.1811210837
libogg 1.3.3.1809090720
libopencv 3.4.1.1809242322
libpcl 1.8.1.1809090720
libpng 1.6.35.1809242322
libraw 0.19.0.1811210804
libtbb-dev 2018.6.0.1812011750
libtheora 1.1.1.1809090720
libwebp 0.6.1.1811210731
log4cxx 0.10.0.1809242322
lz4 1.8.1.1809242322
ogre 1.10.11.1901312214
openexr 2.2.1.1811210832
openjpeg 2.3.0.1811210736
openssl 1.1.1.1809242322
pkg-config 0.29.2.1809242322
poco 1.8.1.1809242322
protobuf 3.6.1.1811130545
pyqt5 5.10.1.1809242322
pyside2 5.10.1.1809242322
qt5-sdk 5.10.1.1903182340
qwt 6.1.3.1811212304
ros-catkin-tools 0.0.1.1903221831
ros-cmake 3.11.4.1809242322
ros-melodic-desktop 20190323.1
ros-melodic-desktop_full 20190323.1
ros-python2 2.7.15.1809242354
sbcl 0.0.0.1809242326
sdl 1.2.15.1809242322
sdl_image 1.2.12.1809242322
sip 4.19.8.1809242322
tiff 4.0.9.1811210742
tinyxml 2.6.2.1809242322
tinyxml2 6.1.0.1809242322
urdfdom 1.0.0.1809242322
urdfdom_headers 1.0.0.1809242322
vcpython27 9.0.30729.1809090720
vcredist140 14.16.27027.1
yaml-cpp 0.6.2.1901170034
zeromq 4.2.5.1811162203
zlib 1.2.11.1809242322
74 packages installed.

bbw2008927, 2019-03-24:
[pcl_node.zip](https://github.com/ms-iot/ROSOnWindows/files/3000348/pcl_node.zip)

seanyen, 2019-03-24:
@bbw2008927 

After some tweak on your CMakeLists.txt, I can compile your program. Here is my catkin_make log: https://gist.github.com/seanyen/9cdf2c9ab7913c1d77542cca9c2c3f92

(I modified the line 17 to `find_package(PCL 1.8 REQUIRED COMPONENTS common)`, otherwise it complains VTK not found, which is something not in our Chocolatey server.)

Can you compare my logs to yours? Particularly the following lines?

https://gist.github.com/seanyen/9cdf2c9ab7913c1d77542cca9c2c3f92#file-build-log-L60-L62

bbw2008927, 2019-03-26:
Hi seanyen
Yes, you are right. VTK isn't in libpcl get from choco.
And Cmake find VTK from my vcpkg folder. If I only use conponent of common, I can compile without error.
But there are other issue.
1. When I try to run choco uninstall libpcl, choco will delete all the thing in the C:\opt\rosdeps\x64\include. It make me to uninstall all the package of choco.
2. After choco upgrade ros-melodic-desktop 20190325. there is new issue about compiling.
This is log:
Base path: E:\ros_ws
Source space: E:\ros_ws\src
Build space: E:\ros_ws\build
Devel space: E:\ros_ws\devel
Install space: E:\ros_ws\install
####
#### Running command: "cmake E:\ros_ws\src -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCATKIN_DEVEL_PREFIX=E:\ros_ws\devel -DCMAKE_INSTALL_PREFIX=E:\ros_ws\install -G NMake Makefiles" in "E:\ros_ws\build"
####
-- Using CATKIN_DEVEL_PREFIX: E:/ros_ws/devel
-- Using CMAKE_PREFIX_PATH: E:/ros_ws/devel;C:/opt/ros/melodic/x64;C:/opt/rosdeps/x64
-- This workspace overlays: E:/ros_ws/devel;C:/opt/ros/melodic/x64
-- Using PYTHON_EXECUTABLE: C:/opt/python27amd64/python.exe
-- Using default Python package layout
-- Using empy: C:/opt/python27amd64/lib/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: E:/ros_ws/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: C:/opt/python27amd64/Scripts/nosetests-2.7.exe
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - pcl_node
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'pcl_node'
-- ==> add_subdirectory(pcl_node)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Configuring done
-- Generating done
-- Build files have been written to: E:/ros_ws/build
####
#### Running command: "nmake" in "E:\ros_ws\build"
####

Microsoft (R)  toolchain 14.16.27026.1 
Copyright (C) Microsoft Corporation.  copyright reserved。

[ 50%] Built target _setup_util.py_executable_install_python
NMAKE : fatal error U1073: don't know how to generate “C:\opt\rosdeps\x64\lib\qhullstatic.lib”
Stop.
NMAKE : fatal error U1077: “"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023 \bin\HostX64\x64\nmake.exe"”: return code “0x2”
Stop.
NMAKE : fatal error U1077: “"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023 \bin\HostX64\x64\nmake.exe"”: return code “0x2”

Stop.
Invoking "nmake" failed

I never see C:\opt\rosdeps\x64\lib\qhullstatic.lib before. And I never see such error before.
The code is attached in this task.
[pcl_node.zip](https://github.com/ms-iot/ROSOnWindows/files/3006462/pcl_node.zip)

Would you tell me how to roll back ros-melodic_desktop to 20190323?

bbw2008927, 2019-03-30:
I use vcpkg to compile qhull static library and copy files into C:\opt\rosdeps\x64\lib.
And I still use ros-melodic-desktop 20190325.
I can compile pcl_ros application.

HarelAshwal, 2020-06-19:
LK"I

Hi All,

Is there any workaround for that issue?
Can someone share a working example on how to run a simple pcl node in windows ros-melodic-desktop?

Thanks,

Harel

ooeygui, 2020-06-22:
Hi @HarelAshwal ,
Ask: I'd love to understand more about the scenario and nodes you are attempting to use.

Background:
pcl is available as a vcpkg - which is C++ library manager from Microsoft. This has been integrated into ROSDep, but does require some Win32 specific changes to link.

We have documented here for leveraging vcpkg and porting nodes:
https://aka.ms/ros/docs

closed 2020-11-20 by ooeygui