ms-iot/ROSOnWindows #88: [MoveIt] UR3 build fails due to missing chainiksolver_pos_nr_jl_mimic.hpp

alecGraves, 2019-03-24:
**Describe the bug**
Compiling universal_robot package fails because of missing moveit .hpp

```
fatal error C1083: Cannot open include file: 'moveit/kdl_kinematics_plugin/chainiksolver_pos_nr_jl_mimic.hpp': No such file or directory
```

<details><summary> (click here) Full build log: </summary>
<p>

```
Base path: C:\Users\alec\Documents\moveit_tutorial_ws
Source space: C:\Users\alec\Documents\moveit_tutorial_ws\src
Build space: C:\Users\alec\Documents\moveit_tutorial_ws\build
Devel space: C:\Users\alec\Documents\moveit_tutorial_ws\devel
Install space: C:\Users\alec\Documents\moveit_tutorial_ws\install
####
#### Running command: "nmake cmake_check_build_system" in "C:\Users\alec\Documents\moveit_tutorial_ws\build"
####

Microsoft (R) Program Maintenance Utility Version 14.16.27024.1
Copyright (C) Microsoft Corporation.  All rights reserved.

####
#### Running command: "nmake" in "C:\Users\alec\Documents\moveit_tutorial_ws\build"
####

Microsoft (R) Program Maintenance Utility Version 14.16.27024.1
Copyright (C) Microsoft Corporation.  All rights reserved.

[  1%] Built target _setup_util.py_executable_install_python
[  1%] Built target _ur_msgs_generate_messages_check_deps_RobotStateRTMsg
[  1%] Built target _ur_msgs_generate_messages_check_deps_IOStates
[  1%] Built target _ur_msgs_generate_messages_check_deps_SetIO
[  1%] Built target _ur_msgs_generate_messages_check_deps_Digital
[  1%] Built target _ur_msgs_generate_messages_check_deps_MasterboardDataMsg
[  1%] Built target _ur_msgs_generate_messages_check_deps_SetPayload
[  1%] Built target _ur_msgs_generate_messages_check_deps_ToolDataMsg
[  1%] Built target std_msgs_generate_messages_cpp
[  1%] Built target _ur_msgs_generate_messages_check_deps_Analog
[ 11%] Built target ur_msgs_generate_messages_cpp
[ 11%] Built target std_msgs_generate_messages_py
[ 24%] Built target ur_msgs_generate_messages_py
[ 24%] Built target std_msgs_generate_messages_nodejs
[ 34%] Built target ur_msgs_generate_messages_nodejs
[ 34%] Built target std_msgs_generate_messages_eus
[ 45%] Built target ur_msgs_generate_messages_eus
[ 45%] Built target std_msgs_generate_messages_lisp
[ 55%] Built target ur_msgs_generate_messages_lisp
[ 55%] Built target ur_msgs_generate_messages
[ 56%] Built target ur_driver_gencfg
[ 78%] Built target ur_driver_common
[ 82%] Built target ur_hardware_interface
[ 82%] Built target geometry_msgs_generate_messages_py
[ 82%] Built target actionlib_generate_messages_cpp
[ 82%] Built target geometry_msgs_generate_messages_nodejs
[ 82%] Built target roscpp_generate_messages_eus
[ 82%] Built target actionlib_msgs_generate_messages_cpp
[ 82%] Built target roscpp_generate_messages_cpp
[ 82%] Built target rosgraph_msgs_generate_messages_eus
[ 82%] Built target rosgraph_msgs_generate_messages_nodejs
[ 82%] Built target roscpp_generate_messages_lisp
[ 82%] Built target geometry_msgs_generate_messages_eus
[ 82%] Built target trajectory_msgs_generate_messages_lisp
[ 82%] Built target rosgraph_msgs_generate_messages_cpp
[ 82%] Built target roscpp_generate_messages_py
[ 82%] Built target roscpp_generate_messages_nodejs
[ 82%] Built target rosgraph_msgs_generate_messages_lisp
[ 82%] Built target rosgraph_msgs_generate_messages_py
[ 82%] Built target sensor_msgs_generate_messages_py
[... 163 lines cut ...]
ros-catkin-tools 0.0.1.1901231908
ros-cmake 3.11.4.1809242322
ros-melodic-desktop_full 20190324.1
ros-melodic-moveit 20190324.1
ros-python2 2.7.15.1809242354
sbcl 0.0.0.1809242326
sdl 1.2.15.1809242322
sdl_image 1.2.12.1809242322
sip 4.19.8.1809242322
swig 3.0.12
tiff 4.0.9.1811210742
tinyxml 2.6.2.1809242322
tinyxml2 6.1.0.1809242322
urdfdom 1.0.0.1809242322
urdfdom_headers 1.0.0.1809242322
vcpython27 9.0.30729.1809090720
vcredist140 14.16.27027.1
yaml-cpp 0.6.2.1901170034
zeromq 4.2.5.1811162203
zlib 1.2.11.1809242322
83 packages installed.
```

</p>
</details>

seanyen, 2019-03-25:
@alecGraves Most likely it is caused by this recent [change](https://github.com/ros-industrial/universal_robot/commit/1f2a2b54c40c4bcadefb5f2f9ce52fc6cf1dc36a) not being rebased into https://github.com/ms-iot/universal_robot.

Can you try to remove those non-existent headers to see if it go through?

alecGraves, 2019-03-25:
Yep, that fixed it. Thanks!

ooeygui, 2019-05-17:
Closing since fixed :-)

closed 2019-05-17 by ooeygui