ms-iot/ROSOnWindows #91: catkin_make failing, looking for argsparse

seanyen, 2019-03-27:
### Problem

A recent release of [catkin_pkg](https://pypi.org/project/catkin-pkg/#history) comes with a [change](https://github.com/ros-infrastructure/catkin_pkg/commit/5cf0ea92d9ff888d48413b2948370a0bbb34abdc) to remove `argparse` from the requirement list from `catkin_pkg`.

And this delta (compared to the ms-iot version) causes symptoms like `argparse` not found: http://answers.ros.org/question/319726/catkin_make-failing-looking-for-argsparse

### Resolution
Since all the catkin_pkg changes are considered upstream'ed, we no longer need to patch the catkin_pkg for `ROS on Windows` installation. (See https://github.com/ms-iot/rosdistro-db/pull/16)

For who already installed the broken builds, please run `choco install ros-catkin-tools -y --force`.

### Details
The ros-catkin-tool install the all the ROS tool from pypi.org and then patch it with the ms-iot version (for catkin_pkg). And since there is a while we haven't rebase to the latest upstream catkin_pkg, this delta mentioned above becomes an inconsistency on what catkin_pkg agrees the dependencies. And when rosdep runs, it calls the old version of catkin_pkg which still says "argparse" to check and it propagates back to cause dependency check errors in rosdep side.

bakerhillpins, 2019-03-28:
The above quoted command to force the reinstall did not work in my situation. The version is/was apparently the most recent already:
```
choco install ros-catkin-tools -y --force
Chocolatey v0.10.13
Installing the following packages:
ros-catkin-tools
By installing you accept licenses for the packages.
ros-catkin-tools v0.0.1.1903221831 already installed. Forcing reinstall of version '0.0.1.1903221831'.
 Please use upgrade if you meant to upgrade to a new version.
```

I see that changes have been made and merged into the referenced issue in rosdistro so I suspect that I'm here before it's available?

seanyen, 2019-03-28:
@bakerhillpins What's your error when you run catkin_make or rosdep? One other workaround is to `pip install -U argparse` to get around the problem.

bakerhillpins, 2019-03-28:
@seanyen Ok, so I went back and tried this again and then noticed I made the same mistake you did in the title, I put argSparse back into requires.txt before I tried your suggestion. So i corrected that and executed the catkin_make again.

It's now reporting the dependency PY_em missing: (I will go back to my issue on ROS Answers and update with this info as well.)

```
catkin_ws>catkin_make
Base path: C:\Users\xx\catkin_ws
Source space: C:\Users\xx\catkin_ws\src
Build space: C:\Users\xx\catkin_ws\build
Devel space: C:\Users\xx\catkin_ws\devel
Install space: C:\Users\xx\catkin_ws\install
####
#### Running command: "cmake C:\Users\xx\catkin_ws\src -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCATKIN_DEVEL_PREFIX=C:\Users\xx\catkin_ws\devel -DCMAKE_INSTALL_PREFIX=C:\Users\xx\catkin_ws\install -G NMake Makefiles" in "C:\Users\xx\catkin_ws\build"
####
-- Using CATKIN_DEVEL_PREFIX: C:/Users/xx/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: C:/opt/ros/melodic/x64
-- This workspace overlays: C:/opt/ros/melodic/x64
-- Using PYTHON_EXECUTABLE: C:/opt/python27amd64/python.exe
-- Using default Python package layout
-- Could NOT find PY_em (missing: PY_EM)
CMake Error at C:/opt/ros/melodic/x64/share/catkin/cmake/empy.cmake:29 (message):
  Unable to find either executable 'empy' or Python module 'em'...  try
  installing the package 'python-empy'
Call Stack (most recent call first):
  C:/opt/ros/melodic/x64/share/catkin/cmake/all.cmake:163 (include)
  C:/opt/ros/melodic/x64/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:56 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/Users/xx/catkin_ws/build/CMakeFiles/CMakeOutput.log".
Invoking "cmake" failed
```

[edit - you update just came through, I'll read that in a sec.]

seanyen, 2019-03-28:
@bakerhillpins 

Good to hear that you found out the `argparse` typo thing. Btw, do you mind to share what's the `requires.txt` file location you touched?

For the problem of missing `empy`, it is most likely also caused by this `argparse` issue, which prevents `rosdep` to run correctly before. (I suspect there are lots of system dependencies not yet deployed to your machine too, `empy` is just one of them.)

In this case, can you do `choco upgrade -y --force ros-melodic-desktop` (or the `ros-melodic-desktop_full`, which depends on what the package you installed before.) to trigger `rosdep` again?

bakerhillpins, 2019-03-28:
@seanyen requires.txt location is as follows:
C:\opt\python27amd64\Lib\site-packages\catkin_pkg-0.4.8-py2.7.egg-info

I just ran the following. 
```
 C:\Windows\System32>choco upgrade -y --force ros-melodic-desktop
Chocolatey v0.10.13
Upgrading the following packages:
ros-melodic-desktop
By upgrading you accept licenses for the packages.
ros-melodic-desktop v20190325.1 is the latest version available based on your source(s).
```
The output of this upgrade is SUBSTANTIALLY different/longer from my initial install which I recorded previously. However, The version number, v20190325., hasn't changed.

seanyen, 2019-03-28:
@bakerhillpins Yes, and please be patient if it takes SUBSTANTIALLY longer time to finish. It sounds like `rosdep` is working on getting everything you need for ROS node development. Let me know how it goes.

bakerhillpins, 2019-03-28:
@seanyen Well something happened during the time I stepped away from my desk and a network disturbance caused a fatal error:

```
Progress: Downloading qt5-sdk 5.10.1.1903182340... 100%

qt5-sdk v5.10.1.1903182340
qt5-sdk package files upgrade completed. Performing other installation steps.
Extracting C:\ProgramData\chocolatey\lib\qt5-sdk\tools\qt5-sdk-5.10.1-msvc2017-64.zip to C:\opt\rosdeps\x64...
Chocolatey timed out waiting for the command to finish. The timeout
 specified (or the default value) was '2700' seconds. Perhaps try a
 higher `--execution-timeout`? See `choco -h` for details.
The upgrade of ros-melodic-desktop was NOT successful.
Error while running 'C:\ProgramData\chocolatey\lib\ros-melodic-desktop\tools\chocolateyInstall.ps1'.
 See log for details.
C:\opt\rosdeps\x64
 The upgrade of qt5-sdk was successful.
  Software installed to 'C:\opt\rosdeps\x64'

Chocolatey upgraded 1/1 packages.
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).

Chocolatey upgraded 0/2 packages. 2 packages failed.
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).

Failures
 - ros-melodic-desktop (exited -1) - Error while running 'C:\ProgramData\chocolatey\lib\ros-melodic-desktop\tools\chocolateyInstall.ps1'.
 See log for details.
 - vcredist140 (exited 1638) - Error while running 'C:\ProgramData\chocolatey\lib\vcredist140\tools\chocolateyInstall.ps1'.
 See log for details.
```
I'll retry the command and hopefully we don't have the same problem.

bakerhillpins, 2019-03-28:
@seanyen  Score!  catkin_make was successful after the `choco upgrade -y --force ros-melodic-desktop` command executed successfully.

Thanks for your help! I'll update the ROS Answers question.

Can you help me understand what I may have missed with the initial depoy of ROS on Windows. I explicitly followed the instructions [here](https://ms-iot.github.io/ROSOnWindows/GettingStarted/Setup.html). 

Ah - After revisiting the page it seems that the command to install ROS for Windows under the Binary Package Installation section changed from:
`choco upgrade ros-melodic-desktop -y`
to
`choco upgrade ros-melodic-desktop_full -y`

ooeygui, 2019-03-28:
Thank you for reporting this and working through the issue with us. 
During ros-melodic-desktop install python scripts are pulled in from pip. There was a change we didn't have a match for in our distribution. Since we have upstreamed our work in that component, we were able to snap to the mainline.

Because chocolatey caches the files, --force was needed to purge the cache.

The only difference between desktop and desktop_full is that the latter contains gazebo and moveit.

bakerhillpins, 2019-03-29:
@ooeygui  NP.  As I was going to pull down gazebo and moveit eventually that's a plus. Appreciate the quick attention and support.

ooeygui, 2019-05-17:
Fixed.

closed 2019-05-17 by ooeygui