ms-iot/ROSOnWindows #92: roscd not working, missing findpath.py

bakerhillpins, 2019-03-29:
**Describe the bug**
Working my way through the ROS [tutorials](http://wiki.ros.org/ROS/Tutorials/NavigatingTheFilesystem) for navigating the filesystem. roscd command returns:

```
C:\Windows\System32>roscd roscpp
python: can't open file 'C:\opt\ros\melodic\x64\bin\\findpath.py': [Errno 2] No such file or directory
( was unexpected at this time.
```

**To Reproduce**
Steps to reproduce the behavior:
Open a command prompt and try the following command - `roscd roscpp`

**Expected behavior**
should have changed to the roscpp package directory

**Environment information:**
 - OS version: `1803`
I just updated to ros-melodic-desktop 20190329.1.

I've searched the entire c:\opt directory for the missing python script findpath.py and it's not found.

bakerhillpins, 2019-03-29:
Apparently my search foo isn't working well for me today: 
#11

kejxu, 2019-03-29:
Thanks for trying ROS on Windows out and reporting this issue! My apologies for this nuisance! While upstreaming our changes to ROS mainline, the script was renamed yet the renaming was not done correctly.

[A fix](https://github.com/ms-iot/ros/pull/27) has been made for our `ms-iot` repo, and that should become part of the binaries on 20190330. It should work by then =)

Meantime, we would be working to upstream this change to ROS mainline. Thanks again!

closed 2019-03-29 by kejxu

bakerhillpins, 2019-04-01:
NP. I updated to Version 20190331.1 and the commands are working now. Thanks!