ms-iot/ROSOnWindows #93: rostest(...) causes invalid file names containing ":"

alecGraves, 2019-03-30:
**Describe the bug**
```rostest``` causes build errors due to making invalid filenames with ":"

At bottom of CMakeLists.txt of [ar_track_alvar](https://github.com/ros-perception/ar_track_alvar) several rostest(...) statements exist, like:
```add_rostest(test/marker_arg_config-basic.test DEPENDENCIES ${PROJECT_NAME}_4markers_tork.bag)```

These tests are causing build errors like:

```
CMake Error: Cannot open file for write: C:/Users/alec/Documents/catkin_ws/build/JACO/ar_track_alvar/ar_track_alvar/CMakeFiles/_run_tests_ar_track_alvar_roslaunch-check_C:_Users_alec_Documents_catkin_ws_src_JACO_ar_track_alvar_ar_track_alvar_test_marker_arg_config-full.test.dir/build.make.tmp
CMake Error: : System Error: Invalid argument
CMake Error: Could not create C:/Users/alec/Documents/catkin_ws/build/JACO/ar_track_alvar/ar_track_alvar/CMakeFiles/_run_tests_ar_track_alvar_roslaunch-check_C:_Users_alec_Documents_catkin_ws_src_JACO_ar_track_alvar_ar_track_alvar_test_marker_arg_config-full.test.dir/cmake_clean.cmake
```

Rebuilding also reveals the problem in catkin_ws/build/Makefile:
```
makefile(3907) : fatal error U1033: syntax error : ':' unexpected
```
In catkin_ws/build/Makefile:
```Makefile
_run_tests_ar_track_alvar_roslaunch-check_C:_Users_alec_Documents_catkin_ws_src_JACO_ar_track_alvar_ar_track_alvar_test_marker_arg_config-full.test: cmake_check_build_system
	$(MAKE) -f CMakeFiles\Makefile2 /nologo -$(MAKEFLAGS) _run_tests_ar_track_alvar_roslaunch-check_C:_Users_alec_Documents_catkin_ws_src_JACO_ar_track_alvar_ar_track_alvar_test_marker_arg_config-full.test
.PHONY : _run_tests_ar_track_alvar_roslaunch-check_C:_Users_alec_Documents_catkin_ws_src_JACO_ar_track_alvar_ar_track_alvar_test_marker_arg_config-full.test

```

**To Reproduce**
```bat
choco upgrade -y ros-melodic-desktop_full
mkdir rostest_ws\src
cd rostest_ws\src
git clone https://github.com/ros-perception/ar_track_alvar
cd ..
catkin_make
catkin_make
```

**Expected behavior**
```rostest``` statements do not cause this kind of error (the build will likely still fail, but due to other problems).

**Environment information:**
 - OS version: Microsoft Windows Version 1809 (OS Build 17763.379)

<details><summary> Chocolatey Versions (click to expand):</summary>
<p>

```
Chocolatey v0.10.11
assimp 4.0.1.1809242322
boost 1.66.0.1809242326
bullet3 2.87.0.1809242326
bzip2 1.0.6.1901242108
cairo 1.15.12.1810221716
ceres 1.14.0.1812111905
chocolatey 0.10.11
chocolatey-core.extension 1.3.3
chocolatey-windowsupdate.extension 1.0.3
clapack 3.2.1.1811082319
console_bridge 0.4.0.1809242322
cppunit 1.12.1.1809242322
cppzmq 4.2.2.1811162207
dlfcn-win32 1.1.1.1811162226
eigen 3.3.4.1809242322
freeglut 3.0.0.1809242322
freeimage 3.17.0.1811160037
gazebo9 9.4.1.1901022119
gflags 2.2.1.1811082321
glog 0.3.5.1811082321
google-mock 1.8.0.1809242326
google-test 1.8.0.1809242322
gtk2 2.22.1.1809242322
hg 4.9
ilmbase 2.2.1.1811210800
jasper 2.0.14.1811210919
jxrlib 1.1.0.1811210809
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.4
KB3035131 1.0.2
lcms 2.8.0.1811210903
libccd 2.0.0.1809242322
libcurl 7.58.0.1809242322
libfcl 0.5.0.1809242322
libflann 1.9.1.1809090720
libglew 2.1.0.1809101757
libgraphviz 2.41.0.1810022039
libjpeg-turbo 1.2.1.201304081
liblzma 5.2.4.1811210837
libogg 1.3.3.1809090720
libompl 1.2.2.1809120057
libopencv 3.4.1.1809242322
libpcl 1.8.1.1809090720
libpng 1.6.35.1809242322
libqglviewer 2.7.1.1809090720
libqhull 2015.2.0.1809242322
libraw 0.19.0.1811210804
libtbb-dev 2018.6.0.1812011750
libtheora 1.1.1.1809090720
libwebp 0.6.1.1811210731
log4cxx 0.10.0.1809242322
lua 5.3.4.1810221716
lz4 1.8.1.1809242322
metis 5.1.0.1811082350
octomap 1.9.0.1809090720
ogre 1.10.11.1901312214
openblas 0.2.20.1811082312
openexr 2.2.1.1811210832
openjpeg 2.3.0.1811210736
openssl 1.1.1.1809242322
pkg-config 0.29.2.1809242322
poco 1.8.1.1809242322
protobuf 3.6.1.1811130545
pyqt5 5.10.1.1809242322
pyside2 5.10.1.1809242322
qt5-sdk 5.10.1.1810112239
qwt 6.1.3.1811212304
ros-catkin-tools 0.0.1.1901231908
ros-cmake 3.11.4.1809242322
ros-melodic-cartographer_ros 20190324.1
ros-melodic-desktop_full 20190329.1
ros-melodic-moveit 20190324.1
ros-python2 2.7.15.1809242354
sbcl 0.0.0.1809242326
sdl 1.2.15.1809242322
sdl_image 1.2.12.1809242322
sip 4.19.8.1809242322
suitesparse 1.4.0.1811082321
swig 3.0.12
tiff 4.0.9.1811210742
tinyxml 2.6.2.1809242322
tinyxml2 6.1.0.1809242322
urdfdom 1.0.0.1809242322
urdfdom_headers 1.0.0.1809242322
vcpython27 9.0.30729.1809090720
vcredist140 14.16.27027.1
yaml-cpp 0.6.2.1901170034
zeromq 4.2.5.1811162203
zlib 1.2.11.1809242322
91 packages installed.
```

</p>
</details>

---

I am commenting out the rostests for now and continuing with the port.

seanyen, 2019-04-03:
Thanks for the reporting! We will take a look soon.

seanyen, 2020-01-17:
The fix is merged from the upstream. Closing it.

closed 2020-01-17 by seanyen