ms-iot/ROSOnWindows #94: Cannot build rosserial due to python installer

ooeygui, 2019-03-31:
**Steps to repro:**
```
mkdir c:\ws\rosserial_ws\src
cd c:\ws\rosserial_ws\src
catkin_init_workspace
git checkout -b init_windows http://github.com/ms-iot/rosserial.git
cd ..
catkin_make_install
```
**Result**
install TARGETS givn target "make_libraries.py_executable_install_python"

**Cause**
make_libraries.py exists in multiple directories.

**workaround**
echo > rosserial_XXX\CATKIN_IGNORE 

Where XXX is what you don't want to build.

kejxu, 2019-04-09:
this duplicate-macro issue was hit because `catkin_make` treats all packages in the workspace [like a single project](https://answers.ros.org/question/320613/catkin_make-vs-catkin_make_isolated-which-is-preferred/)

I will check if the macros could be renamed, that seems like the easiest way to resolve this

kejxu, 2019-04-09:
> isn't the suggestion in the ROS Answers question you link to not to use `catkin_make`, specifically (but not exclusively) for the reported issue?

thanks for pointing this out! We are still considering a fix for this change because `catkin_make` would work in this case before this [Windows-specific change](https://github.com/ros/catkin/commit/a4e2b7c0a44d68f8a0054e091c0c0755aa63398b), making this issue seemingly a regression; meantime, I think it could/might be a pretty simple fix.

as suggested in the linked question (thanks a lot for answering that!), we will be looking at `catkin_make_isolated` and/or `colcon` for our VS Code integration

gavanderhoorn, 2019-04-09:
I think it was @dirk-thomas who answered that specific question though.

seanyen, 2019-09-25:
I verified it is no longer repro, so I close it.

closed 2019-09-25 by seanyen

kejxu, 2019-09-25:
> I verified it is no longer repro, so I close it.

this is fixed in https://github.com/ros/catkin/commit/3d4fa72a18c7faa061f431d7727c5a87ba1a5b33