ms-iot/ROSOnWindows #95: rostopic using deprecated API

bakerhillpins, 2019-04-02:
**Describe the bug**
using rostopic pub issues warning about using unsafe/deprecated API.

**To Reproduce**
Start up the TurtleSim tutorial,
Goto the [Understanding Topics tutorial](http://wiki.ros.org/ROS/Tutorials/UnderstandingTopics), pub section.
Execute the pub command with a 1hz rate and note the warning:

```
C:\Windows\System32>rostopic pub /turtle1/cmd_vel geometry_msgs/Twist -r 1 -- "[2.0, 0.0, 0.0]" "[0.0, 0.0, 1.8]"
C:\opt\ros\melodic\x64\lib\site-packages\rostopic\__init__.py:1782: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  pub_args.append(yaml.load(arg))
```

**Expected behavior**
no warning

**Environment information:**
ros-melodic-desktop v20190331.1

ooeygui, 2019-04-03:
Hi @bakerhillpins 
Thanks for the report. I don't believe this problem is specific to Windows. Looking upstream it doesn't seem to be corrected.

ooeygui, 2019-04-03:
@kejxu We should remove the ros_comm repo since it has been upstreamed. When this is fixed in main, Windows will inherit the fix.

bakerhillpins, 2019-04-03:
Sorry about that. My project is specifically targeted at ROS for Windows so I was wondering when I would get to issues that were upstream related. Looks like I got there sooner than I expected. I'll look into getting a Docker container with a linux ROS node on it to compare to but that's probably not in the near term.
Thanks for your prompt attention.

seanyen, 2019-04-03:
https://github.com/ros/ros_comm/issues/1686 This is also recently reported to the upstream. Added it here for reference.

kejxu, 2019-04-03:
Thanks for reporting this @bakerhillpins, and @ooeygui thanks for the reminder!

A part of the reason why we have not switched our build to point to `ros/ros_comm` is because `ms-iot/ros_comm` still has some test code changes. `ms-iot/ros_comm` will be deleted once there is no difference between our build and ros mainline.

phamkhactu, 2019-09-26:
I do follow tutorials rosbridge + roslibjs . But  i Pub message can not subcribe in websocket 
[image]. I dont kown why. Can you help me. Thanks more for helping. I have faced with bugs 2 days

phamkhactu, 2019-09-26:
here your link [tutorial](https://msadowski.github.io/ros-web-tutorial-pt1/?fbclid=IwAR2EHUnTJ9fISQ03lt58qxKrMEsxvfXZEJLYGnrEGWzz0tUsxeKF6koUUno)

bakerhillpins, 2019-09-26:
@phamkhactu This thread should be confined to discussions related to the deprecated API as described in the issues first post. 
Please ask a new question concerning your particular problem on [ROS Answers](https://answers.ros.org/questions/) where you are likely to get a better response from the community.

ooeygui, 2019-09-26:
@bakerhillpins thanks for replying to the thread.  I'm going to close this issue as it is not Windows specific.

closed 2019-09-26 by ooeygui