ms-iot/ROSOnWindows #98: ROSOnWindows does not work well in VS code

arwtyxouymz, 2019-04-07:
I'm a newer of ROSOnWindows and Windows Development and I'm sorry if I asked easy questions.

I have two questions about development in VS code.

# ROS for VSCode Plugin does not work
I'm going to use [ROS for VSCode Plugin](https://marketplace.visualstudio.com/items?itemName=ajshort.ros) which you recommended in [here](https://ms-iot.github.io/ROSOnWindows/GettingStarted/UsingROSonWindows.html), but it cannot setup.

### Error message
```
Could not source the setup file for ROS distro "melodic".
```
I specified ROS Distro to `melodic` in `settings.json` like
```json
{
    "ros.distro": "melodic",
    "C_Cpp.intelliSenseEngineFallback": "Disabled",
    "C_Cpp.errorSquiggles": "Disabled"
}
```
But this plugin can't setup or find `melodic` distribution in my local.

# Include Error
I created `main.cpp` and `<ros/ros.h>` is included in main, but this error occurred.
```
cannot open source file "boost/math/special_functions/round.hpp" (dependency of "ros/ros.h")
```
I set the `includePath` in `c_cpp_properties.json` like below, but error occurred.
```json
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "${workspaceFolder}/**",
                "C:/opt/ros/melodic/x64/include"
            ],
            "defines": [
                "_DEBUG",
                "UNICODE",
                "_UNICODE"
            ],
            "windowsSdkVersion": "10.0.17763.0",
            "compilerPath": "C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx64/x64/cl.exe",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "intelliSenseMode": "msvc-x64"
        }
    ],
    "version": 4
}
```

How can i fix it?

My environment is:

```
OS: Windows 10 Home
Visual Studio: 2017 Community

VS Code: 
Version: 1.33.0 (user setup)
Commit: 0dd516dd412d42323fc3464531b1c715d51c4c1a
Date: 2019-04-04T15:14:28.026Z
Electron: 3.1.6
Chrome: 66.0.3359.181
Node.js: 10.2.0
V8: 6.6.346.32
OS: Windows_NT x64 10.0.17134
```
and I installed prebuild ROSOnWindows via Chocolatey in `C:\opt\`.

ooeygui, 2019-04-07:
Hi @arwtyxouymz ,
Thanks for the report. 
We're working on updating this plugin to work with Windows. The original author of the plugin has transfered the project to microsoft for ongoing maintanance, and we're working on the logistics of getting it published. 

For now, using the command line is the best option.

I'm finalizing some videos which we will post to our youtube channel showing the basic flow, and we'll update as new behaviors come online.

We'll post back when it is updated.

ooeygui, 2019-05-17:
We will be releasing an updated VSCode ROS extension shortly.

Brabalawuka, 2019-07-15:
> We will be releasing an updated VSCode ROS extension shortly.

This problem is still not fixed as I face the same exact issue

kejxu, 2019-07-15:
> This problem is still not fixed as I face the same exact issue

Hi @Brabalawuka, thanks for reaching out! Quick question, which issue are you hitting? I noticed that you have kindly shared your setup in https://github.com/ms-iot/vscode-ros/pull/73 so I'm guessing you're having problem with the `IncludePath` setting. Are you also hitting the `Could not source the setup file for ROS distro "melodic".` issue?

ooeygui, 2019-09-19:
We've been updating the vscode ROS extension. IF you find issues in the extension, please create an issue in https://github.com/ms-iot/vscode-ros

closed 2019-09-19 by ooeygui