ms-iot/ROSOnWindows #100: Can gsk_recogniztion_msgs be installed in ros for Windows?

JackRen88, 2019-04-09:
Hello.
I run choco install ros-mentric-gsk-recogniztion-msgs to install gsk_recogniztion_msgs,there is a problem that the package is not included in source list. How should I install the package?

seanyen, 2019-04-09:
@JackRen88  Thanks for your interest in this project! Currently we are releasing daily builds for common ROS variants (http://www.ros.org/reps/rep-0150.html). For additional stacks, I would suggest to start from `choco install ros-melodic-desktop_full` (https://ms-iot.github.io/ROSOnWindows/GettingStarted/Setup.html)  and try to make Windows ports for the extra packages.

ooeygui, 2019-05-17:
Hello @JackRen88,
You can clone this repository into your workspace and build it as part of your project.

closed 2019-05-17 by ooeygui