ms-iot/ROSOnWindows #101: missing libgen.h

Pyrokine, 2019-04-10:
when catkin the package, cmd shows that it lacks libgen.h.

seanyen, 2019-04-10:
@Pyrokine what package are you trying to build?

Pyrokine, 2019-04-11:
move_server package in the navigation package

Pyrokine, 2019-04-11:
I comment the wrong name. It's map_server

seanyen, 2019-04-11:
Hi @Pyrokine, can you share more information where you checkout the source code for `navigation` packages and the steps you took to build?

JFYI, as of today, the Windows build `navigation` code is in `ms-iot` fork: https://github.com/ms-iot/navigation/tree/init_windows, and the upstream repository doesn't have the ported changes yet (but the PRs are out, awaiting feedback.)

Pyrokine, 2019-04-12:
Thanks a lot. I built the original package from ros-planning/navigation. I found those package you forked from ros and it works.

closed 2019-04-12 by Pyrokine