ms-iot/ROSOnWindows #102: _tf2.pyd was replaced with tf2.lib, causes error

alecGraves, 2019-04-12:
**Describe the bug**
C:\opt\ros\melodic\x64\lib\site-packages\tf2_py needs a file "_tf2.pyd", but in recent choco upgrade, "tf2.lib" is copied into this folder instead.

traceback:

```
Traceback (most recent call last):
  File "C:\opt\ros\melodic\x64\lib\gazebo_ros\spawn_model", line 31, in <module>
    from tf.transformations import quaternion_from_euler
  File "C:\opt\ros\melodic\x64\lib\site-packages\tf\__init__.py", line 30, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "C:\opt\ros\melodic\x64\lib\site-packages\tf2_ros\__init__.py", line 38, in <module>
    from tf2_py import *
  File "C:\opt\ros\melodic\x64\lib\site-packages\tf2_py\__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: No module named _tf2
```

**To Reproduce**
```
*delete ros, start with new machine*
choco upgrade -y ros-melodic-desktop_full
python
>>> import tf
```

**Expected behavior**
"_tf2.pyd" is found in folder C:\opt\ros\melodic\x64\lib\site-packages\tf2_py

**Environment information:**
 - OS version: 1809 (OS Build 17763.437)
 - Chocolatey Versions: `choco list --local-only`
```
Chocolatey v0.10.11
...
ros-melodic-desktop_full 20190411.1
```

seanyen, 2019-04-12:
@alecGraves Thanks for the reporting! I will take a look what's going wrong soon.

seanyen, 2019-06-12:
Closed this issue since it is fixed already in the latest ROS installation.

closed 2019-06-12 by seanyen