ms-iot/ROSOnWindows #103: [roscore] The code execution cannot proceed because DLL not found

sgolodetz, 2019-04-12:
After following the standard installation instructions, I've been trying to run `roscore`, but I get the following errors when it gets to starting `/rosout`:

> The code execution cannot proceed because {boost_date_time-vc141-mt-x64-1_66.dll/boost_regex-vc141-mt-x64-1_66.dll/boost_system-vc141-mt-x64-1_66.dll/boost_thread-vc141-mt-x64-1_66.dll/log4cxx.dll} was not found. Reinstalling the program may fix this problem.

Please do you know how I should go about installing the Boost 1.66 dependency? I was hoping it would be installed by `rosdep`, but it doesn't seem to have worked. I'm loath to build Boost manually and copy across the relevant DLLs - doesn't seem like a very clean solution.

seanyen, 2019-04-19:
@sgolodetz Thank for reporting this issue. Do you have the output for your `rosdep`? Did it try to `choco install` packages from `https://roswin.azurewebsites.net/`?

budongyue, 2019-04-27:
I also meet this problem.What I should do can make it ok?Thanks.

sgolodetz, 2019-04-27:
Thanks for the response @seanyen, and sorry for the delay - I'll get the rosdep output off my work laptop next week when I'm back in the office and post it.

ooeygui, 2019-05-17:
Closing due to inactivity.

AmandaLW, 2019-06-12:
What was the solution for this? I am having the same problem.

budongyue, 2019-06-13:
@AmandaLW  Installation is not successful.You can install it again.If your Internet speed is slow,you can set the time limit of timeout when you use choco,

gptshubham595, 2019-08-21:
same problem

 I would like to say that the folder 
C:\opt\rosdeps\x64\bin\
doesn't contain any dll files even on reinstalling

gptshubham595, 2019-09-18:
SAME PROBLEM

ooeygui, 2019-09-18:
Reopening. 
ROS is very sensitive to the environment, and environmental conflicts can cause issues like this. We're trying to identify the environment that leads to this condition.

gptshubham595, 2019-09-18:
MY FILES ARE LISTED HERE https://github.com/gptshubham595/ROS-WINDOWS-FILES
(*I WILL DELETE THIS REPO)

Chocolatey v0.10.15
assimp 4.0.1.1809242322
boost 1.66.0.1904102339
bullet3 2.87.0.1809242326
bzip2 1.0.6.1901242108
chocolatey 0.10.15
chocolatey-core.extension 1.3.3
chocolatey-windowsupdate.extension 1.0.4
console_bridge 0.4.0.1809242322
cppunit 1.12.1.1809242322
cppzmq 4.2.2.1811162207
dlfcn-win32 1.1.1.1811162226
eigen 3.3.4.1809242322
freeimage 3.17.0.1811160037
gazebo9 9.4.1.1903240636
git 2.22.0
git.install 2.22.0
google-mock 1.8.0.1809242326
google-test 1.8.0.1809242322
gtk2 2.22.1.1904131648
hg 5.0.2
ilmbase 2.2.1.1811210800
jasper 2.0.14.1811210919
jxrlib 1.1.0.1811210809
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
lcms 2.8.0.1811210903
libcurl 7.58.0.1809242322
libflann 1.9.1.1809090720
libgraphviz 2.41.0.1810022039
libjpeg-turbo 1.5.3.1809242322
liblzma 5.2.4.1811210837
libogg 1.3.3.1809090720
libopencv 3.4.1.1809242322
libpcl 1.8.1.1809090720
libpng 1.6.35.1809242322
libraw 0.19.0.1811210804
libtbb-dev 2018.6.0.1812011750
libtheora 1.1.1.1809090720
libwebp 0.6.1.1811210731
log4cxx 0.10.0.1809242322
lz4 1.8.1.1809242322
ogre 1.10.11.1901312214
openexr 2.2.1.1811210832
openjpeg 2.3.0.1811210736
openssl 1.1.1.1809242322
pkg-config 0.29.2.1809242322
poco 1.8.1.1809242322
protobuf 3.6.1.1811130545
pyqt5 5.10.1.1809242322
pyside2 5.10.1.1809242322
qt5-sdk 5.10.1.1903182340
qwt 6.1.3.1811212304
ros-catkin-tools 0.0.1.1903221831
ros-cmake 3.11.4.1809242322
ros-melodic-desktop_full 20190729.1
ros-melodic-moveit 20190729.1
ros-python2 2.7.15.1904300710
ros-vcpkg 2018.11.23.1906112226
sbcl 0.0.0.1809242326
sdl 1.2.15.1809242322
sdl_image 1.2.12.1809242322
sip 4.19.8.1809242322
tiff 4.0.9.1811210742
tinyxml 2.6.2.1809242322
tinyxml2 6.1.0.1809242322
urdfdom 1.0.0.1809242322
urdfdom_headers 1.0.0.1809242322
vcpython27 9.0.30729.1809090720
vcredist140 14.22.27821
yaml-cpp 0.6.2.1901170034
zeromq 4.2.5.1811162203
zlib 1.2.11.1809242322
75 packages installed.

gptshubham595, 2019-09-18:
ONLY Error due to boost*.DLL files

I have put them now in vcpkg/packages

ooeygui, 2019-09-18:
Can you post the output of of the command environment?
open a command window with 'run as administrator'
Inside:
```
c:\opt\ros\melodic\x64\setup.bat
set
```

Post the output. (Please remove any personal information from the output)

gptshubham595, 2019-09-18:
ALLUSERSPROFILE=C:\ProgramData
APPDATA=C:\Users\Shubham\AppData\Roaming
ChocolateyInstall=C:\ProgramData\chocolatey
ChocolateyLastPathUpdate=132097646948150083
CMAKE_PREFIX_PATH=D:/opt/ros/melodic/x64;D:/opt/rosdeps/x64;D:/opt/vcpkg/installed/x64-windows
CommonProgramFiles=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
COMPUTERNAME=SHUBHAM
ComSpec=C:\WINDOWS\system32\cmd.exe
CUDA_PATH=C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v9.0
CUDA_PATH_V9_0=C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v9.0
DriverData=C:\Windows\System32\Drivers\DriverData
FPS_BROWSER_APP_PROFILE_STRING=Internet Explorer
FPS_BROWSER_USER_PROFILE_STRING=Default
GTK_BASEPATH=C:\Program Files (x86)\GtkSharp\2.12\
HOMEDRIVE=C:
HOMEPATH=\Users\Shubham
JAVA_HOME=c:\Program Files (x86)\Java\jre1.8.0_211
LD_LIBRARY_PATH=D:/opt/ros/melodic/x64\lib
LOCALAPPDATA=C:\Users\Shubham\AppData\Local
LOGONSERVER=\\SHUBHAM
NUMBER_OF_PROCESSORS=12
OneDrive=C:\Users\Shubham\OneDrive - Indian Institute of Technology Guwahati
OneDriveCommercial=C:\Users\Shubham\OneDrive - Indian Institute of Technology Guwahati
OneDriveConsumer=C:\Users\Shubham\OneDrive
OS=Windows_NT
Path=c:\opt\vcpkg;D:/opt/ros/melodic/x64\bin;D:/opt/ros/melodic/x64\lib;D:/opt/rosdeps/x64\bin;D:\opt\python27amd64\;D:\opt\python27amd64\Scripts;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\ProgramData\Oracle\Java\javapath;C:\ProgramData\Anaconda3;C:\ProgramData\Anaconda3\Library\mingw-w64\bin;C:\ProgramData\Anaconda3\Library\usr\bin;C:\ProgramData\Anaconda3\Library\bin;C:\ProgramData\Anaconda3\Scripts;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\iCLS\;C:\Program Files\Intel\Intel(R) Management Engine Components\iCLS\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files (x86)\GtkSharp\2.12\bin;D:\Program Files\nodejs\;C:\Program Files\MATLAB\R2019a\bin;C:\ProgramData\chocolatey\bin;D:\Program Files (x86)\Universal Extractor;D:\Program Files (x86)\Universal Extractor\bin;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;D:\Program Files\Microsoft VS Code\bin;D:\Program Files\CMake\bin;C:\Program Files\Git\cmd;C:\Program Files\dotnet\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;%SystemRoot%\system32;%SystemRoot%;%SystemRoot%\System32\Wbem;%SYSTEMROOT%\System32\WindowsPowerShell\v1.0\;%SYSTEMROOT%\System32\OpenSSH\;C:\Users\Shubham\AppData\Local\Programs\Python\Python37-32\Scripts\;C:\Users\Shubham\AppData\Local\Programs\Python\Python37-32\;C:\Users\Shubham\AppData\Local\Microsoft\WindowsApps;D:\Program Files\Microsoft VS Code\bin;C:\Users\Shubham\AppData\Roaming\npm;c:\Program Files\nc111nt;c:\adb\adb.exe;c:\Program Files (x86)\Java\jre1.8.0_211\bin\java.exe;c:\Program Files (x86)\Java\jre1.8.0_211\bin\;C:\Program Files\Mercurial\;D:\opt\ros\melodic\x64\bin\roscore;D:\opt\ros\melodic\x64\bin;C:\Users\Shubham\AppData\Local\Programs\Python\Python37-32\python.exe;D:\opt\rosdeps\x64\bin;D:\opt\ros\melodic\x64\lib;D:\flutter\bin;c:\Windows\System32;c:\Windows\SysWOW64;D:\flutter;C:\;c:\Program Files (x86)\Java\jre1.8.0_211\bin;c:\Program Files (x86)\Java\jre1.8.0_211;C:\opt\python27amd64\Scripts;C:\opt\python27amd64;C:\opt\ros\melodic\x64\lib;D:\Program Files\heroku\bin;D:\Program Files (x86)\CodeBlocks\MinGW\bin;D:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\;D:\Program Files (x86)\Microsoft Visual Studio\2019\Community;D:\;D:\Program Files (x86)\Microsoft Visual Studio\2019;D:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin;
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
PKG_CONFIG_PATH=D:/opt/ros/melodic/x64\lib\pkgconfig
PROCESSOR_ARCHITECTURE=AMD64
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 158 Stepping 10, GenuineIntel
PROCESSOR_LEVEL=6
PROCESSOR_REVISION=9e0a
ProgramData=C:\ProgramData
ProgramFiles=C:\Program Files
ProgramFiles(x86)=C:\Program Files (x86)
ProgramW6432=C:\Program Files
PROMPT=$P$G
PSModulePath=C:\Program Files\WindowsPowerShell\Modules;C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules
PUBLIC=C:\Users\Public
PYTHONHOME=D:\opt\python27amd64\
PYTHONPATH=D:/opt/ros/melodic/x64\lib/site-packages
QT_PLUGIN_PATH=C:\ProgramData\Anaconda3\Library\plugins
ROS_DISTRO=melodic
ROS_ETC_DIR=D:/opt/ros/melodic/x64/etc/ros
ROS_MASTER_URI=http://127.0.0.1:11311
ROS_PACKAGE_PATH=D:\opt\ros\melodic\x64\share
ROS_PYTHON_VERSION=2
ROS_ROOT=D:/opt/ros/melodic/x64/share/ros
ROS_VERSION=1
SESSIONNAME=Console
SystemDrive=C:
SystemRoot=C:\WINDOWS
TEMP=C:\Users\Shubham\AppData\Local\Temp
TMP=C:\Users\Shubham\AppData\Local\Temp
USERDOMAIN=SHUBHAM
USERDOMAIN_ROAMINGPROFILE=SHUBHAM
USERNAME=Shubham
USERPROFILE=C:\Users\Shubham
VBOX_MSI_INSTALL_PATH=C:\Program Files\Oracle\VirtualBox\
VCPKG_ROOT=c:\opt\vcpkg
windir=C:\WINDOWS

ooeygui, 2019-09-18:
In your output you have a mix of C drives and D drives. 

*ASK:* Could you tell me how you got into this situation? ROS install is currently only supported on the C drive.

*Long story*
During the build, catkin (the ROS build system) bakes into the output files the hard coded 'install path', so it expects to install into c:\opt. Your install seems to be redirected to D which breaks that assumption.

This assumption is baked into the Linux codebase, which was migrated to the Windows build. Because of this, installs of ROS into something other than c:\opt is unfortunately not supported.

ooeygui, 2019-09-18:
Linking https://github.com/ms-iot/ROSOnWindows/issues/142

ooeygui, 2019-09-19:
@gptshubham595 I see that you have anaconda installed. In the ROS/VS window, could you run 
```
where python.exe
```
What I suspect is happening is that during the ROS install, we initialize the sideloaded python in c:\opt with things like rosdeps. However, when running python the anaconda version is used, which doesn't include rosdeps, so that portion fails.

@gptshubham595 Could you try the following to confirm this theory?

Please remove the d:\opt install all together. It isn't currently supported.
Run 
```
c:\opt\ros\melodic\x64\setup.bat
c:\ProgramData\chocolatey\lib\ros-melodic-desktop_full\tools\rosdepInstall.bat
```

Paste the output.

Then in that same window set the environment back to c:\opt and remove anaconda:
```
set path=c:\opt\vcpkg;c:/opt/ros/melodic/x64\bin;c:/opt/ros/melodic/x64\lib;c:/opt/rosdeps/x64\bin;c:\opt\python27amd64;c:\opt\python27amd64\Scripts;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\iCLS;C:\Program Files\Intel\Intel(R) Management Engine Components\iCLS;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0;C:\Windows\System32\OpenSSH;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\WiFi\bin;C:\Program Files\Common Files\Intel\WirelessCommon;C:\Program Files (x86)\GtkSharp\2.12\bin;D:\Program Files\nodejs;C:\Program Files\MATLAB\R2019a\bin;C:\ProgramData\chocolatey\bin;D:\Program Files (x86)\Universal Extractor;D:\Program Files (x86)\Universal Extractor\bin;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;D:\Program Files\Microsoft VS Code\bin;D:\Program Files\CMake\bin;C:\Program Files\Git\cmd;C:\Program Files\dotnet;C:\Program Files\Microsoft SQL Server\130\Tools\Binn;%SystemRoot%\system32;%SystemRoot%;%SystemRoot%\System32\Wbem;%SYSTEMROOT%\System32\WindowsPowerShell\v1.0;%SYSTEMROOT%\System32\OpenSSH;C:\Users\Shubham\AppData\Local\Microsoft\WindowsApps;D:\Program Files\Microsoft VS Code\bin;
```

Then run rosdep again and paste the output:
```
c:\ProgramData\chocolatey\lib\ros-melodic-desktop_full\tools\rosdepInstall.bat
```
Thank you.

seanyen, 2020-09-22:
Closing it and in favor of using the meta ticket.

closed 2020-09-22 by seanyen