ms-iot/ROSOnWindows #104: There is a error for running gazebo on windows10 ros.

JackRen88, 2019-04-13:
When running `roslaunch gazebo_ros empty_world.launch` on windows ros, there is a error that is that because of not finding jpeg62.dll,can't continue to execute code.How should I solved it?

ancabilloni, 2019-04-13:
I ran into this issue as well. Found out there was a package was not installed correctly which is listed in  `C:\ProgramData\chocolatey\lib-bad`. This one https://chocolatey.org/packages/libjpeg-turbo , after choco install it, the issue was resolved. 

It also has something to do with ros update version to me. When I was using the update version in March, I did not have this issue. After uninstall ros and choco upgrade to the latest update this week, I ran into that issue.

JackRen88, 2019-04-15:
@ancabilloni .
Thanks!

seanyen, 2019-04-15:
@JackRen88 Thank for reporting this issue!

@ancabilloni Just curious, do you still keep the logs to show why it was listed in `lib-bad`? That may help for future improvement.

ancabilloni, 2019-04-16:
@seanyen i did not find any log on terminal indicating to the error. When trying to run gazebo, a windows just popped up saying jpeg62.dll not found and ask to re-install to resolve, nothing else. Finding the relevant package to reinstall on `lib-bad` was pure luck as I search around for relevant jpeg package. If you can instruct any log file I should look at to detect on the problem, I can send the info. Thanks.

ooeygui, 2019-05-17:
Closing as gazebo is working

closed 2019-05-17 by ooeygui