ms-iot/ROSOnWindows #105: build errors with moveit_tutorials

bakerhillpins, 2019-04-17:
**Describe the bug**
I have been following the windows tutorial for move it: 
[ROSOnWindows - MoveIt Tutorials](https://ms-iot.github.io/ROSOnWindows/Moveit/moveit_tutorials.html)
During the catkin_make process the following build error occurs on several different elements:

> c:\opt\rosdeps\x64\include\eigen3\eigen\src/Geometry/Transform.h(331): error C2338: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
C:\Users\xx\catkin_ws\src\moveitTutorials\src\moveit_tutorials\doc\robot_model_and_robot_state\src\robot_model_and_robot_state_tutorial.cpp(135): note: see reference to function template instantiation 'Eigen::Transform<double,3,1,0>::Transform<2,0>(const Eigen::Transform<double,3,2,0> &)' being compiled

**To Reproduce**
Steps to reproduce the behavior:
Simply follow the workspace setup, git clone, and build process outlined at [ROSOnWindows - MoveIt Tutorials](https://ms-iot.github.io/ROSOnWindows/Moveit/moveit_tutorials.html)

**Expected behavior**
No errors during build of the moveit_tutorials project

bakerhillpins, 2019-04-17:
The problem above boils down to a previously solved issue in the upstream [ros-planning/moveit_tutorials](https://github.com/ros-planning/moveit_tutorials) project. Specifically, replacing all occurrences of `Affine3d `with `Isometry3d`

See upstream pull request:
https://github.com/ros-planning/moveit_tutorials/pull/255

I've made my changes locally and have successfully executed the tutorial as it sits. I'm hoping that this defect will help others past the problem and I'd also like to understand how I could possibly help by putting together a PR bringing in the upstream changes.

Just forking the ms-iot repos and pulling in from the upstream isn't really what's confusing me on this en devour.  I've spent some time trying to understand how you might like us to make contributions in this way but I'm not quite sure about what branches you all are working from (both upstream and locally) and if directly pulling from the upstream is proper. Short story- I couldn't find or possibly I didn't find or understand, how to make contributions to the ROSonWindows project in this manner.

bakerhillpins, 2019-04-23:
I'm willing to try to help put together a pull request for this but I'd like to understand what project/branches I should be working against.

ooeygui, 2019-05-17:
HI @bakerhillpins,
Sorry for the delay. We've recently upstreamed the core of ROS and are working to expand the projects we're upstreaming. The next batch is moveit - which includes rebasing the init_windows branch and doing the pull requests; then moving our builds over.

mgdelmonte, 2020-02-24:
What's the status on moveit_tutorials for windows?  The docs at https://ms-iot.github.io/ROSOnWindows/Moveit/moveit_tutorials.html currently say to use

`git clone https://github.com/ms-iot/moveit_tutorials -b init_windows`

but there is no `init_windows` branch in that repo -- and never has been.

Attempting to build from `master` fails with `fatal error C1083: Cannot open include file: 'moveit/collision_detection_fcl/collision_env_fcl.h': No such file or directory`

Any suggestions?

wkeram, 2020-02-24:
I'm facing the same issue as @mgdelmonte. Any help would be great.

ooeygui, 2020-02-24:
Thanks for the ping on this. @lilustga is working on the MoveIt tutorials.

wkeram, 2020-03-18:
Thanks for updating the docs. I was now able to build it. 
However, if I follow the MoveIt Tutorials and try

`roslaunch panda_moveit_config demo.launch rviz_tutorial:=true`

it fails with
`RLException: included file [C:\moveit_ws\src\panda_moveit_config\launch\planning_context.launch] requires the 'True' arg to be set`.

From the log file:

> File "C:\opt\ros\melodic\x64\lib\site-packages\roslaunch\__init__.py", line 330, in main
> p.start()
> File "C:\opt\ros\melodic\x64\lib\site-packages\roslaunch\parent.py", line 289, in start
> self._start_infrastructure()
> File "C:\opt\ros\melodic\x64\lib\site-packages\roslaunch\parent.py", line 238, in _start_infrastructure
> self._load_config()
> File "C:\opt\ros\melodic\x64\lib\site-packages\roslaunch\parent.py", line 144, in _load_config
> roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
> File "C:\opt\ros\melodic\x64\lib\site-packages\roslaunch\config.py", line 462, in load_config_default
> raise RLException(e)
> RLException: while processing C:\moveit_ws\src\panda_moveit_config\launch\demo.launch:
> included file [C:\moveit_ws\src\panda_moveit_config\launch\planning_context.launch] requires the 'True' arg to be set

Any ideas how to fix that error?

lilustga, 2020-03-18:
Hi @wkeram,

Glad the updated docs worked to fix your build issues.

I haven't seen the "requires the 'True' arg error before but I'll look into it.

For now, one thing to try would be remove the single quotes (or replace with `&quot;`) in the commands in planning_context.launch as they don't work correctly on Windows. This open PR has the single quote replacement. Still discussing the best long term solution.
https://github.com/ros-planning/panda_moveit_config/pull/60

wkeram, 2020-03-18:
Hi @lilustga,

I already removed the single quotes as I encountered the same error as described in issue #106. 
After that I had the "requires the 'True' arg" error.

It would be great if you could help me.

lilustga, 2020-03-18:
Which branch/remote repository did you clone from?

wkeram, 2020-03-18:
I followed the instructions in the docs:
`git clone https://github.com/ros-planning/moveit_tutorials.git -b melodic-devel`
`git clone https://github.com/ros-planning/panda_moveit_config.git -b melodic-devel`
`git clone https://github.com/frankaemika/franka_ros-release -b release/kinetic/franka_description`

I also tried `git clone https://github.com/frankaemika/franka_ros-release -b release/melodic/franka_description` instead but the error remains.

lilustga, 2020-03-21:
hi @wkeram,

I've tried to reproduce your error but so far been unable to. Is it possible that an error was introduced in planning_context.launch? The error message seems nonsensical to me. Perhaps try replacing it with this version:
https://github.com/ms-iot/panda_moveit_config/blob/melodic-devel/launch/planning_context.launch

lilustga, 2020-03-27:
@wkeram, any update on this? Did that work?

Chemaslane, 2020-05-14:
> @wkeram, any update on this? Did that work?

It worked for me. I had the same error as wkeram, just removing all the single quotes. With the file you posted it works.

lilustga, 2020-05-14:
Thanks @Chemaslane! will mark as closed for now. Fix for the single quotes issue is in development.

closed 2020-05-14 by lilustga