ms-iot/ROSOnWindows #106: invalid mode ('r') or filename:

JackRen88, 2019-04-18:
When I start a launch file(<param command="$(find xacro)/xacro  '$(find car_model)/robots/Car.xacro'" name="robot_description"/> is included),there is a issue that 
`invalid mode ('r') or filename: 'C:\catkin_ws\src\car_model\robots\Car.xacro'
RLException: Invalid <param> tag: Cannot load command parameter [robot_description]: command [C:/opt/ros/melodic/x64\lib\xacro\xacro.exe  'C:\catkin_ws\src\car_model\robots\Car.xacro'] returned with code [2].

Param xml is <param command="$(find xacro)/xacro  '$(find car_model)/robots/Car.xacro'" name="robot_description"/>
The traceback for the exception was written to the log file`
How should I solve it?

seanyen, 2019-04-19:
@JackRen88 Thank for reporting the issue. Can you try to run `rosrun xacro xacro -vvv C:\catkin_ws\src\car_model\robots\Car.xacro` to see more verbose logs?

bakerhillpins, 2019-06-05:
@JackRen88 
I think the problem you're seeing relates to the single quote characters in the command line of the xacro file.  Take them out and it will load the xacro file.

Not sure what issues this might have with windows and spaces in path names.

Ex:
```
<?xml version="1.0"?>
<launch>
  <param name="robot_description" command="$(find xacro)/xacro --inorder '$(find mybot_description)/urdf/mybot.xacro'" />
</launch>
```
Causes:
> 
> [91minvalid mode ('r') or filename: 'C:\Users\x\catkin_ws2\src\mybot_description\urdf\mybot.xacro'[0m
> [31mRLException: Invalid <param> tag: Cannot load command parameter [robot_description]: command [C:/opt/ros/melodic/x64\lib\xacro\xacro.exe --inorder 'C:\Users\x\catkin_ws2\src\mybot_description\urdf\mybot.xacro'] returned with code [2].

Remove the extra single quotes:
```
<param name="robot_description" command="$(find xacro)/xacro --inorder $(find mybot_description)/urdf/mybot.xacro" />
```

ooeygui, 2019-09-19:
@bakerhillpins is correct. the Single quotes are not needed and cause issues on Windows. (also --inorder is not needed on melodic)

closed 2019-09-19 by ooeygui

scottnortman, 2020-07-17:
Want to confirm the above solution; I too had the same problem when running on Windows after following these instructions:

https://moveit.ros.org/install/
https://ros-planning.github.io/moveit_tutorials/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html

By editing lines of the 

C:\moveit_ws\src\panda_moveit_config\launch\planning_context.launch

 with the *.udrf.xacro and removing the single quotes I was able to run the launch file and RViz