ms-iot/ROSOnWindows #107: choco upgrade ros-melodic-desktop_full can timeout and fail to install

ooeygui, 2019-04-18:
During the installation of ros-melodic-desktop_full, the rosdep install can timeout causing chocolatey to report an install failure. Rerunning the command can succeed depending on how many rosdeps have been installed.

Attempt to add a timeout bypass inside the chocolatey package, otherwise update doc to add the chocolatey flag `--execution-timeout=0` to the recommended command line.

ooeygui, 2019-04-18:
[chocolatey.log](https://github.com/ms-iot/ROSOnWindows/files/3095935/chocolatey.log)

budongyue, 2019-05-31:
You can try again by adding --execution-timeout=0 to your install command's end.For an example:
 choco upgrade ros-melodic-desktop_full -y --execution-timeout=0

As for choco command,it use 45 minutes as its fault timeout.And the flag --execution-timeout=0  can set the timeout as endless.

seanyen, 2019-06-04:
I closed this issue since I updated the installation Wiki. http://wiki.ros.org/Installation/Windows?action=diff&rev1=7&rev2=8

closed 2019-06-04 by seanyen

ig-or, 2020-02-06:
Hello! still having this issue; tried to install ROS about 50 times in a different ways; the result is the same

```
2020-02-06 12:49:14,878 12364 [INFO ] - [NuGet] Attempting to resolve dependency 'KB3033929 (= 1.0.3)'.
2020-02-06 12:50:54,936 12364 [ERROR] - ros-melodic-desktop_full not installed. An error occurred during installation:
 The operation has timed out
```

ooeygui, 2020-02-06:
@ig-or Are you by chance behind a proxy?

ig-or, 2020-02-06:
THANKS a lot for a so quick answer.....  Actually I did it successfully on two Windows10 PCs, but with different Window10 build version and everything was OK..  But on this PC there were issues.

Now I suspect I just forget to install git first, I can see installation progress.. Let you know it it installed OK in 20 minutes.