ms-iot/ROSOnWindows #110: choco upgrade ros-melodic-desktop_full -y  errors:Error while running 'C:\ProgramData\chocolatey\lib\ros-python2\tools\chocolateyinstall.ps1'.

budongyue, 2019-04-27:
[image]

gavanderhoorn, 2019-04-27:
Appears to be a cross-post of [Error while running 'C:\ProgramData\chocolatey\lib\ros-python2\tools\chocolateyinstall.ps1'](https://answers.ros.org/question/322035) on ROS Answers.

seanyen, 2019-04-27:
@budongyue can you do `set ComSpec` and `where cmd` to see if `cmd.exe` visible in your command prompt? Usually it exists at `c:\windows\syetem32\cmd.exe`.

ooeygui, 2019-04-28:
Investigating.

ooeygui, 2019-04-28:
@seanyen We should use the %comspec% variable in the install scripts in case the path isn't set correctly.

ooeygui, 2019-04-28:
Confirmed that this is environment specific. It does not repro on clean installs or upgrades.

budongyue, 2019-04-30:
@seanyen Yes,I can see the cmd after do set ComSpec and where cmd

seanyen, 2019-04-30:
@budongyue Thanks for the confirm. I have updated the `ros-python2` package to use `%ComSpec%` to run batch scripts. Can you run again `choco upgrade ros-python2 -y` to see if you still see the error?

ooeygui, 2019-05-17:
Please reopen if this isn't fixed for you

closed 2019-05-17 by ooeygui