ms-iot/ROSOnWindows #116: How do I use the VS in RosOnWindow

budongyue, 2019-05-04:
I want to know the method of using VS in RosOnWindows. I want to know how to edit and compile the code by using VS and add the code compiled by VS to ROS. Thanks very much.

gavanderhoorn, 2019-05-04:
Cross-post of [How do I use the VS in RosOnWindows](https://answers.ros.org/question/322449) on ROS Answers.

traversaro, 2019-05-08:
(Disclaimer: I am not related in any way to the ROSOnWindows project, so this is not an official answer). 

I have successfully used the C++ libraries installed by ROSOnWindows's Chocolatey packages in a plain CMake project simply by running the `"C:\opt\ros\melodic\x64\setup.bat"` script in a terminal and then by launching CMake from that terminal. This was possible because the setup scripts sets the [`CMAKE_PREFIX_PATH` environmental variable](https://cmake.org/cmake/help/latest/variable/CMAKE_PREFIX_PATH.html) that is used by [CMake's `find_package`command](https://cmake.org/cmake/help/v3.10/command/find_package.html) to find its dependencies, including the ROS's libraries. For an example of using ROS's libraries from a plain CMake project, see https://github.com/gerkey/ros1_external_use.

Once you have a plain CMake project that use ROS's libraries, you can easily generate a Visual Studio solution for it. For more details on this, see the [related CGold section](https://cgold.readthedocs.io/en/latest/first-step/generate-native-tool/gui-visual-studio.html). 

Note that this is a bit different from the classical strategy of using `catkin` to build ROS-dependent code. I personally have no experience of using `catkin` to generate Visual Studio solutions. 

**Note: the ROSOnWindows Chocolatey repository is currently not providing Debug libraries binaries, so in general you cannot use the Debug Visual Studio configuration to build code, for more info see https://github.com/ms-iot/ROSOnWindows/issues/31.**

ooeygui, 2019-05-17:
We are working on releasing a ROS vscode extension which supports windows. in the mean time, you can build it yourself from here: https://github.com/ms-iot/vscode-ros

closed 2019-05-17 by ooeygui