ms-iot/ROSOnWindows #117: When I use cartographer on gazebo,there are some problems.

budongyue, 2019-05-08:
I use the tutorials：https://ms-iot.github.io/ROSOnWindows/Turtlebot/Turtlebot3.html
When I use cartographer on gazebo,I use "roslaunch turtlebot3_gazebo turtlebot3_gazebo_cartographer_demo.launch"
But I get some problems:
[ INFO] [1557301488.083588696, 8.307000000]: Using plugin "inflation_layer"
[spawn_urdf-4] process has finished cleanly
log file: C:\Users\syx\.ros\log\117d7121-7165-11e9-bfb5-9822eff15670\spawn_urdf-4*.log
[FATAL] [1557301488.219322779, 8.411000000]: Failed to create the dwa_local_planner/DWAPlannerROS planner, are you sure it is properly registered and that the containing library is built? Exception: Could not find library corresponding to plugin dwa_local_planner/DWAPlannerROS. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[31m[move_base-6] process has died [pid 6676, exit code 1, cmd C:\opt\ros\melodic\x64\lib\move_base\move_base.exe cmd_vel:=/cmd_vel odom:=odom __name:=move_base __log:=C:\Users\syx\.ros\log\117d7121-7165-11e9-bfb5-9822eff15670\move_base-6.log].
log file: C:\Users\syx\.ros\log\117d7121-7165-11e9-bfb5-9822eff15670\move_base-6*.log[0m
F0508 15:44:48.868984 17992 map_by_time.h:43] Check failed: data.time > std::prev(trajectory.end())->first (621355968085380000 vs. 621355968085380000)
[FATAL] [1557301489.434209753, 8.603000000]: F0508 15:44:48.000000 17992 map_by_time.h:43] Check failed: data.time > std::prev(trajectory.end())->first (621355968085380000 vs. 621355968085380000)
*** Check failure stack trace: ***
[31m[cartographer_node-7] process has died [pid 3948, exit code -1073740791, cmd C:\opt\ros\melodic\x64\lib\cartographer_ros\cartographer_node.exe -configuration_directory C:/opt/ros/turtlebot3_ws/turtle3/src\turtlebot3\turtlebot3_slam\config -configuration_basename turtlebot3_lds_2d_gazebo.lua /imu:=/flat_imu __name:=cartographer_node __log:=C:\Users\syx\.ros\log\117d7121-7165-11e9-bfb5-9822eff15670\cartographer_node-7.log].
log file: C:\Users\syx\.ros\log\117d7121-7165-11e9-bfb5-9822eff15670\cartographer_node-7*.log
What happened?
Thanks very much.

ooeygui, 2019-05-16:
We had a bug with the dwa planner. We're working on moving it to public. For now, you can upgrade it using:
choco upgrade ros-melodic-desktop_full -y --pre

Then the DWA planner will be present.

closed 2019-05-16 by ooeygui

c650, 2019-07-29:
I think we should keep this issue open until the normal release is working.. As of last week it still wasn't. :)