ms-iot/ROSOnWindows #118: When I catkin_make the navigation's codes,there are some errores occured.

budongyue, 2019-05-11:
When I catkin_make the navigation's codes,there are some errores occured.
Now,I am trying to use the navigation's codes,but some errores comed.
The source is https://github.com/ms-iot/navigation.
I can't complie the code.I hope some persons can help me.Thanks very much.
[image]

budongyue, 2019-05-13:
NMAKE : fatal error U1073: don't know how to create “gtest-NOTFOUND”
Stop.
NMAKE : fatal error U1077: “"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe"”: return code“0x2”
Stop.
NMAKE : fatal error U1077: “"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe"”: return code“0x2”
Stop.
Invoking "nmake" failed

If you have any ideas,please tell me.Thanks very much.

ooeygui, 2019-05-16:
Hi there,
Did you run rosdep in the catkin directory?
rosdep update
rosdep install --from-paths src --ignore-src -r -y

budongyue, 2019-05-18:
I did it just now.But there are some problems:
1. rosdep install --from-paths src --ignore-src -r -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
navfn: No definition of [netpbm] for OS [windows]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

And I install the netpbm by pip.There are some errors.
2.  C:\navigation_ros>pip install netpbm
Collecting netpbm
  Using cached https://files.pythonhosted.org/packages/e4/85/5922ed47c1cfd659b7cff42fe17749bd4e08ad4bf2cc3f318d3236670b17/netpbm-0.0.1.zip
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "c:\users\syx\appdata\local\temp\pip-install-sop3x9\netpbm\setup.py", line 54, in <module>
        with open(str(MODULE_PATH), "r", encoding="utf-8") as source_code_file:
    TypeError: 'encoding' is an invalid keyword argument for this function

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in c:\users\syx\appdata\local\temp\pip-install-sop3x9\netpbm\
You are using pip version 18.1, however version 19.1.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

Thanks for your answer.

seanyen, 2019-05-20:
@budongyue 
Did you try to checkout the navigation code `init_windows` branch? For example, https://github.com/ms-iot/navigation/tree/init_windows?

We typically don't directly check-in changes to `melodic-devel` and we push the windows specific changes to `init_windows` instead.

budongyue, 2019-05-21:
seanyen
Yes,you are right.Thanks very much.I have successed.

budongyue, 2019-05-21:
@seanyen 
Yes,you are right.Thanks very much.

seanyen, 2019-05-21:
@budongyue Sounds good. I will close this issue. And thanks for your confirmation.

closed 2019-05-21 by seanyen