ms-iot/ROSOnWindows #122: DEBUG versions for *core* ROS libraries would be VERY helpful (and static too)

codenotes, 2019-06-01:
It would be *extremely* helpful to include debug versions/builds of the *core* ROS libraries (not all of rosdeps) and, if possible, static library versions.  I have a lot of years working with ROS on windows (I worked a little with Brian et. al in the past and was responsible for the cygwin port of all of ROS that was out there for Indigo a few years back).   I am super excited to see what MSFT is doing here, saved me and others an enormous amount of work and will really help the community.  Thank you. 

But back to debug...

Most of us developing ROS on windows will be using VS when authoring our own nodes.  We will still use catkin, but for debugging VS is king.  It is easy enough at present to use commandline cmake to generate VS projects for a package.  But DEBUG can not be used in the corresponding VS projects for a number of reasons (debug/release mismatch, class size/memory allocator differences, etc.).   No one needs to be debugging the ROS source code, but to debug one's own code, it is kind of important to important to be in DEBUG. (In fact, you can't even call ros::init without a "invalid size" exception being thrown immediately if you have DEBUG enabled...which makes sense as we're mixing Release libraries with Debug client code) 

What I would suggest:
1) compile just the *core* ROS libraries in debug x64 and include in the chocolaty package. They can be anywhere, no one needs to make catkin do this.  They can be included just for those who know what they are doing with VS generally and require it...no need to make everything more complex.
2) you do NOT need to include debug versions of the boost libraries: it would be great, but most folks can build their own boost libraries. 

In my experience with ROS on windows (having ported the core libs and TF2 et. al) they are very happy and stable in debug mode and only really require debug boost builds.  The tools like rviz and such have far more dependencies, but I am suggesting just the core ROS libraries like:

#pragma comment(lib, "roscpp.lib")
#pragma comment(lib, "rosconsole.lib")
#pragma comment(lib, "roscpp_serialization.lib")
#pragma comment(lib, "rostime.lib")
//and the TF libs

folks can do 99pct of what they need with just these.  So static and debug versions of these would be a dream and should not be tough to do.   I am *not* suggesting debug versions of all rosdeps.  Just a handful of core libraries so folks like me can write and debug our own code in ROS nodes.

traversaro, 2019-06-01:
Related issue and discussion: https://github.com/ms-iot/ROSOnWindows/issues/31 .

codenotes, 2019-06-02:
Yeah, I read the tinyXML thread...it is related.   I understand the perspective of not including the DEBUG as doing so could add a lot of complexity and maintenance for something only a few use,  but I would say again to powers that be: just include the debug versions in a special place/folder...don't go  and modify catkin or worry about the toolset and other dependencies.   Just a few core libraries.  It may be that to compile DEBUG for even a few libraries requires a lot of catkin/cmake headaches...but I found that simply importing the cmake files into CMAKE GUI, it was pretty trivial to generate VS projects and then compile them to debug (and also for static libs).  I did this for kinetic and indigo and I can do it for melodic if need be. Someone would just have a guide me a bit as to how to best collaborate. 

On the subject of static libs, these can be more than nice to have.  When integrating ROS in complicated host systems (like game or simulation engines or embedded/mobile devices [like android NDK]) it can be really problematic to have a ton of DLLs hanging around.   In some cases, like for mobile, it can be impossible to handle.  I did a ROS port to android that required all static as there are restrictions on the .so loading for that platform that got in the way. And, anyway, there could be no .so versions of ROS without android-arm-compiled boost libraries...

So I would say: ROS for Windows really means ROS for VS and VS is really about cross-compilation and targeting everything these days and towards that end, we could really use the debug and static lib versions of ROS.

traversaro, 2019-06-04:
I totally agree that having libraries in Debug will be extremely convenient. With respect to that, I guess something is moving:
* A GSoC student is working on adding vcpkg support to bloom ( https://summerofcode.withgoogle.com/projects/#5110132554334208 , https://www.openrobotics.org/blog/tag/GSoC )
* According to a recent ROS2 TSC meeting minute, Microsoft is working on "a vcpkg feature to export chocolatey packages" ( https://discourse.ros.org/t/tsc-meeting-minutes-2019-04-18/8832 )

IoTDan, 2022-10-11:
Thank you for the suggestion. we understand the desire and appreciate the time you took writing this up. At this time, we recommend that if you need the ability to debug an unoptimized distribution, it would be more beneficial to clone the source code and build a local ROS distribution. You can leverage the build scripts http://github.com/ms-iot/ros-windows-build.

closed 2022-10-11 by IoTDan