ms-iot/ROSOnWindows #124: [DevOps] Could NOT find PythonInterp: Found unsuitable version "1.4", but required is at least "2"

seanyen, 2019-06-07:
**Describe the bug**
By using Azure DevOps pipelines for CI, I added the following step in the flow:
```yaml
  - script: |
      call "C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Auxiliary\Build\vcvars64.bat"
      call "C:\opt\ros\melodic\x64\setup.bat"
      pushd "%Build_StagingDirectory%\catkin_ws"
      catkin_make_isolated --only-pkg-with-deps kobuki_core --install
```

Then, when I kickoff a build, this error shows up:
```
-- This workspace overlays: C:/opt/ros/melodic/x64
CMake Error at C:/opt/rosdeps/x64/share/cmake-3.11/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find PythonInterp: Found unsuitable version "1.4", but required
  is at least "2" (found C:/ProgramData/chocolatey/bin/python2.exe)
```

**Expected behavior**
Kick off a DevOps build without this error.

seanyen, 2019-06-07:
https://github.com/Kitware/CMake/blob/v3.11.0/Modules/FindPythonInterp.cmake#L90-L108

FindPythonInterp.cmake is using system registry key to search for `PYTHON_EXECUTABLE`. On machines with multiple pythons installed system-wide, it could eventually pick a wrong one than the one from ROS installation.

Adding `-DPYTHON_EXECUTABLE=c:/opt/python27amd64/python.exe` can override this behavior to get around this problem. For example,
```yaml
  - script: |
      call "C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Auxiliary\Build\vcvars64.bat"
      call "C:\opt\ros\melodic\x64\setup.bat"
      catkin_make_isolated -DPYTHON_EXECUTABLE=c:/opt/python27amd64/python.exe
```

closed 2019-06-07 by seanyen

846524417, 2020-07-18:
Have you solved the mistake? I'm making the same mistake now. I hope I can get your help. thank you!

ooeygui, 2020-07-18:
Hi @846524417 ,
I think you are asking about a different problem. This bug is about using Python in a ROS DevOps environment, which @seanyen offered a correction for.

846524417, 2020-07-20:
@ooeygui Yes, but it's the same for me to report an error. I can't find a solution yet. Do you know what caused this mistake?

ooeygui, 2020-07-20:
@846524417 The solution presented here, is to pass the python executable to catkin.
```
catkin_make_isolated -DPYTHON_EXECUTABLE=c:/opt/python27amd64/python.exe
```

846524417, 2020-07-21:
@ooeygui  My platform is Android studio, can I handle it like this?

ooeygui, 2020-07-21:
@846524417 I'm sorry, we can't help you. This bug is unrelated. Perhaps try stack overflow or the android studio github.