ms-iot/ROSOnWindows #131: Developing ROSpackage in VIsual Studio

Brabalawuka, 2019-07-15:
**Describe the bug**
I am a rookie and it may be due to my misoperation. May I ask is there any way to get a intellience when I create a .cpp file under the new package src folder?

**To Reproduce**
Steps to reproduce the behavior:
-Create a new package in command line  
-Open devenv and create a new cpp node  
-The new cpp file is not getting inteliience such as when including header files  

**Environment information:**
 - OS version: `windows 10 1903`
 - Chocolatey Versions: `choco list --local-only`

Brabalawuka, 2019-07-15:
Fixed by adding lines in Cmakelist.txt

closed 2019-07-15 by Brabalawuka

seanyen, 2019-07-15:
@Brabalawuka Do you mind to share the lines you added in `CMakeLists.txt`? It might help for others who are also running into the same issue.

Brabalawuka, 2019-07-16:
Haha is it a simple solution just remember to

`add_executable(@packagename src/@filename.cpp)`