ms-iot/ROSOnWindows #133: boost_system-vc141-mt-x64-1_66.dll

jchen706, 2019-07-17:
Ran roscore: 
This happen:

The code execution cannot proceed because boost_system-vc141-mt-x64-1_66.dll was not found. Reinstalling the program may fix it this problem.

Then I try the force install: ran roscore same problem

seanyen, 2019-07-17:
@jchen706 Thanks for the reporting! To understand the situation, can you run `choco list -lo` to list what's installed in your system? And when you said the force install, did you do `choco upgrade ros-melodic-desktop_full -y --force`?

And please note that `git` needs to be available in your command prompt before the installation, you can run `git --version` to check.

majid1268, 2019-07-19:
Dear @seanyen,  I have the same issue, please find more information in following

```
C:\Windows\System32>choco list -lo
Chocolatey v0.10.15
boost-msvc-12 1.58.0
chocolatey 0.10.15
chocolatey-core.extension 1.3.3
chocolatey-windowsupdate.extension 1.0.4
git 2.22.0
git.install 2.22.0
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
ros-catkin-tools 0.0.1.1903221831
ros-melodic-desktop_full 20190415.1
ros-python2 2.7.15.1904300710
vcredist140 14.21.27702.2
15 packages installed.

C:\Windows\System32>choco upgrade ros-melodic-desktop_full -y --force
Chocolatey v0.10.15
Upgrading the following packages:
ros-melodic-desktop_full
By upgrading you accept licenses for the packages.
ros-melodic-desktop_full v20190415.1 is the latest version available based on your source(s).
Progress: Downloading ros-melodic-desktop_full 20190415.1... 100%

ros-melodic-desktop_full v20190415.1 (forced)
ros-melodic-desktop_full package files upgrade completed. Performing other installation steps.
Extracting 64-bit C:\ProgramData\chocolatey\lib\ros-melodic-desktop_full\tools\drop.zip to c:\opt\ros...
c:\opt\ros

running rosdep...
"Could not create temporary file: C:\Users\Braendler Annotator\AppData\Local\Temp\chocolatey\setup.18575061.bat"
 The upgrade of ros-melodic-desktop_full was successful.
  Software installed to 'c:\opt\ros'

Chocolatey upgraded 1/1 packages.
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).

C:\Windows\System32>

```

majid1268, 2019-07-19:
```running rosdep..."Could not create temporary file: C:\Users\Braendler Annotator\AppData\Local\Temp\chocolatey\setup.18575061.bat" The upgrade of ros-melodic-desktop_full was successful.  Software installed to 'c:\opt\ros'Chocolatey upgraded 1/1 packages. See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).C:\Windows\System32>```

seanyen, 2019-07-19:
@majid1268 I think the caveat is that your account with spaces and somehow the installation doesn't work with that. Before the team figure out the root cause, can you try to create another local account without spaces and retry again?

@jchen706 Could you check if you were running into that same case?

ooeygui, 2019-08-01:
I have a local repro of this. It is related to the space in the name. Root causing

majid1268, 2019-08-02:
@ooeygui and @seanyen thank you for your responses. I have not had a chance to test it, but it seems correct and it is related to space in the name.

gptshubham595, 2019-08-13:
sorry but @majid1268  can u tell how you resolved this

seanyen, 2019-08-13:
@gptshubham595 
After you run `c:\opt\ros\melodic\x64\setup.bat` do you the following related ROS paths in `%PATH%`?

```
> set PATH
Path=C:/opt/ros/melodic/x64\bin;C:/opt/ros/melodic/x64\lib;C:/opt/rosdeps/x64\bin;C:/opt/rosdeps/x64\lib;C:\opt\python27amd64\;C:\opt\python27amd64\Scripts;...

> where boost_system-vc141-mt-x64-1_66.dll
C:\opt\rosdeps\x64\bin\boost_system-vc141-mt-x64-1_66.dll
```

gptshubham595, 2019-08-13:
inside my this folder there is only these files 
`C:\opt\rosdeps\x64\bin`

`cmake, cmake-gui,cmcldeps,cpack,ctest`

rx78kok, 2019-09-19:
Hi im having the same problem too. Force install `choco upgrade ros-melodic-desktop_full -y --force` didnt fix the issue.

Folder `C:\opt\rosdeps\x64\bin` does not have boost_system-vc141-mt-x64-1_66.dll.
 
I hope there is some solution for this.
```
C:\Windows\System32>choco list -lo
Chocolatey v0.10.15
asio 1.12.1
assimp 4.0.1.1809242322
boost 1.66.0.1904102339
bullet3 2.87.0.1809242326
bzip2 1.0.6.1901242108
chocolatey 0.10.15
chocolatey-core.extension 1.3.3
chocolatey-windowsupdate.extension 1.0.4
cmake 3.15.3
cmake.install 3.15.3
console_bridge 0.4.0.1809242322
cppunit 1.12.1.1809242322
cppzmq 4.2.2.1811162207
dlfcn-win32 1.1.1.1811162226
eigen 3.3.4.1809242322
freeimage 3.17.0.1811160037
gazebo9 9.4.1.1909060324
git 2.23.0
git.install 2.23.0
google-mock 1.8.0.1809242326
google-test 1.8.0.1809242322
gtk2 2.22.1.1904131648
hg 5.1.1
ilmbase 2.2.1.1811210800
jasper 2.0.14.1811210919
jxrlib 1.1.0.1811210809
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
lcms 2.8.0.1811210903
libcurl 7.58.0.1809242322
libflann 1.9.1.1809090720
libgraphviz 2.41.0.1810022039
libjpeg-turbo 1.5.3.1809242322
liblzma 5.2.4.1811210837
libogg 1.3.3.1809090720
libopencv 3.4.1.1809242322
libpcl 1.8.1.1809090720
libpng 1.6.35.1809242322
libraw 0.19.0.1811210804
libtbb-dev 2018.6.0.1812011750
libtheora 1.1.1.1809090720
libwebp 0.6.1.1811210731
log4cxx 0.10.0.1809242322
lz4 1.8.1.1809242322
ogre 1.10.11.1901312214
openexr 2.2.1.1811210832
openjpeg 2.3.0.1811210736
openssl 1.1.1.1809242322
pkg-config 0.29.2.1809242322
poco 1.8.1.1809242322
protobuf 3.6.1.1811130545
pyqt5 5.10.1.1809242322
pyside2 5.10.1.1809242322
python 3.7.4
python3 3.7.4
qt5-sdk 5.10.1.1903182340
qwt 6.1.3.1811212304
ros-catkin-tools 0.0.1.1903221831
ros-colcon-tools 0.0.1.1905240527
ros-melodic-desktop_full 20190729.1
ros-python2 2.7.15.1904300710
ros-python37 3.7.2.1903082244
ros-vcpkg 2018.11.23.1906112226
sbcl 0.0.0.1809242326
sdl 1.2.15.1809242322
sdl_image 1.2.12.1809242322
sip 4.19.8.1809242322
tiff 4.0.9.1811210742
tinyxml 2.6.2.1809242322
tinyxml-usestl 2.6.2
tinyxml2 6.1.0.1809242322
urdfdom 1.0.0.1809242322
urdfdom_headers 1.0.0.1809242322
vcpython27 9.0.30729.1809090720
vcredist140 14.22.27821
yaml-cpp 0.6.2.1901170034
zeromq 4.2.5.1811162203
zlib 1.2.11.1809242322
81 packages installed
```

ooeygui, 2019-09-19:
@rx78kok We're working to understand what is happening. We do know that once a system gets into this stage, chocolatey won't repair itself. Could you post your environment?
From within your visual studio command prompt:
```
c:\opt\ros\melodic\x64\setup.bat
set
```

Please remove any personally identifiable information like tokens and usernames.

Do you happen to have python installed before attempting to install ROS?

rx78kok, 2019-09-20:
```
C:\Windows\System32>set
ALLUSERSPROFILE=C:\ProgramData
APPDATA=C:\Users\syafiq\AppData\Roaming
asl.log=Destination=file
ChocolateyInstall=C:\ProgramData\chocolatey
ChocolateyLastPathUpdate=132132666303836011
CMAKE_PREFIX_PATH=C:/opt/ros/melodic/x64;C:/opt/rosdeps/x64;C:/opt/vcpkg/installed/x64-windows
CommandPromptType=Native
CommonProgramFiles=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
COMPUTERNAME=DESKTOP-A1I1L5G
ComSpec=C:\WINDOWS\system32\cmd.exe
DevEnvDir=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\
DriverData=C:\Windows\System32\Drivers\DriverData
ExtensionSdkDir=C:\Program Files (x86)\Microsoft SDKs\Windows Kits\10\ExtensionSDKs
Framework40Version=v4.0
FrameworkDir=C:\WINDOWS\Microsoft.NET\Framework64\
FrameworkDIR64=C:\WINDOWS\Microsoft.NET\Framework64
FrameworkVersion=v4.0.30319
FrameworkVersion64=v4.0.30319
HOMEDRIVE=C:
HOMEPATH=\Users\syafiq
INCLUDE=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.22.27905\ATLMFC\include;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.22.27905\include;C:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um;C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt;C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared;C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um;C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt;C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt
LD_LIBRARY_PATH=C:/opt/ros/melodic/x64\lib;C:/opt/rosdeps/x64\lib
LIB=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.22.27905\ATLMFC\lib\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.22.27905\lib\x64;C:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\lib\um\x64;C:\Program Files (x86)\Windows Kits\10\lib\10.0.18362.0\ucrt\x64;C:\Program Files (x86)\Windows Kits\10\lib\10.0.18362.0\um\x64;
LIBPATH=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.22.27905\ATLMFC\lib\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.22.27905\lib\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.22.27905\lib\x86\store\references;C:\Program Files (x86)\Windows Kits\10\UnionMetadata\10.0.18362.0;C:\Program Files (x86)\Windows Kits\10\References\10.0.18362.0;C:\WINDOWS\Microsoft.NET\Framework64\v4.0.30319;
LOCALAPPDATA=C:\Users\syafiq\AppData\Local
LOGONSERVER=\\DESKTOP-A1I1L5G
MOZ_PLUGIN_PATH=C:\Program Files (x86)\Foxit Software\Foxit Reader\plugins\
NETFXSDKDir=C:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\
NUMBER_OF_PROCESSORS=4
OneDrive=C:\Users\syafiq\OneDrive
OpenCV_DIR=C:\opencv-3.4.1-vc15.VS2017\opencv
OPENSSL_CONF=C:\OpenSSL-Win64\bin\openssl.cfg
OS=Windows_NT
Path=c:\opt\vcpkg;C:/opt/ros/melodic/x64\bin;C:/opt/ros/melodic/x64\lib;C:/opt/rosdeps/x64\bin;C:/opt/rosdeps/x64\lib;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.22.27905\bin\HostX64\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\VC\VCPackages;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\TestWindow;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\TeamFoundation\Team Explorer;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\bin\Roslyn;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Team Tools\Performance Tools\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Team Tools\Performance Tools;C:\Program Files (x86)\Programming\visual studio\2017\Shared\Common\VSPerfCollectionTools\vs2019\\x64;C:\Program Files (x86)\Programming\visual studio\2017\Shared\Common\VSPerfCollectionTools\vs2019\;C:\Program Files (x86)\Microsoft SDKs\Windows\v10.0A\bin\NETFX 4.6.1 Tools\x64\;C:\Program Files (x86)\Windows Kits\10\bin\10.0.18362.0\x64;C:\Program Files (x86)\Windows Kits\10\bin\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\\MSBuild\Current\Bin;C:\WINDOWS\Microsoft.NET\Framework64\v4.0.30319;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\Tools\;C:\opt\python27amd64\;C:\opt\python27amd64\Scripts;C:\Python37\Scripts\;C:\Python37\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Programming\Matlab\2018\bin;C:\Program Files\dotnet\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\ProgramData\chocolatey\bin;C:\OpenSSL-Win64\bin\;C:\Users\syafiq\.dnx\bin;C:\Program Files\Microsoft DNX\Dnvm\;C:\opencv-3.4.1-vc15.VS2017\opencv\x64\vc15\bin;c:\opencv-2.4.13.2-vc14.VS2015\x64\vc14\bin;C:\Program Files\CMake\bin;C:\ProgramData\chocolatey\lib\tinyxml2\lib;C:\ProgramData\chocolatey\lib\log4cxx\lib;C:\nuget;C:\Program Files\Git\cmd;C:\opt\rosdeps\x64\bin;C:\opt\rosdeps\x64\lib;C:\opt\ros\melodic\x64\lib;C:\opt\ros\melodic\x64\bin;C:\Users\syafiq\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Mercurial\;C:\Program Files (x86)\Microsoft Visual St [... cut ...] xml2\lib;C:\ProgramData\chocolatey\lib\log4cxx\lib;C:\nuget;C:\Program Files\Git\cmd;C:\opt\rosdeps\x64\bin;C:\opt\rosdeps\x64\lib;C:\opt\ros\melodic\x64\lib;C:\opt\ros\melodic\x64\bin;C:\Users\syafiq\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Mercurial\
```
Yes i have python installed before installing ROS because i need it to run rosdep when installing ROS. If not i cant installed the dependency package for ROS. Is there any other way to run rosdep without python when installing ROS?

ooeygui, 2019-09-20:
@rx78kok Thank you for the environment. During the install, the ROS Chocolatey package installs Python and seeds it with pips needed to initialize the environment - including rosdeps. What we've seen is with a mixed environment an unexpected install of python gets seeded, and rosdeps fails. 

While we have code to protect against this situation, we need to replicate this scenario to see how we can guard against it.

Can you tell me how you installed python before running the ROS chocolatey package? 
Was there something you hit that made you install it before running rosdeps?

To fix this, remove python from your system environment before running the Visual Studio command prompt. Then 

```
c:\ProgramData\chocolatey\lib\ros-melodic-desktop_full\tools\rosdepInstall.bat
```

If this command fails, could you post the output?

rx78kok, 2019-09-23:
Hi. Sorry for the late reply. 
I just installed python normally by downloading from www.python.org. Dont think i did anything that would affect the installation.

This is the error i get when execute command above.
```
C:\Windows\System32>c:\ProgramData\chocolatey\lib\ros-melodic-desktop_full\tools\rosdepInstall.bat
Fatal error in launcher: Unable to create process using '"c:\opt\python27amd64\python.exe"  "c:\opt\python27amd64\Scripts\rosdep.exe" init'
Fatal error in launcher: Unable to create process using '"c:\opt\python27amd64\python.exe"  "c:\opt\python27amd64\Scripts\rosdep.exe" update'
Fatal error in launcher: Unable to create process using '"c:\opt\python27amd64\python.exe"  "c:\opt\python27amd64\Scripts\rosdep.exe" install --from-paths c:\opt\ros\melodic\x64\share --ignore-src -r -y'
```

rx78kok, 2019-09-23:
@ooeygui Hi. Sorry for the late reply. 
I just installed python normally by downloading from www.python.org. Dont think i did anything that would affect the installation.

After i remove any related to python in my environment variables, i tried to execute the command

`c:\ProgramData\chocolatey\lib\ros-melodic-desktop_full\tools\rosdepInstall.bat`

The command above install all the package normally. 
Then i tried execute `c:\opt\ros\melodic\x64\setup.bat` and then `roscore`. But it still giving the same error as before and the missing .dll still not there. Should i try uninstall the python also?

TYIA.

ooeygui, 2020-01-21:
Closing due to inactivity. Please reactivate if you are still hitting this issue.

closed 2020-01-21 by ooeygui