ms-iot/ROSOnWindows #136: Catkin tools: catkin build support

ltstein, 2019-07-26:
Is Catkin build supported in ROSonWindows or only catkin_make commands?

kejxu, 2019-07-30:
Thanks for asking! Not exactly familiar with voxblox (can't open readthedocs.io, maybe the server is temporarily down) or your use case, but I have come to notice that [`catkin_tools`](https://github.com/catkin/catkin_tools) itself is a standalone package aside from `catkin`. Since I don't have too much experience using the `catkin build` command (`catkin_make_isolated` generally provides sufficient functionality), so I'm not sure if it's only because the package itself did not get installed correctly.

Maybe it would be helpful to check if the entry script for `catkin_tools` is existing in the Python environment that this package was installed.

Thanks again

ooeygui, 2019-07-31:
Hi @ltstein,
We are not planning on windows enabling the catkin tools project. There is a thread on [ros answers](https://answers.ros.org/question/289655/what-is-the-status-of-catkin-tools/) about `catkin_make` vs `catkin build`. 

For Windows, we are optimizing for the Open Robotics recommended build tools for both ROS1 and ROS2.

closed 2019-07-31 by ooeygui