ms-iot/ROSOnWindows #141: There is a error for running gazebo on windows10 ros when running Turtlebot3

kyawdevi, 2019-08-14:
Gazebo not working 
When running roslaunch gazebo_roslaunch turtlebot3_gazebo turtlebot3_gazebo_cartographer_demo.launch on windows ros, ,can't continue to execute code.How should I solved it?
Rviz and other tools working fine but gaze shows loading screen like gaze is loading 
and then it stopped 
I reinstall many times but gazebo specifically didnt work for me

I tried this method also which is in C:\ProgramData\chocolatey\lib-bad. This one https://chocolatey.org/packages/libjpeg-turbo , after choco install it, but it didnt work

Error on terminal 

'gazebo' is not recognized as an internal or external command,
operable program or batch file.

I am new to this

seanyen, 2019-08-14:
> Rviz and other tools working fine but gaze shows loading screen like gaze is loading
> and then it stopped

@kyawdevi could you share a screenshot for the loading screen? And there should be some logs for `gazebo` under `C:\Users\<your name>\.gazebo`. Are you able to share them as well?

ooeygui, 2019-10-24:
@kyawdevi Please reopen if you can still repro

closed 2019-10-24 by ooeygui